espressif/arduino-esp32 #7551: RTC_DATA_ATTR not incremented

0xPr0xy, 2022-12-05:
### Board

ESP32S3

### Device Description

Development board, nothing else

### Hardware Configuration

Just the micro usb connected to uart

### Version

v2.0.5

### IDE Name

VSCode

### Operating System

MacOS 12.6.1

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

RTC_DATA_ATTR isn't incremented when using deep sleep

### Sketch

```cpp
#include "Arduino.h"

RTC_DATA_ATTR int bootCount = 0;

void sleep(void *ptr)
{
  esp_sleep_enable_timer_wakeup(10000000);

  Serial.println("Setup ESP32 to sleep for every " + String(10000000) +
               " Seconds");

  Serial.println("Going to sleep now");
  Serial.println("current state" + String(bootCount));

  esp_deep_sleep_start();
}

void setup()
{
  Serial.begin(115200);

  ++bootCount;

  xTaskCreatePinnedToCore(sleep, "sleepTask", 4095, NULL, 3, NULL, 1);
}

void loop()
[... 306 lines cut ...]
16:34:32.481 -> mode:DIO, clock div:1
16:34:32.481 -> load:0x3fce3808,len:0x43c
16:34:32.481 -> load:0x403c9700,len:0xbec
16:34:32.481 -> load:0x403cc700,len:0x2a3c
16:34:32.481 -> entry 0x403c98d8
16:34:32.507 -> Setup ESP32 to sleep for every 10000000 Seconds
16:34:32.507 -> Going to sleep now
16:34:32.507 -> current state6
16:34:42.491 -> ESP-ROM:esp32s3-20210327
16:34:42.491 -> Build:Mar 27 2021
16:34:42.491 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
16:34:42.528 -> SPIWP:0xee
16:34:42.528 -> mode:DIO, clock div:1
16:34:42.528 -> load:0x3fce3808,len:0x43c
16:34:42.528 -> load:0x403c9700,len:0xbec
16:34:42.528 -> load:0x403cc700,len:0x2a3c
16:34:42.528 -> entry 0x403c98d8
16:34:42.567 -> Setup ESP32 to sleep for every 10000000 Seconds
16:34:42.567 -> Going to sleep now
16:34:42.567 -> current state7
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-12-05:
I see you are using arduino-esp32 2.0.5 + IDF 4.4.2-dirty here.  What is the target framework for ArduinoIDE?
I recall something about RTC variables in the base IDF project.  This may be an IDF issue.

0xPr0xy, 2022-12-06:
Am using version 1.8.19 for Arduino IDE. 
What do you recall exactly?
Let me also note that I am using a task in the IDF example due to [this](https://github.com/espressif/esp-idf/issues/8569).

Randalix, 2023-07-04:
I'm having the same issue with version 2.0.8. Did you find a solution?

closed 2025-01-16 by Parsaabasi