espressif/arduino-esp32 #7568: Need help with ESP32 DevkitV1

JRock57, 2022-12-09:
### Board

ESP32 DEVKIT V1

### Device Description

I can't get the device to connect using Arduino IDE v.1.8.19  I am unable to select a port to connect.

### Hardware Configuration

Nothing else is attached

### Version

other

### IDE Name

Arduino IDE v. 1.8.19

### Operating System

Windows 11

### Flash frequency

40-60Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Unanble to select a COM Port

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Invalid library found in C:\Users\jfo84\OneDrive\Documents\Arduino\libraries\libraries: no headers files (.h) found in C:\Users\jfo84\OneDrive\Documents\Arduino\libraries\libraries
Invalid library found in C:\Users\jfo84\OneDrive\Documents\Arduino\libraries\libraries: no headers files (.h) found in C:\Users\jfo84\OneDrive\Documents\Arduino\libraries\libraries
```

### Other Steps to Reproduce

I tried multiple versions of the IDE

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-12-10:
It seems to be a hardware problem. 
It can be the USB cable, the ESP32 board or the Host USB port in the computer.

Try replacing each one of those.

VojtechBartoska, 2022-12-15:
any updates on this @JRock57?

RoCorbera, 2023-01-28:
@JRock57 - Have you installed the USB device driver for the ESP32 DEVKIT V1 board you have?
Some boards use the CP210x and other use the CH340 chips for UART-USB.

You can check if the dirver is missing when opening the Device Manager and checking "Other Devices". It shall be empty.
If it has something, that means that the necessary device driver isn't installed.

After installing the driver, the COM port shall show up in the Arduino IDE.

Other problem may be the USB cable. Sometimes those cables only have the charging wires and miss the data wires.
You can check the cable by using it with a smartphone and checking if the phone shows up in Windows as a folder.
If it doesn't, that means that the cable works only for charging.

JRock57, 2023-01-28:
Yes thank you. That's actually an old post I will remove.

closed 2023-01-28 by SuGlider