espressif/arduino-esp32 #7569: Erro ao iniciar SD card e Data e Hora irregulares na ESP32

HellryMoraes, 2022-12-10:
### Board

Esp32 dev Module

### Device Description

MicroSD Card adapter com cartão de memória de 2GB e Real Time Clock RTC DS3231 com Bateria (CR2032)

### Hardware Configuration

[image]

esse é esquema que estou usando

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Ao testar o código, o SD card não se comunica com placa e não gera as informações que deveria no monitor serial, além do mais, a data e a hora estão erradas 

### Sketch

```cpp
já pesquisei sobre esse tipo de erro, mas até agora não encontrei nada
```

### Debug Message

```plain
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
Erro ao iniciar a comunicacao com o cartao SD...
ets Jun  8 2016 00:22:57
```

### Other Steps to Reproduce

O código só funciona quando é apagada a linha de ação do SD Card no código fonte.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

HellryMoraes, 2022-12-10:
Não funciona sem o DS3231, o código não compila sem ele. Mas sem o SD, funciona

SuGlider, 2022-12-10:
@HellryMoraes - all communication in this repository is done in English, please. 
As said, please provide the smallest possible code that reproduces such issue.

SuGlider, 2022-12-10:
I think that the DS3231 is 3.3v instead of 5v.
I also think that the solenoid shouldn't be directly connected to the ESP32... It draws a lot of current.
Use some relay to activate it.

VojtechBartoska, 2023-02-16:
Hello, closing this issues as expired due to no answer. If needed, you can reopen this.

closed 2023-02-16 by VojtechBartoska