espressif/arduino-esp32 #7572: Platformio broken with actual master

Jason2866, 2022-12-10:
### Board

every

### Device Description

not relevant

### Hardware Configuration

not relevant

### Version

latest master (checkout manually)

### IDE Name

Platformio (custom self build)

### Operating System

macOS 12.3

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

@me-no-dev 
Compile fails with Platformio at the stage where esptool.py merge command is done.
https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py#L91 

There are no `bootloader*.bin` files anymore in folder `sdk/esp*/bin` just `*.elf` files

### Sketch

```cpp
Every sketch
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-12-11:
@Jason2866 that is true. We need PIO to update their stuff to generate bins, like we do in the IDE.

cc @ivankravets @valeros

Jason2866, 2022-12-11:
@me-no-dev Interested in reason for the change. What is the benefit for using *.elf files?

me-no-dev, 2022-12-12:
@Jason2866 if we use precompiled bins, the bootloader complains that the checksum of the firmware is wrong any time that settings (partitions, size, speed, etc.) differ from what the bootloader was compiled with. In order to get around that, we need to use elfs and turn them into bins with the same settings as the firmware at build time.

@ivankravets you are welcome :) Sorry I did not mention this earlier!

TD-er, 2022-12-12:
What does this do to OTA updates?

me-no-dev, 2022-12-12:
@TD-er as long as you keep the settings the same, it does nothing to OTA. You will see the error if you change flash speed/size.

TD-er, 2022-12-12:
So no issues with checksums?

me-no-dev, 2022-12-12:
no issues with checksums. elf2bin is done based on flash/psram settings, not based on the firmware image.

closed 2022-12-12 by me-no-dev

Jason2866, 2022-12-12:
Thx! Tested and working.

PR #7579 (2022-12-12): Update PlatformIO integration scripts
This PR brings updates according to the latest changes in the Arduino core:
- Bootloader binaries are now generated from elf files
- Updated CI scripts
- Updated esptoolpy to v4.2.1
- Minor clean-up by removing obsolete code

Resolves #7572
commit 3af0b44f (2022-12-12): Update PlatformIO build script (#7579)

This PR brings updates according to the latest changes in the Arduino core:
- Bootloader binaries are now generated from elf files
- Updated CI scripts
- Updated esptoolpy to v4.2.1
- Minor clean-up by removing obsolete code

Resolves #7572