espressif/arduino-esp32 #7578: USBHIDKeyboard print function does not reliably output uppercase characters

dbhaig, 2022-12-11:
### Board

ESP32-S3-WROOM1

### Device Description

ESP32-S3-DevKitC-1

### Hardware Configuration

SPI TFT is connected to GPIOs 13, 14, 15, 16, 21, 35, 36, 47

### Version

v2.0.5

### IDE Name

arduino-cli 0.29

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Keyboard.println("abcdefghijklmnopqrstuvwxyz"); 
Gives the expected output of:  'abcdefghijklmnopqrstuvwxyz'

Keyboard.println("ABCDEFGHIJKLOMNPQRSTUVWXYZ");
Gives an output with some characters lower case, such as:   
'abCdEFGHIjklOMNPQRStUvwXYZ' 
 or
 ' aBcdeFGHIjKLOMNPQrsTuVWXYz'

### Sketch

```cpp
#include "USB.h"
#include "USBHIDKeyboard.h"

USBHIDKeyboard Keyboard;

void setup() {
    Serial.begin(9600);
    Serial.println("Keyboard begin");

    Keyboard.begin();
    USB.begin();
}

void loop() {
    Keyboard.println("abcdefghijklmnopqrstuvwxyz");
    delay(2000);
    Keyboard.println("ABCDEFGHIJKLOMNPQRSTUVWXYZ");
    delay(2000);
}
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

No

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

RefactorFactory, 2022-12-12:
Does #6964 help?

dbhaig, 2022-12-12:
@RefactorFactory It is unclear to me how #6964 could be related to the issue I'm observing where the ESP32-S3 is recognized as a keyboard but uppercase characters are sometimes rendered as their corresponding lowercase character.

dbhaig, 2022-12-14:
This appears to be an arduino-libraries/Keyboard issue, as code compiled for the ATmega32u4 exhibits similar behaviour.

closed 2022-12-14 by dbhaig