espressif/arduino-esp32 #7582: Mounting SPIFFS failed on ESP32-C3

amibk, 2022-12-12:
### Board

ESP32-C3FN4

### Device Description

Nothing Attached

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello,Attempts to mount a LittleFS filesystem fails on ESP32-C3...
and ,,  I noticed that there was a similar problem [#6579](https://github.com/espressif/arduino-esp32/issues/6579) but when I set it to the QIO mode, it still did not work 
and the platformio.ini content is:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:airm2m_core_esp32c3]
platform = espressif32
board = airm2m_core_esp32c3
framework = arduino
board_build.partitions = partition.csv
monitor_speed = 115200
board_build.flash_mode = qio
board_build.f_cpu = 160000000L
board_build.filesystem = littlefs
monitor_filters = esp32_exception_decoder
lib_deps = 
	bblanchon/ArduinoJson@^6.19.4
	arduino-libraries/NTPClient@^3.2.1
	mathertel/OneButton@^2.0.3
	h2zero/NimBLE-Arduino@^1.4.0
	knolleary/PubSubClient@^2.8
```
use partition.csv content is:
```
# Name,   Type, SubType, Offset,    Size,   Flags
nvs,      data, nvs,     0x9000,    0x3000,
phy_init, data, phy,     ,          0x1000,

factory,  app,  factory, ,          3M,

spiffs,   data, spiffs,  ,          0xF0000,
```

### Sketch

```cpp
#include "SPIFFS.h"
#include "FS.h"

void setup() {
    SPIFFS.begin(false, "/spiffs", 5, NULL);
    File file = SPIFFS.open("/spiffs/config.txt", FILE_READ);
    if(!file){
        Serial.println("There was an error opening the file for writing");
        return;
    }
}
```

### Debug Message

```plain
E (198) SPIFFS: mount failed, -10025
[   194][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[   194][E][vfs_api.cpp:24] open(): File system is not mounted
There was an error opening the file for writing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2022-12-12:
Hello @amibk

I think this line `board_build.filesystem = littlefs` in your `platformio.ini` file sets the filesystem up as `littlefs` and I don't think you can mix it with SPIFFS unless I am mistaken.

Thanks
Felix

amibk, 2022-12-12:
> Hello @amibk
> 
> I think this line `board_build.filesystem = littlefs` in your `platformio.ini` file sets the filesystem up as `littlefs` and I don't think you can mix it with SPIFFS unless I am mistaken.
> 
> Thanks Felix

Hello @felmue ,
I deleted this line and tried again, but SPIFFS still doesn't work
Thanks

lbernstone, 2022-12-12:
Are you uploading a filesystem?  If not, then you need to initialize the fs on first boot (`SPIFFS.begin(true);`)

lbernstone, 2022-12-12:
Does the [SPIFFS example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_Test/SPIFFS_Test.ino) work?

amibk, 2022-12-13:
> Does the [SPIFFS example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_Test/SPIFFS_Test.ino) work?

Oh heck.. this demo is working, i create new project in PlatformIO IDE,no partition,no upload filesystem image , but it's working , dont why...

closed 2022-12-13 by amibk