espressif/arduino-esp32 #7585: undefined reference to 'vtable for fs::File'

tempdoom, 2022-12-13:
### Board

ESP32-S3

### Device Description

DevKitC

### Hardware Configuration

Nothing attached

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

While in the "linking" step of building and uploading the code from PlatformIO to the chip, LittleFS seems to have problems with a reference to FS. The same issue occurs if I switch to SPIFFS as well. I can't find any specific references to this problem anywhere else. 

### Sketch

```cpp
#include <LittleFS.h>

void setup() {
  LittleFS.begin();
}

void loop() {}
```

### Debug Message

```plain
c:/users/user/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\main.cpp.o:(.literal._ZN2fs4FileD5Ev[_ZN2fs4FileD5Ev]+0x0): undefined reference to `vtable for fs::File'

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tempdoom, 2022-12-14:
It's there.

igrr, 2022-12-15:
The sketch compiles successfully in Arduino IDE (https://wokwi.com/projects/351118022985908824), so I'd say this is a PlatformIO-specific issue. (I'm not a PIO expert, but I think you can find the solution to this by searching for `lib_ldf_mode=deep`)

Jason2866, 2022-12-16:
Can not reproduce
```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
Compiling .pio/build/esp32-s3-devkitc-1/src/main.cpp.o
Generating partitions .pio/build/esp32-s3-devkitc-1/partitions.bin
Compiling .pio/build/esp32-s3-devkitc-1/libbd5/FS/FS.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/libbd5/FS/vfs_api.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/libbd9/LittleFS/LittleFS.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/IPv6Address.cpp.o
Archiving .pio/build/esp32-s3-devkitc-1/libbd9/libLittleFS.a
Archiving .pio/build/esp32-s3-devkitc-1/libbd5/libFS.a
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/MD5Builder.cpp.o
Indexing .pio/build/esp32-s3-devkitc-1/libbd5/libFS.a
Indexing .pio/build/esp32-s3-devkitc-1/libbd9/libLittleFS.a
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/Print.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/USB.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/USBCDC.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/USBMSC.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/WString.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/base64.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-gpio.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-i2c-slave.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-i2c.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-ledc.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-matrix.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-misc.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-rgb-led.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32-s3-devkitc-1/libFrameworkArduino.a
Indexing .pio/build/esp32-s3-devkitc-1/libFrameworkArduino.a
Linking .pio/build/esp32-s3-devkitc-1/firmware.elf
Retrieving maximum program size .pio/build/esp32-s3-devkitc-1/firmware.elf
Checking size .pio/build/esp32-s3-devkitc-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.0% (used 9856 bytes from 327680 bytes)
Flash: [          ]   4.3% (used 143333 bytes from 3342336 bytes)
Building .pio/build/esp32-s3-devkitc-1/firmware.bin
esptool.py v4.4
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
======================================== [SUCCESS] Took 3.63 seconds ========================================
 *  Terminal will be reused by tasks, press any key to close it. 
 ```
Seems your platformio setup is screwed up.

igrr, 2022-12-17:
@tempdoom could you please post the full build log, like @Jason2866 did? Specifically interested in this part:
```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
```

(yours will probably not contain the last line with `FS`)

thaanstad, 2023-02-12:
I have a similar error with the FS libraries:
Error linking for board ESP32 Wrover Module(esp32_esp32wrover)
Build failed for project 'esp32InitialTest'
ld.exe: FS\\vfs_api.cpp.o:(.literal._ZN2fs8FileImplD5Ev[_ZN2fs8FileImplD5Ev]+0x0): undefined reference to vtable for fs::FileImpl

I can see this error when building the following code:
```
#include <FS.h>
#include <FSImpl.h>
#include <vfs_api.h>
#include <SPI.h>
#include <SD.h>
#include <sd_defines.h>
#include <sd_diskio.h>

void setup()
{
    delay(20000);
	Serial.begin(115200);
	SPI.begin(18, 19, 23, 32);
    

	pinMode(4, OUTPUT);  // Set D4 on Arduino to Output (4D Arduino Adaptor V2 - Display Reset)
	digitalWrite(4, 0);  // Reset the Display via D4
	delay(100);
	digitalWrite(4, 1);  // unReset the Display via D4
	delay(3500);
}

// Add the main program code into the continuous loop() function
void loop()
{
    static long pastTime = 0;
    static long interval = 1000;
	static int val = 1;

    if (millis() > pastTime + interval)
    {
        pastTime = millis();
        Serial.print("hello");
        Serial.println(val);
        val++;
    }

}
```

I do not see this error when building this code:
```
#include <FS.h>
#include <FSImpl.h>
#include <vfs_api.h>
#include <SPI.h>
//#include <SD.h>
//#include <sd_defines.h>
//#include <sd_diskio.h>

void setup()
{
    delay(20000);
	Serial.begin(115200);
	SPI.begin(18, 19, 23, 32);
    

	pinMode(4, OUTPUT);  // Set D4 on Arduino to Output (4D Arduino Adaptor V2 - Display Reset)
	digitalWrite(4, 0);  // Reset the Display via D4
	delay(100);
	digitalWrite(4, 1);  // unReset the Display via D4
	delay(3500);
}

// Add the main program code into the continuous loop() function
void loop()
{
    static long pastTime = 0;
    static long interval = 1000;
	static int val = 1;

    if (millis() > pastTime + interval)
    {
        pastTime = millis();
        Serial.print("hello");
        Serial.println(val);
        val++;
    }

}
```

This happens when I am using no compiler optimization for these libraries when I am trying to perform debugging using an ESP32 Wrover.  I am using Visual Micro in VS2022 running this arduino core.  Not sure if the issue is relating to visual micro or if there could be an improvement to the library to address this issue so the code can be compiled with no optimization.  It seems that including the SD library is causing the error (I'll try looking into this more)

Looking to learn and see what I can do to help resolve this issue but not sure where to get started.  It appears that this error is stating that the linker cannot find the function that is overwriting a virtual function in the FileImpl header.  I'll try looking into the SD library to see if I can figure anything out there.

Is there a method of obtaining additional information about which function is not being identified by the linker?  Why is compiler optimization allowing this function to be found and operate correctly but without the optimization the linker cannot find it?  Any help with this or direction to investigate would be much appreciated!

Jason2866, 2023-02-12:
What compiler options (flags) are set? Probably used one which is not supported.

thaanstad, 2023-02-12:
@Jason2866  Thanks for getting back to me so quickly.  Here is the portion of the build output that shows the issue I am seeing with the vtable linker error:

Optimization off:

```
Building libraries ...

Using library FS version 2.0.0 in folder "file:///C:/Arduino/arduino-1.8.13/portable/packages/esp32/hardware/esp32/2.0.6/libraries/FS"
"C:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.3\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/heap/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/log/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/soc/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/soc/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/hal/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_common/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_system/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/xtensa/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/driver/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/driver/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_pm/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_ringbuf/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools [... cut ...] hardware\\esp32\\2.0.6/tools/sdk/esp32/include/bt/esp_ble_mesh/api" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/cbor/port/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/unity/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/unity/unity/src" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hard

thaanstad, 2023-02-12:
(had to spilt post into two due to length) Default optimization on:
```

Building libraries ...

Using library FS version 2.0.0 in folder "file:///C:/Arduino/arduino-1.8.13/portable/packages/esp32/hardware/esp32/2.0.6/libraries/FS"
"C:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch5/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.3\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/heap/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/log/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/soc/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/soc/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/hal/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_common/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_system/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/xtensa/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/driver/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/driver/esp32/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_pm/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/esp_ringbuf/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/efuse/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\ [... cut ...] kages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/cbor/port/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/unity/include" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/unity/unity/src" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\esp32\\hardware\\esp32\\2.0.6/tools/sdk/esp32/include/cmock/CMock/src" "-IC:\\Arduino\\arduino-1.8.13\\portable\\packages\\e

TKTF50, 2023-02-13:
I am having a similar problem when compiling some projects for debugging. I reproduced it using @tempdoom's code with this platform.ini file:
```
[env:esp32]
; platform = espressif32 @ 5.2.0
platform = espressif32 @ 5.3.0
board = mhetesp32devkit
framework = arduino

build_type = debug
debug_build_flags = -O0 -ggdb3 -g3
```
There is no problem if I build the project using this configuration with the V5.2.0 platform, but with the V5.3.0 platform I get the linker error:

> c:/users/user/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32\lib874\libFS.a(vfs_api.cpp.o):(.literal._ZN2fs8FileImplD5Ev[_ZN2fs8FileImplD5Ev]+0x0): undefined reference to `vtable for fs::FileImpl'

The linker error also goes away with V5.3.0 if I comment out either of the **build_type** or **debug_build_flags** parameters. Based on the PlatformIO documentation, I didn't think there was any issue with combining these two parameters together.

Jason2866, 2023-02-13:
@TKTF50  Between 5.2.0 and 5.3.0 the toolchain changed from 8.4.0+2021r2-patch3 to 8.4.0+2021r2-patch.
Try if the error goes away with using the toolchain used in 5.2.0
```
platform_packages = platformio/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

Jason2866, 2023-02-13:
@thaanstad  You are using Arduino IDE. So it is NOT a Platformio issue.
This issue needs to be opened in [Espressif Arduino Github!](https://github.com/espressif/arduino-esp32/issues)

mrengineer7777, 2023-02-13:
> @thaanstad You are using Arduino IDE. So it is NOT a Platformio issue. This issue needs to be opened in [Espressif Arduino Github!](https://github.com/espressif/arduino-esp32/issues)

@Jason2866  This IS the Espressif Arduino Github.

@TKTF50 IDF 5.X isn't officially supported by this repository yet, but the devs are working on it in a new [branch](https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1) .

Jason2866, 2023-02-13:
@mrengineer7777
  > @Jason2866 This IS the Espressif Arduino Github. Nice catch on the toolchain change.

Ohhh yes, was jumping between issues ;-)

TKTF50, 2023-02-13:
@mrengineer7777 Sorry, but I'm not sure what you mean by this:
`@TKTF50 IDF 5.X isn't officially supported by this repository yet, but the devs are working on it in a new [branch] .`

If I change my platform.ini to refer to the latest version of expressif32 platform, ie:
```
[env:esp32]
platform = espressif32
board = mhetesp32devkit
framework = arduino

build_type = debug
debug_build_flags = -O0 -ggdb3 -g3
```
I get the same linker problem. And I see that the PACKAGES information dumped before the build is identical between 5.3.0 and the 6.0.1 that is automatically chosen when I don't specify a platform version:

### Platform version not specified
```
PLATFORM: Espressif 32 (6.0.1) > MH ET LIVE ESP32DevKIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```

### Platform version set to 5.3.0
```
PLATFORM: Espressif 32 (5.3.0) > MH ET LIVE ESP32DevKIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```
I assume PlatformIO will choose only **supported** framework/toolchains when I don't specify an explicit version.

Jason2866, 2023-02-13:
Yes, @mrengineer7777 was wrong regarding IDF 5.0 use. Some coffee too ;-)
Yes, using a specific platform version installs the recommended tools to the framework.
This can be overriden with `platform_packages`
For this case with
```
platform_packages = toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

TKTF50, 2023-02-13:
@Jason2866 Thanks for restoring my sanity. :)

Regarding the **platform_packages** setting - I tried it and got these messages in the Output window:

```
Resolving esp32 dependencies...
Tool Manager: Installing platformio/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Error: Could not find the package with 'platformio/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'
```
Suggestions?

mrengineer7777, 2023-02-13:
@TKTF50 My apologies on the version confusion.  My toolchain compile shows `PLATFORM: Espressif 32 (2.0.5) > Adafruit ESP32 Feather`.
However I'm using a custom platform by TASMOTA: `platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5/platform-espressif32-2.0.5.zip`. That may be why the version prints differently.

I'll setup a separate project and see if I can replicate this issue.

Jason2866, 2023-02-13:
@TKTF50  Ahh, the toolchains are released from espressif. Try
```
platform_packages = espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```
https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/versions

mrengineer7777, 2023-02-13:
@tempdoom Unable to reproduce.

```
#include <Arduino.h>
#include <LittleFS.h>

void setup() {
  LittleFS.begin();
}

void loop() {

}
```

platformio.ini:
```
[env:s3]
platform = espressif32
board = adafruit_feather_esp32s3
framework = arduino
monitor_speed = 115200
check_skip_packages = yes
monitor_filters = time, colorize, esp32_exception_decoder
build_flags = -DCORE_DEBUG_LEVEL=5
```

Will try thaanstad's code next.

Jason2866, 2023-02-13:
@mrengineer7777 

> @TKTF50 My apologies on the version confusion. My toolchain compile shows `PLATFORM: Espressif 32 (2.0.5) > Adafruit ESP32 Feather`. However I'm using a custom platform by TASMOTA: `platform = [...]

The are slightly different ;-). I build the Tasmota framework with fixed frozen parts. One of the frozen is LittleFS (release version). Espressif Arduino builder uses latest commit at build time for all components. So it can happen that a issue is in my framework build which is not in official. Or the other way around. Some sdkconfig settings are different choosen too. Can result in different behaviour too.  So if issues arrive it may help to try the other.
BTW. latest version is a full platform build and core is 2.0.6 based. Can be used with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.01.01/platform-espressif32.zip
```

TKTF50, 2023-02-13:
@Jason2866 Linker error still occurs. Just to confirm, here is my platformIO.ini and dump from start of compile:

```
[env:esp32]
platform = espressif32 @ 5.3.0
board = mhetesp32devkit
framework = arduino
platform_packages = espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

build_type = debug
debug_build_flags = -O0 -ggdb3 -g3

```
```
PLATFORM: Espressif 32 (5.3.0) > MH ET LIVE ESP32DevKIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```
I also tried building with the V5.2.0 platform and the V5.3.0 toolchain (sort of the reverse of what you asked me to do). It still built successfully.
```
[env:esp32]
platform = espressif32 @ 5.2.0
board = mhetesp32devkit
framework = arduino
platform_packages = espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

build_type = debug
debug_build_flags = -O0 -ggdb3 -g3
```

mrengineer7777, 2023-02-13:
@thaanstad @TKTF50  Unable to reproduce.  Code below compiles successfully.

```
#include <Arduino.h>
#include <FS.h>
#include <FSImpl.h>
#include <vfs_api.h>
#include <SPI.h>
#include <SD.h>
#include <sd_defines.h>
#include <sd_diskio.h>

void setup()
{
    delay(20000);
	Serial.begin(115200);
	SPI.begin(18, 19, 23, 32);
    
	pinMode(4, OUTPUT);  // Set D4 on Arduino to Output (4D Arduino Adaptor V2 - Display Reset)
	digitalWrite(4, 0);  // Reset the Display via D4
	delay(100);
	digitalWrite(4, 1);  // unReset the Display via D4
	delay(3500);
}

// Add the main program code into the continuous loop() function
void loop()
{
    static long pastTime = 0;
    static long interval = 1000;
	static int val = 1;

    if (millis() > pastTime + interval)
    {
        pastTime = millis();
        Serial.print("hello");
        Serial.println(val);
        val++;
    }
}
```

platformio:
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
monitor_speed = 115200
check_skip_packages = yes
monitor_filters = time, colorize, esp32_exception_decoder
build_flags = -DCORE_DEBUG_LEVEL=5

[env:esp32]
platform = espressif32 @ 5.3.0
board = mhetesp32devkit
framework = arduino
platform_packages = espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
build_type = debug
debug_build_flags = -O0 -ggdb3 -g3
```

All environments compile successfully on my machine. VSCode with PIO extension version v2.5.5.

---------------------

> ```
> platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.01.01/platform-espressif32.zip
> ```

@Jason2866 Thanks.  I'm waiting for 2.0.7 to drop.

Jason2866, 2023-02-13:
Good, NOT a (new introduced) toolchain issue!

Jason2866, 2023-02-13:
@mrengineer7777  What OS do you use?  Tried too. Can't reproduce the issue. Tried variants. None did fail.  Compiled on Mac Air M1. Platformio 6.1.6 Latest VSCODE and PIO extension 3.0

mrengineer7777, 2023-02-13:
> @mrengineer7777 What OS do you use? Tried too. Can't reproduce the issue. Tried variants. None did fail. Compiled on Mac Air M1. Platformio 6.1.6 Latest VSCODE and PIO extension 3.0

OS: Windows 11 Pro 22H2.  OS Build 22621.1105
Processor: 12th Gen Intel(R) Core(TM) i7-12700.

mrengineer7777, 2023-02-13:
@TKTF50 Can you publish a demo project reproducing the issue on Github?

thaanstad, 2023-02-13:
@mrengineer7777 thanks for checking on this.  I'll look more into the compiler settings.  Really appreciate your help!  Is it possible for you to include your build output (showing compiler flags) so I can compare them to the ones that I obtained?

mrengineer7777, 2023-02-13:
@thaanstad See attached

[Compile Output.txt](https://github.com/espressif/arduino-esp32/files/10724681/Compile.Output.txt)

TKTF50, 2023-02-13:
@mrengineer7777 Can you tell me what you are looking for? My test project consists of the code that you posted yourself (of tempdoom's test case) and the platformio.ini file I posted earlier. That's it.

Since it seems to be something particular to my environment, I don't know what more I could provide that would help out. Are you looking for something in the .vscode or .platformio directories?

mrengineer7777, 2023-02-13:
@TKTF50 Just trying to eliminate any configuration errors.  Having an example project will allow anyone that uses PIO to test the issue.

I don't expect any issues in the .pio or .vscode folders.  I exclude those folders from repository backups with the following `.gitignore` settings:
```
.pio
.vscode/.browse.c_cpp.db*
.vscode/c_cpp_properties.json
.vscode/launch.json
.vscode/ipch

# Prerequisites
*.d

# Compiled Object files
*.slo
*.lo
*.o
*.obj

# Precompiled Headers
*.gch
*.pch

# Compiled Dynamic libraries
*.so
*.dylib
*.dll

# Fortran module files
*.mod
*.smod

# Compiled Static libraries
*.lai
*.la
*.a
*.lib

# Executables
*.exe
*.out
*.app
```

tktf50, 2023-02-14:
@mrengineer7777 Ok. After much mucking about (I'm still a newbie at this Git stuff) here is a repository with a demo project:
[link_error_issue_7585](https://github.com/tktf50/link_error_issue_7585.git). Hope it helps.

Jason2866, 2023-02-14:
Can reproduce with this settings
```
[env:esp32]
platform = espressif32 @ 6.0.1
platform_packages = framework-arduinoespressif32 @ ~3.20006.0
board = esp-wrover-kit
framework = arduino
build_type = debug
debug_build_flags = -O0 -ggdb3 -g3
```
fails switching back to ~3.20005.0 does work. The trigger is build `type = debug`
core 2.0.6 does fail when building in debug mode.

mrengineer7777, 2023-02-14:
@tktf50 Demo works. Issue duplicated.  Fails at linking.

```
Linking .pio\build\esp32\firmware.elf
c:/users/username/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32\lib060\libFS.a(vfs_api.cpp.o):(.literal._ZN2fs8FileImplD5Ev[_ZN2fs8FileImplD5Ev]+0x0): undefined reference to `vtable for fs::FileImpl'
```

`vtable for fs::FileImpl` is in folder \.pio\build\esp32\firmware.map line 89266:
`vtable for fs::FileImpl                           .pio\build\esp32\lib060\libFS.a(vfs_api.cpp.o)`

The same error occurs with any function call on LittleFS, so I believe the issue is in the constructor.
`LittleFSFS::LittleFSFS() : FS(FSImplPtr(new LittleFSImpl())), partitionLabel_(NULL)`
`FS(FSImplPtr impl) : _impl(impl) { }`
`typedef std::shared_ptr<FSImpl> FSImplPtr;`

My guess is that the pointer `FSImplPtr` is somehow optimized out.

Note that compiling with SPIFFS also fails to compile with the same error.

```
#include <Arduino.h>
//#include <LittleFS.h>
#include <SPIFFS.h>

void setup() {
  //LittleFS.begin();
  SPIFFS.begin();
}
```

`c:/users/username/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32\lib060\libFS.a(vfs_api.cpp.o):(.literal._ZN2fs8FileImplD5Ev[_ZN2fs8FileImplD5Ev]+0x0): undefined reference to `vtable for fs::FileImpl'`

Both LittleFS and SPIFFS compile fine if I remove the line `debug_build_flags = -O0 -ggdb3 -g3` from platformio.ini.

Jason2866, 2023-02-14:
Just found out too, that removing solves. Should not needed regarding https://docs.platformio.org/en/latest/projectconf/build_configurations.html

mrengineer7777, 2023-02-14:
Confirmed demo compiles correctly with option `debug_build_flags = -Og -ggdb3 -g3`.
Resolution: change -O0 -> -Og.

rvbc1, 2023-02-16:
> What compiler options (flags) are set? Probably used one which is not supported. For example you can not use `rtti`. Xtensa toolchains do not support.

Is there any way to enable RTTI? I'm not using the flag -frtti, but the IDF configuration has the setting CONFIG_COMPILER_CXX_RTTI=y. Everything is compiling below Arduino Core 2.0.7. IDF examples also compile with dynamic_cast and CONFIG_COMPILER_CXX_RTTI=y. Is it possible that the Xtensa toolchains don't support this, even though it works in IDF projects? I'm using IDF 5.0 and had to downgrade Arduino to 2.0.6 to compile my huge project, so refactoring to remove all dynamic_cast at this moment is a big problem, especially given that the project has been compiling and working fine with this for the last two years of development on Arduino.

Jason2866, 2023-02-17:
@rvbc1  Disabled already in core 2.0.5 As can be seen here https://github.com/espressif/arduino-esp32/blob/2.0.5/platform.txt#L33

nocs85, 2023-03-20:
Hello, I have a similar Linker error:
```
.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib875\libFS.a(vfs_api.cpp.o):(.rodata._ZTI11VFSFileImpl[_ZTI11VFSFileImpl]+0x8): undefined reference to `typeinfo for fs::FileImpl'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1
```

Using the following test-code:

```
#include <Arduino.h>
#include "FS.h"
#include <LITTLEFS.h>
#include <any>

void setup()
{
    LittleFS.begin();

    delay(20000);
    Serial.begin(115200);
    
    pinMode(4, OUTPUT);  // Set D4 on Arduino to Output (4D Arduino Adaptor V2 - Display Reset)
    digitalWrite(4, 0);  // Reset the Display via D4
    delay(100);
    digitalWrite(4, 1);  // unReset the Display via D4
    delay(3500);
}

// Add the main program code into the continuous loop() function
void loop()
{
    static long pastTime = 0;
    static long interval = 1000;
    static int val = 1;

    if (millis() > pastTime + interval)
    {
        pastTime = millis();
        Serial.print("hello");
        Serial.println(val);
        val++;
    }
}
```

And the following platformio.ini File:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
build_unflags = -std=gnu++11 -fno-rtti
build_flags = -std=gnu++17 -DCORE_DEBUG_LEVEL=0
framework = arduino
monitor_speed = 115200
```

I have to use c++17, so this build_flags are needed.
The Linker-Problem does NOT exist with the following platformio.ini File:

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
build_unflags = -std=gnu++11 -fno-rtti
build_flags = -std=gnu++17 -DCORE_DEBUG_LEVEL=0
framework = arduino
monitor_speed = 115200
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.5
```

Starting with 2.0.6 the Linker error occurs. With 2.0.5 everything links without problems. So I suppose something in 2.0.6 was added that breaks the linking.

Jason2866, 2023-03-20:
@nocs85  Read the previous posts. The reason unbuild `-fno-rtti` is not possible.
Actual toolchains removed the faulty support which should have been never been there as long it is not bug free.

nocs85, 2023-03-20:
I see, sorry for having read over that part.
Are you aware if there is any plan/timeline for a working rtti support?

igrr, 2023-03-20:
@nocs85 could you please try the above suggestion with setting `lib_ldf_mode=deep` in the platformio INI file? 

If that still leads to the same error, could you please post the complete build log, especially the part highlighted in https://github.com/espressif/arduino-esp32/issues/7585#issuecomment-1356331147?

Jason2866, 2023-03-20:
> > Actual toolchains removed the faulty support which should have never been there as long it is not bug free.
> 
> I'm sorry, could you explain this sentence? I am not aware of any issues with RTTI support in our toolchains
> 
> AFAICT, the reported error is simply a library linking order issue.

Great to hear when misunderstanding this commit https://github.com/espressif/gcc/commit/a7d184b4f6504529046efc24e0bd169fdbbf2a58

nocs85, 2023-03-20:
The linker error still exists with a toolchain newer than 2.0.5 with `lib_ldf_mode=deep` active. Here part of the build log:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.7+sha.213f976
 - tool-esptoolpy @ 1.40500.0 (4.5.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
 - toolchain-xtensa32 @ 3.80200.200512 (8.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32Ping @ 1.7.0+sha.357e5f5
|   |-- WiFi @ 2.0.0
|-- ArduinoJson @ 6.21.0
|-- SPI @ 2.0.0
|-- NTPClient @ 3.2.1+sha.1c04e07
|-- Adafruit ADS1X15 @ 2.4.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- DallasTemperature @ 3.11.0
|   |-- OneWire @ 2.3.7
|-- OneWire @ 2.3.7
|-- HX711_ADC @ 1.2.12
|-- PubSubClient @ 2.8.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- WebServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
```

russelljahn, 2023-05-02:
Noting that I encountered the `undefined reference to vtable for fs::FileImpl` linkage error with `framework-arduinoespressif32 @ 2.0.8` when using the `-fno-inline` flag. 

Removing `-fno-inline` fixed it in my case.

lsyin, 2023-06-05:
I used
```
platform = espressif32@5.2.0
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32#2.0.5
```
the bug then disappered

moritz89, 2024-01-16:
The source of the bug is in [libraries/FS/src/FSImpl.h:46-48](https://github.com/espressif/arduino-esp32/blob/04693c666020128f0b8b69d44a8b2144e6e6a20c/libraries/FS/src/FSImpl.h#L46-L47). It did not add `= 0`. This has been resolved in https://github.com/espressif/arduino-esp32/commit/4b3aebe11fa5acc0f8b259105b4d1990b95c399b. So just wait until that lands in the next release. Currently available in 3.0.0-alpha3.