espressif/arduino-esp32 #7587: fail to properly reconnect WiFi from ARDUINO_EVENT_WIFI_STA_DISCONNECTED event handler

FranckFreiburger, 2022-12-13:
### Board

ESP32 dev module

### Device Description

ESP-WROOM-32

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

For some reasons, I need to handle auto reconnect my myself.
I use the ARDUINO_EVENT_WIFI_STA_DISCONNECTED event handler to initiate a reconnection to the WiFi.
Once reconnect is running, I got the following state:

isConnected:0 / localIP:0.0.0.0
isConnected:0 / localIP:0.0.0.0
... 3 seconds later ...
isConnected:0 / localIP:192.168.0.33

(WiFi.isConnected() returns false but WiFi.localIP() is a valid IP)

at  this point, WiFi.isConnected() should ber true because WiFi is connected.

### Sketch

```cpp
#include <WiFi.h>

bool waitForConnected() {
    
    Serial.println("connecting...");

    for (;;) {

        bool isConnected = WiFi.isConnected();
        IPAddress localIP(WiFi.localIP());

        printf("isConnected:%d / localIP:%s\n", isConnected, localIP.toString().c_str());

        if ( isConnected )
            break;

        delay(1000);
    }
    return true;
}

extern "C" void app_main() {

  initArduino();

  Serial.begin(115000);

  WiFi.onEvent([](arduino_event_id_t event) {

    WiFi.begin();
    waitForConnected();

  }, ARDUINO_EVENT_WIFI_STA_DISCONNECTED);

  WiFi.setAutoReconnect(false);

  WiFi.begin(); // connect with previous settings ...
  waitForConnected();
  Serial.println("disconnect");
  WiFi.disconnect();

  for ( ;; )
    delay(1);

}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6656
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (26) boot: ESP-IDF 4.4.2 2nd stage bootloader
I (26) boot: compile time 10:59:08
I (27) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (74) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (81) boot:  2 app0             OTA app          00 10 00010000 00140000
I (89) boot:  3 app1             OTA app          00 11 00150000 00140000
I (96) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (104) boot: End of partition table
I (108) boot_comm: chip revision: 1, min. application chip revision: 0
I (115) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1af4ch (110412) map
I (164) esp_image: segment 1: paddr=0002af74 vaddr=3ffb0000 size=0391ch ( 14620) load
I (170) esp_image: segment 2: paddr=0002e898 vaddr=40080000 size=01780h (  6016) load
I (172) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=82bc0h (535488) map
I (372) esp_image: segment 4: paddr=000b2be8 vaddr=40081780 size=13c80h ( 81024) load
I (405) esp_image: segment 5: paddr=000c6870 vaddr=50000000 size=00010h (    16) load
I (416) boot: Loaded app from partition at offset 0x10000
I (416) boot: Disabling RNG early entropy source...
I (427) cpu_start: Pro cpu up.
I (428) cpu_start: Starting app cpu, entry point is 0x40081dac
I (0) cpu_start: App cpu up.
I (444) cpu_start: Pro cpu start user code
I (444) cpu_start: cpu freq: 160000000
I (444) cpu_start: Application information:
I (449) cpu_start: Project name:     wifi
I (453) cpu_start: App version:      4fad1e7-dirty
I (459) cpu_start: Compile time:     Dec 13 2022 10:55:50
I (465) cpu_start: ELF file SHA256:  8e2afee7ff5fb506...
I (471) cpu_start: ESP-IDF:          4.4.2
I (476) heap_init: Initializing. RAM available for dynamic allocation:
I (483) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (489) heap_init: At 3FFB7D18 len 000282E8 (160 KiB): DRAM
I (495) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (501) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (508) heap_init: At 40095400 len 0000AC00 (43 KiB): IRAM
I (515) spi_flash: detected chip: generic
I (519) spi_flash: flash io: dio
I (524) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (5I (606) wifi:wifi driver task: 3ffc2680, prio:23, stack:6656, core=0
I (621) wifi:wifi firmware version: eeaa27d
I (621) wifi:wifi certification version: v7.0
I (622) wifi:config NVS flash: enabled
I (622) wifi:config nano formating: disabled
I (626) wifi:Init data frame dynamic rx buffer num: 32
I (631) wifi:Init management frame dynamic rx buffer num: 32
I (636) wifi:Init management short buffer num: 32
I (640) wifi:Init dynamic tx buffer num: 32
I (645) wifi:Init static rx buffer size: 1600
I (649) wifi:Init static rx buffer num: 4
I (652) wifi:Init dynamic rx buffer num: 32
I (657) wifi_init: rx ba win: 6
I (660) wifi_init: tcpip mbox: 32
I (664) wifi_init: udp mbox: 6
I (668) wifi_init: tcp mbox: 6
I (671) wifi_init: tcp tx win: 5744
I (676) wifi_init: tcp rx win: 5744
I (680) wifi_init: tcp mss: 1440
I (684) wifi_init: WiFi IRAM OP enabled
I (688) wifi_init: WiFi RX IRAM OP enabled
I (695) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (807) wifi:mode : sta (c8:f0:9e:9b:55:90)
I (808) wifi:enable tsf
I (809) wifi:Set ps type: 1

connecting... isConnected:0 / localIP:0.0.0.0
I (825) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
I (827) wifi:state: init -> auth (b0)
I (832) wifi:state: auth -> assoc (0)
I (837) wifi:state: assoc -> run (10)
I (853) wifi:connected with MYSSID, aid = 6, channel 9, BW20, bssid = 34:27:92:38:57:b3
I (854) wifi:security: WPA2-PSK, phy: bgn, rssi: -73
I (861) wifi:pm start, type: 1

W (870) wifi:<ba-add>idx:0 (ifx:0, 34:27:92:38:57:b3), tid:0, ssn:0, winSize:64
I (901) wifi:AP's beacon interval = 122880 us, DTIM period = 2
isConnected:0 / localIP:0.0.0.0
isConnected:0 / localIP:0.0.0.0
isConnected:0 / localIP:0.0.0.0
I (4104) esp_netif_handlers: sta ip: 192.168.0.33, mask: 255.255.255.0, gw: 192.168.0.254
isConnected:1 / localIP:192.168.0.33
disconnect
I (4818) wifi:state: run -> init (0)
I (4819) wifi:pm stop, total sleep time: 1381694 us / 3958322 us

W (4820) wifi:<ba-del>idx
I (4822) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
connecting... isConnected:0 / localIP:0.0.0.0
I (4842) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
I (4844) wifi:state: init -> auth (b0)
I (4848) wifi:state: auth -> assoc (0)
I (4854) wifi:state: assoc -> run (10)
I (4868) wifi:connected with MYSSID, aid = 6, channel 9, BW20, bssid = 34:27:92:38:57:b3
I (4869) wifi:security: WPA2-PSK, phy: bgn, rssi: -71
I (4876) wifi:pm start, type: 1

W (4884) wifi:<ba-add>idx:0 (ifx:0, 34:27:92:38:57:b3), tid:0, ssn:0, winSize:64
I (4956) wifi:AP's beacon interval = 122880 us, DTIM period = 2
isConnected:0 / localIP:0.0.0.0
isConnected:0 / localIP:0.0.0.0
isConnected:0 / localIP:0.0.0.0
I (8103) esp_netif_handlers: sta ip: 192.168.0.33, mask: 255.255.255.0, gw: 192.168.0.254
isConnected:0 / localIP:192.168.0.33
isConnected:0 / localIP:192.168.0.33
isConnected:0 / localIP:192.168.0.33
...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-12-13:
Since you are using PlatformIO with the Arduino core, you should be able to modify the Arduino libraries (make a backup copy of the folder) to get some additional insight into what is going on.  Be aware that if you change Arduino version in PIO, this folder will be overwritten.  Files are located here: C:\Users\YOUR_USER_NAME\.platformio\packages\framework-arduinoespressif32\

`WiFi.isConnected()` is actually WiFiSTAClass::isConnected(), which calls `WiFiSTAClass::status()`.  This gets flags on `_sta_status_group`.

The flags for connected event are updated in `WiFiGenericClass::_eventCallback(arduino_event_t *event)`.  I would add debug statements here or in WiFiSTAClass::_setStatus(wl_status_t status);

My best guess is you are disconnecting too quickly after the connection finishes (add a delay?), and the flags get out of sync.  It will be interesting to see the state of the flag bits.

mrengineer7777, 2022-12-13:
`_arduino_event_task` is what allows the Arduino WiFi module to run.  You want that.

`_eventCallback` is what calls your disconnect handler registered with `WiFi.onEvent`.

What makes you think eventGroup bits aren't set properly?

closed 2023-04-13 by mrengineer7777