espressif/arduino-esp32 #7588: HTTPS_OTA_Update and ESPAsyncWebServer

bearkillerPT, 2022-12-13:
### Board

ESP32 Dev Module

### Device Description

plain model with wifi

### Hardware Configuration

standard

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi! I currently have a web server running with some data of some sensors. 
I'm sending said data to my server and on the response body I can indicate which is the last available firmware version.
I've tested [https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/HTTPS_OTA_Update/HTTPS_OTA_Update.ino](this) example and got it working fine with my certificates and credentials. 
But now I'm trying to get both working and can't because it doesn't compile...
Is it not possible to have both? 
I found no other alternative to download the firmware from my server...
Thank you!

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include "HttpsOTAUpdate.h"
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "SPIFFS.h"

#define statusTaskDefaultPeriod 30000
#define WifiScanTaskDefaultPeriod 30000

#define FW_VERSION 1
static HttpsOTAStatus_t otastatus;

String serverName = "https://example.com";

static const char *oldServerCertificate = "xxxx";

IPAddress PageIP(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
[... 743 lines cut ...]
   HTTP_OPTIONS = 0b01000000,
                  ^~~~~~~~~~
In file included from C:\Users\gil-t\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/sdk/esp32/include/esp_http_client/include/esp_http_client.h:11,
                 from C:\Users\gil-t\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\Update\src/HttpsOTAUpdate.h:3,
                 from C:\Users\gil-t\Desktop\dev\HeabooComercial\esp32_code\esp32_code.ino:4:
C:\Users\gil-t\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:31: note: previous declaration 'http_method HTTP_OPTIONS'
 #define XX(num, name, string) HTTP_##name = num,
                               ^~~~~
C:\Users\gil-t\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:102:3: note: in expansion of macro 'XX'
   XX(6,  OPTIONS,     OPTIONS)      \
   ^~
C:\Users\gil-t\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
```

### Other Steps to Reproduce

A combination of trying to use the HTTPS_OTA_Update and ESPAsyncWebServer in the same sketch,
My idea is to have a local server that shows some sensor data and then there's a loop that sends said data to my server.
In the response body it will be included the last available version served on my server. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-12-13:
A better title for your issue is:
Conflicting defines for HTTP Methods http_parser.h, ESPAsyncWebServer.h

Description:
Conflicting defines for HTTP Methods between http_parser.h and ESPAsyncWebServer.h.

     \esp32\2.0.5\libraries\Update\src/HttpsOTAUpdate.h:3
   ->\esp32\2.0.5/tools/sdk/esp32/include/esp_http_client/include/esp_http_client.h
  -->\esp32\2.0.5/tools/sdk/esp32/include/nghttp/port/include/http_parser.h
        HTTP_METHOD_MAP
           HTTP_GET, etc.

http_parser.h:
```
/* Request Methods */
#define HTTP_METHOD_MAP(XX)         \
  XX(0,  DELETE,      DELETE)       \
  XX(1,  GET,         GET)          \
  XX(2,  HEAD,        HEAD)         \
  XX(3,  POST,        POST)         \
  XX(4,  PUT,         PUT)          \
  /* pathological */                \
  XX(5,  CONNECT,     CONNECT)      \
  XX(6,  OPTIONS,     OPTIONS)      \
  XX(7,  TRACE,       TRACE)        \
  /* WebDAV */                      \
  XX(8,  COPY,        COPY)         \
  XX(9,  LOCK,        LOCK)         \
  XX(10, MKCOL,       MKCOL)        \
  XX(11, MOVE,        MOVE)         \
  XX(12, PROPFIND,    PROPFIND)     \
  XX(13, PROPPATCH,   PROPPATCH)    \
  XX(14, SEARCH,      SEARCH)       \
  XX(15, UNLOCK,      UNLOCK)       \
  XX(16, BIND,        BIND)         \
  XX(17, REBIND,      REBIND)       \
  XX(18, UNBIND,      UNBIND)       \
  XX(19, ACL,         ACL)          \
  /* subversion */                  \
  XX(20, REPORT,      REPORT)       \
  XX(21, MKACTIVITY,  MKACTIVITY)   \
  XX(22, CHECKOUT,    CHECKOUT)     \
  XX(23, MERGE,       MERGE)        \
  /* upnp */                        \
  XX(24, MSEARCH,     M-SEARCH)     \
  XX(25, NOTIFY,      NOTIFY)       \
  XX(26, SUBSCRIBE,   SUBSCRIBE)    \
  XX(27, UNSUBSCRIBE, UNSUBSCRIBE)  \
  /* RFC-5789 */                    \
  XX(28, PATCH,       PATCH)        \
  XX(29, PURGE,       PURGE)        \
  /* CalDAV */                      \
  XX(30, MKCALENDAR,  MKCALENDAR)   \
  /* RFC-2068, section 19.6.1.2 */  \
  XX(31, LINK,        LINK)         \
  XX(32, UNLINK,      UNLINK)       \

enum http_method
  {
#define XX(num, name, string) HTTP_##name = num,
  HTTP_METHOD_MAP(XX)
#undef XX
  };
```

ESPAsyncWebServer.h:
```
#ifndef WEBSERVER_H
typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
  HTTP_PATCH   = 0b00010000,
  HTTP_HEAD    = 0b00100000,
  HTTP_OPTIONS = 0b01000000,
  HTTP_ANY     = 0b01111111,
} WebRequestMethod;
#endif
```

Unfortunately ESPAsyncWebServer isn't part of this repository, and it doesn't look like it will be updated in the near future.

We are successfully using both ESPAsyncWebServer.h and Update.h (HTTP not HTTPS), but they are in different modules and compiled using VSCode + PlatformIO.

mrengineer7777, 2022-12-13:
Sigh... After looking at those files now my project doesn't compile either.

mrengineer7777, 2022-12-13:
Ok I was able to get my project to compile by removing the following includes from my command parser module:

```
#include <AsyncJson.h>
#include <ESPAsyncWebServer.h>
```

Instead I use `#include <ArduinoJson.h>` there.  I only call `ESPAsyncWebServer.h` from the module that serves webpages.  Caution: `AsyncJson.h` includes `ESPAsyncWebServer.h`.

bearkillerPT, 2022-12-13:
I can't seem to find those includes on my http_parser.h:

```c
/* Copyright Joyent, Inc. and other Node contributors. All rights reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to
 * deal in the Software without restriction, including without limitation the
 * rights to use, copy, modify, merge, publish, distribute, sublicense, and/or
 * sell copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 * FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS
 * IN THE SOFTWARE.
 */
#ifndef http_parser_h
#define http_parser_h
#ifdef __cplusplus
extern "C" {
#endif

/* Also update SONAME in the Makefile whenever you change these. */
#define HTTP_PARSER_VERSION_MAJOR 2
#define HTTP_PARSER_VERSION_MINOR 7
#define HTTP_PARSER_VERSION_PATCH 0

#include <sys/types.h>
#if defined(_WIN32) && !defined(__MINGW32__) && \
  (!defined(_MSC_VER) || _MSC_VER<1600) && !defined(__WINE__)
#include <BaseTsd.h>
#include <stddef.h>
typedef __int8 int8_t;
typedef unsigned __int8 uint8_t;
typedef __int16 int16_t;
typedef unsigned __int16 uint16_t;
typedef __int32 int32_t;
typedef unsigned __int32 uint32_t;
typedef __int64 int64_t;
typedef unsigned __int64 uint64_t;
#else
#include <stdint.h>
#endif

/* Compile with -DHTTP_PARSER_STRICT=0 to make less checks, but run
 * faster
 */
#ifndef HTTP_PARSER_STRICT
# define HTTP_PARSER_STRICT 1
#endif

/* Maximium header size allowed. If the macro is not defined
 * before including this header then the default is used. To
 * change the maximum header size, define the macro in the build
 * environment (e.g. -DHTTP_MAX_HEADER_SIZE=<value>). To remove
 * the effective limit on the size of the header, define the macro
 * to a very large number (e.g. -DHTTP_MAX_HEADER_SIZE=0x7fffffff)
 */
#ifndef HTTP_MAX_HEADER_SIZE
# define HTTP_MAX_HEADER_SIZE (80*1024)
#endif

typedef struct http_parser http_parser;
[... 256 lines cut ...]
const char *http_errno_name(enum http_errno err);

/* Return a string description of the given error */
const char *http_errno_description(enum http_errno err);

/* Initialize all http_parser_url members to 0 */
void http_parser_url_init(struct http_parser_url *u);

/* Parse a URL; return nonzero on failure */
int http_parser_parse_url(const char *buf, size_t buflen,
                          int is_connect,
                          struct http_parser_url *u);

/* Pause or un-pause the parser; a nonzero value pauses */
void http_parser_pause(http_parser *parser, int paused);

/* Checks if this is the final chunk of the body. */
int http_body_is_final(const http_parser *parser);

#ifdef __cplusplus
}
#endif
#endif
```
I'm somewhat new to the arduino environment :)

mrengineer7777, 2022-12-13:
You won't find it by searching.  It's a macro "HTTP_METHOD_MAP" that creates the definitions.

bearkillerPT, 2022-12-13:
So, to confirm, the only way would be to switch to VSCode + PlatformIO? My boss doens't want me to use this option.... I personally love vscode!
If i use the switch to the normal webServer will it work?

bearkillerPT, 2022-12-14:
I've already given a shot at that and I managed to compile the sketch but unfortunately it crashed when updating... The asying web server was working properly but then i got a Guru panick....

Yeh I wish I could give at a try but I'm working with a guy that really doesn't want to go through the hassle  😂.
I'll probably just rewrite everything with the standard Webserver in Arduino. 

Was making good use of the processor function that allowed to substitute vars in the html for values like a StringBuilder. Should i build the html in the on() function with the variable manually or is there a better way? I'm really restricted on space aswell.... Can't really afford to add more libs...

Thank you for your help!