espressif/arduino-esp32 #7596: Number of SPI channels on ESP32C3 is inconsistent

ZinggJM, 2022-12-17:
### Board

ESP32C3

### Device Description

any ESP32C3 board

### Hardware Configuration

any

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

any

### PSRAM enabled

no

### Upload speed

irrelevant

### Description

The documentation https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf claims there are 3 SPI channels.
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c supports only one:
```
#elif CONFIG_IDF_TARGET_ESP32C3
// ESP32C3
#define SPI_COUNT           (1)
```
and would fail for HSPI
```
spi_t * spiStartBus(uint8_t spi_num, uint32_t clockDiv, uint8_t dataMode, uint8_t bitOrder)
{
    if(spi_num >= SPI_COUNT){
        return NULL;
    }
```

### Sketch

```cpp
irrelevant
```

### Debug Message

```plain
irrelevant
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-12-17:
Two of the SPI channels are for memory.  There is only one general purpose SPI.  Check the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf)

ZinggJM, 2022-12-18:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.h
```
#if CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32S3
#define FSPI  0
#define HSPI  1
#else
#define FSPI  1 //SPI bus attached to the flash (can use the same data lines but different SS)
#define HSPI  2 //SPI bus normally mapped to pins 12 - 15, but can be matrixed to any pins
#if CONFIG_IDF_TARGET_ESP32
#define VSPI  3 //SPI bus normally attached to pins 5, 18, 19 and 23, but can be matrixed to any pins
#endif
#endif
```
Either makes no sense, or is misleading.

ZinggJM, 2022-12-18:
BTW: I was surprised to see that ESP32 has 4 SPI channels:
```
#else
// ESP32
#define SPI_COUNT           (4)
```

ZinggJM, 2022-12-19:
In [https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf](url) Table 10, Page 31:
SPI0/1 : Support Standard SPI, Dual SPI, Quad SPI, and QPI that allow connection to external flash

**Support Standard SPI** : could be used for standard SPI (maybe with some limitations, e.g. CPOL, CPHA)

ZinggJM, 2022-12-19:
I did not read attentively enough this bottom remark on Table 2:
3 Ports of embedded flash correspond to pins of ESP32-C3FN4 and ESP32-C3FH4 as follows:
• CS# = SPICS0
• IO0/DI = SPID
• IO1/DO = SPIQ
• CLK = SPICLK
• IO2/WP# = SPIWP
• IO3/HOLD# = SPIHD
These pins are not recommended for other uses.
This may explain why SPI0/SPI1 are not supported for general use.
Maybe HSPI should be removed for C3.

davidbcf, 2023-07-05:
i got the same problem with esp32-c3, so in conclusion,esp32-c3 dont support general SPI  using?

me-no-dev, 2023-07-05:
ESP32-C3 has only one available GPSPI bus. We will fix the definitions in `esp32-hal-spi`. The other two SPIs are for Flash/PSRAM only

davidbcf, 2023-07-05:
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.h
>
> ```
> #if CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32S3
> #define FSPI  0
> #define HSPI  1
> #else
> #define FSPI [...]

maybe for esp32-c3 need add one  more spi enum definition and change SPI_COUNT to 2, will it be work?

ushalev001, 2023-10-27:
The issue I see is here:
spi_t * spiStartBus(uint8_t spi_num, uint32_t clockDiv, uint8_t dataMode, uint8_t bitOrder)
{
    if(spi_num >= SPI_COUNT){
        return NULL;
    }
HSPI is defined as 1, and SPI_COUNT for ESP32C3 is also 1. this means the bus will never start unless you initialize it with '0' but then will the bus actually work? I tried it and it didn't work. Is there a simple example that was tested on ESP32c3?

morcibacsi, 2023-11-15:
A related issue is the following case: 
Board used ESP32-C3 SuperMini: https://www.aliexpress.com/i/1005005319963906.html
In Arduino the board is selected as LOLIN C3 Mini

Newing up an SPI instance uses the HSPI as a default, so it won't work. Passing the parameter as FSPI, the code start working.

https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h#L64

```
SPIClass* spi;
spi = new SPIClass();      // not working
spi = new SPIClass(FSPI);  // working
```

Example sketch which uses MCP2515 CAN library from CoryJFowler: https://github.com/coryjfowler/MCP_CAN_lib/

```
#include <Arduino.h>
#include <SPI.h>
#include <mcp_can.h>

SPIClass* spi;
MCP_CAN *can;

bool ledStatus = false;

uint8_t CS_PIN   = 5;
uint8_t MOSI_PIN = 7;
uint8_t SCK_PIN  = 6;
uint8_t MISO_PIN = 2;

uint8_t STATUS_LED_PIN = 8;

uint8_t data_036[] = { 0x0E, 0x00, 0x00, 0x21, 0x21, 0x80, 0x00, 0xA0 };
uint8_t data_0f6[] = { 0x8E, 0x71, 0x21, 0x8D, 0xF3, 0x65, 0x65, 0x50 };

void setup() {
  // put your setup code here, to run once:

    delay(5000);
    spi = new SPIClass(); // <-------------  no parameter passed to the constructor, and the default constructor uses HSPI which is not available
    spi->begin(SCK_PIN, MISO_PIN, MOSI_PIN, CS_PIN);

    can = new MCP_CAN(spi, CS_PIN);
    uint8_t err = can->begin(MCP_ANY, CAN_125KBPS, MCP_8MHZ);
    can->setMode(MCP_NORMAL);

    pinMode(STATUS_LED_PIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:

  digitalWrite(STATUS_LED_PIN, ledStatus);
  ledStatus = !ledStatus;

  can->sendMsgBuf(0x036, 0, 8, data_036);
  can->sendMsgBuf(0x0F6, 0, 8, data_0f6);

  delay(50);
}

```

twaymouth, 2024-01-27:
@morcibacsi Thanks, your insight solved my issue with SPI not functioning on the ESP32C3 using the Arduino core.

 I was able to get it working using esp-idf framework but not the Arduino core however passing the parameter FSPI to the SPI class constructor resolved the issue.

VojtechBartoska, 2024-02-20:
Closing this in favor of https://github.com/espressif/arduino-esp32/issues/8192, we will track this there.

closed 2024-02-20 by VojtechBartoska