espressif/arduino-esp32 #7602: Ethernet + WiFi: Shared DNS.

gonzabrusco, 2022-12-19:
### Board

Custom ESP32 Board with a LAN8720

### Device Description

Custom ESP32 Board with a LAN8720

### Hardware Configuration

Custom ESP32 Board with a LAN8720

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello. I'm doing some tests with a custom board with the LAN8720 onboard. 

I noticed that ETH.dnsIP(uint8_t dns_no) and WiFi.dnsIP(uint8_t dns_no) always return the same IPs. I think there's a problem in the way the DNS are being handled. **Is it normal for both interfaces to share the same DNS?**

The DNS set are always the ones of the latest interface that got IP (by DHCP). For example take a look at the following screenshot:
[image]
As you can see, the IP of the Ethernet and the WiFi interface are different, but the DNS are the same for some reason. In that image, the Ethernet connected first and then the WiFi, overriding the DNS that got Ethernet first (which does not make sense because those IPs are not even in the same subnet of the Ethernet IP).

Also I noticed another problem while using both interfaces at the same time.  If the network cable is plugged before powering everything up, I can't never connect to my server using the following code:

```
ETH.begin();
WiFi.begin("Unexistent SSID", "Random Password");
```
The error the debug gives me is:
`E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for test.mosquitto.org`

The only way for this to start working, if by unplugging and plugging the Ethernet cable. Just then the DNS are set correctly.

But if I modify the code and add some lines, Ethernet works from the start (without needing to unplug and plug the Ethernet cable):

```
ETH.begin();
ETH.config(IPAddress(0u),IPAddress(0u),IPAddress(0u),IPAddress(0u),IPAddress(0u));
WiFi.config(IPAddress(0u),IPAddress(0u),IPAddress(0u),IPAddress(0u),IPAddress(0u));
WiFi.begin("Unexistent SSID", "Random Password");
```

This clearly seems like a bug or a bad managment of DNS records. If DNS are shared, then it this means that Ethernet and WiFi can't be used together on the first place. And even if that was not possible, when running WiFi.begin() without a previous config, seems to reset or delete the DNS of the Ethernet interface.

Finally as a **bonus bug** WiFiClientSecure localIP() method always returns **0.0.255.0** but if I modify the script to connect without TLS, the localIP() method works fine on a regular WiFiClient.

[... 291 lines cut ...]
                    Serial.println("Failed to connect!");
                }
            }
        }
        
    }
}
```

### Debug Message

```plain
WiFi IP: 0.0.0.0
WiFi DNS: 0.0.0.0 0.0.0.0
Ethernet IP: 192.168.10.74
Ethernet DNS: 0.0.0.0 0.0.0.0
I'm not connected!
[403506][E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for test.mosquitto.org
[403506][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1
Failed to connect!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-12-19:
Ethernet would need to be updated [here](https://github.com/espressif/arduino-esp32/blob/9275dbff942bb2ccce62632607500ad27dd0513a/libraries/Ethernet/src/ETH.cpp#L446-L456) and [here](https://github.com/espressif/arduino-esp32/blob/9275dbff942bb2ccce62632607500ad27dd0513a/libraries/Ethernet/src/ETH.cpp#L488-L492) to use the netif API, like it's done for WiFi. That should fix the problem 🤞

gonzabrusco, 2022-12-20:
@me-no-dev not sure if it's "just that". Because as you can see in my tests, Ethernet does not work if you power on everything with the Ethernet cable and just calling ETH.begin() and WiFi.begin() with an unexistent SSID. 

As you can see, that particular bug occurs without calling any config(). So I think the bug it's a bit more complex. It's not just the setters and getters of the DNS. It's the way Ethernet and Wifi initializes. 

If I just call ETH.begin() without calling WiFi.begin() it works right away (with the Ethernet cable connected before powering up). But if you call WiFi.begin() with an unexistent SSID right after ETH.begin() the DNS resolution starts to fail. 

So I think WiFi.begin() is also messing with the Ethernet DNS.

TD-er, 2022-12-23:
When closing WiFi, you also erase the DNS entries.

What I do in ESPEasy as a work-around is that I keep a copy of the received DNS records from the DHCP.
Then I restore the DNS settings whenever a DNS lookup failed.

Typical use case:

- ESP is booted with Ethernet cable not connected.
- WiFi is started to find an AP to connect to.
- Eth gets connected + gets IP from DHCP server
- WiFi is turned off -> clears DNS entries.

TD-er, 2023-01-07:
I have done some more research on this topic.

For sure the DNS server IPs get reset (or set to some random value) when the WiFi interface is turned off.
However, it used to be that on the next DHCP lease renew, those would be restored.

On the latest SDK revisions, things have changed for the worse.
Where these DNS IP addresses used to be cleared to 0.0.0.0, on my nodes they now get some seemingly random value of 253.0.0.0
No idea if that's part of another field in the DHCP reply, or perhaps some uninitialized value somewhere.

Another change is that this mangling of the DNS server IPs is done on every DHCP lease renewal.
Thus in my network, which has the DHCP lease time set to 10 minutes, this already occurs after 5-ish minutes.

So not only is it a really poor design decision to only have a single set of DNS IPs instead of per interface, but also clearing it when shutting down one interface will mess up the DNS config of another.
I think a better solution would be to clear (or set with static IP config) the DNS server IPs when starting a new connection, not when closing the interface.

Also the DHCP lease renewal apparently differs from the initial DHCP request.
Or at least handling of the result differs.

The work-around I just implemented for ESPEasy is to cache the DNS server IPs per interface.
Every time I intend to do a DNS lookup, or make a connection, I check to see if the DNS server IPs should be restored from the cache.
When initiating a new connection on that interface, I clear the cached values.

If/When someone is going to look into this, I would also make a feature request to get access to some extra DHCP optional fields, like syslog server IP and NTP server IP.

s-hadinger, 2023-01-07:
The limitation comes from LWIP which uses the same pair of DNS servers for all interfaces, whether IPv4 or IPv6.

In Tasmota, I had to make sure that we keep a copy of 2x DNS IPv4 and 2x DNS IPv6 servers, and force them back in place whenever LWIP clears them.

https://github.com/arendst/Tasmota/blob/7465ff0c7f3341555d11c219e0a6b5f60c7ff6e8/lib/libesp32/ESP32-to-ESP8266-compat/src/ESP32Wifi.cpp#L64-L129

TD-er, 2023-01-07:
Ah, that's even nastier. Did not realize you only had the same 2 for both IPv4 and IPv6.
Can you "combine" them? Like 1st one for IPv6 and 2nd one for IPv4 if you prefer IPv6 lookup first?

s-hadinger, 2023-01-07:
No you can't. The resolver always prefers the first one and moved to second if the first times out

Jason2866, 2023-01-09:
@VojtechBartoska @me-no-dev Can this bug added too milestone 2.0.7?
There is alot of workaround code needed to handle this bug.

me-no-dev, 2023-01-09:
@Jason2866 it will not make it to 2.0.7 unless someone submits a PR with fixes. We already have quite a bit to look into and 2.0.7 needs to come out fast

mrengineer7777, 2023-12-06:
Possibly related #8672

VojtechBartoska, 2023-12-11:
Hi @gonzabrusco is this still valid in version 2.0.14? Can you please test if you still face the issue on latest release of 2.X? Thanks

VojtechBartoska, 2024-02-20:
This will be probably covered by https://github.com/espressif/arduino-esp32/pull/8760

gonzabrusco, 2024-02-20:
Sorry. I can't test it right now. I will let you know later or when 3.0.0 is released. @VojtechBartoska

VojtechBartoska, 2024-04-12:
Hello, can you please validate this against 3.0.0-RC1 version? Thanks

VojtechBartoska, 2024-05-14:
This is supposed to be solved, I am closing this ticket. If needed, you can reopen. Thanks.

closed 2024-05-14 by VojtechBartoska