espressif/arduino-esp32 #7604: ESP32 C3-WROOM-02-H4 antenna issue ??

spilz87, 2022-12-20:
### Board

C3-WROOM-02-H4

### Device Description

remplacing an ESP32-C3-13 (4MB) (which is out of stock) by an ESP32 C3-WROOM-02-H4 

### Hardware Configuration

ESP32-C3-13 (4MB) VS C3-WROOM-02-H4 

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

HI everybody !

I try do replace on a board an ESP32-C3-13 (4MB) (which is out of stock) by an ESP32 C3-WROOM-02-H4.

I change it on the board.

I load the example WiFiScan.ino (which worked before) using board "ESP32C3 Dev Module".
The loading was succesfull.

but the board restart when it try to use the WiFi.
If I touch the antenna, it works well.

I change again to ESP32-C3-13, and it works well without touching the antenna.
I tryed on 3 differents C3_WROOM-02-H4 chip to be sure

I tryed to switch chips 3 times and always the same result : 
- with ESP32-C3-13 : everything work well
- with C3_WROOM-02-H4 : works only if I touch the antenna

questions : 
- should I use an other board instead of "ESP32C3 Dev Module" for C3_WROOM-02-H4 ?
- someone else observed this issue ?
- is there a known issue about antenna design on C3_WROOM-02-H4 ?

thanks in advance

Spilz

### Sketch

```cpp
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```

### Debug Message

```plain
no debug
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

spilz87, 2023-01-01:
in an other project they find an issue similar to the one I have.
[https://github.com/zephyrproject-rtos/zephyr/issues/43655](url)

Does someone have an idea about where it can come from ?

thanks in advence

spilz87, 2023-02-13:
I update the link (it didn't work)

nobody has an idea about it ??

rftafas, 2025-01-10:
@spilz87 Your report on the Zephyr project (fingering the board makes it work) points to a bad soldering / hardware issue and unrelated to the actual issue reported there.

If you are not experiencing issues on other boards / devices, seems the issue is specific to this unit.

Unless there is an update pointing towards a bug (and not a hardware deffect), re-open this.

closed 2025-01-10 by rftafas