espressif/arduino-esp32 #7605: ESP-WROOM-32 WiFi.begin() issue AUTH_EXPIRE

dvarrel, 2022-12-20:
### Board

ESP32 dev module

### Device Description

Wemos D1 R32 ( ESP-WROOM-32  )
and small ESP-WROOM-32 module

### Hardware Configuration

card without add-on

### Version

v1.0.6

### IDE Name

Arduino IDE 1.8.19 and also 2.03

### Operating System

ubuntu 22.04

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

with Wemos D1 R32 ( ESP-WROOM-32 too ! ) no AP connexion

- code works with small ESP-WROOM-32 module
-  Wifi scan ok 
-  test with phone AP, Orange AP ( Open, WPA, and WPA2 )

### Sketch

```cpp
#ifdef ESP32
#include <WiFi.h>
#endif
#ifdef ESP8266
#include <ESP8266WiFi.h>
#endif

void setup()
{  
  Serial.begin(115200);
  delay(100);
  WiFi.begin("Livebox-C806","*******");

  // attente connexion
  Serial.print("\nConnecting HotSpot");
  while(WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.print("\nWifi OK, IP: ");
  Serial.print(WiFi.localIP());
  Serial.print(" Passerelle: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("Try to ping me !");
}

void loop() {}
```

### Debug Message

```plain
`scan start
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
scan done
1 networks found
1: Livebox-C806 (-94)*`

`st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

Connecting HotSpot......[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 2 - AUTH_EXPIRE
`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dvarrel, 2022-12-20:
tested also with v2.0.5 :

` 41824][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 41832][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 41839][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
.[ 41847][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
..[ 42951][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Livebox-C806, BSSID: 7c:c1:77:be:70:c0, Reason: 2
[ 42951][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED`

lbernstone, 2022-12-20:
Please try (in v2.0.5) enabling "erase all flash before upload" in the Tools menu, and then upload the firmware again.  It looks like you have cached expired credentials.

dvarrel, 2022-12-20:
thank you.
done, but not efficient.

dvarrel, 2022-12-20:
`esptool.py v4.2.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:d7:eb:0f:34:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 14.3s
Compressed 18912 bytes to 13025...
Writing at 0x00001000... (100 %)
Wrote 18912 bytes (13025 compressed) at 0x00001000 in 0.4 seconds (effective 368.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 555.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 766.9 kbit/s)...
Hash of data verified.
Compressed 700096 bytes to 452940...
Writing at 0x00010000... (3 %)
Writing at 0x0001c558... (7 %)
Writing at 0x000277d3... (10 %)
Writing at 0x000308f8... (14 %)
Writing at 0x0003631e... (17 %)
Writing at 0x0003ba3d... (21 %)
Writing at 0x00040f99... (25 %)
Writing at 0x00046168... (28 %)
Writing at 0x0004b32b... (32 %)
Writing at 0x000505b6... (35 %)
Writing at 0x00055999... (39 %)
Writing at 0x0005aaa6... (42 %)
Writing at 0x0005fc72... (46 %)
Writing at 0x0006586b... (50 %)
Writing at 0x0006b037... (53 %)
Writing at 0x000702f5... (57 %)
Writing at 0x00075668... (60 %)
Writing at 0x0007a90e... (64 %)
Writing at 0x0007ff90... (67 %)
Writing at 0x0008572d... (71 %)
Writing at 0x0008b110... (75 %)
Writing at 0x00090d2c... (78 %)
Writing at 0x00096520... (82 %)
Writing at 0x0009f4ca... (85 %)
Writing at 0x000a6c87... (89 %)
Writing at 0x000ac306... (92 %)
Writing at 0x000b1d6a... (96 %)
Writing at 0x000b735d... (100 %)
Wrote 700096 bytes (452940 compressed) at 0x00010000 in 7.6 seconds (effective 733.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

mrengineer7777, 2022-12-21:
-94 is a terrible signal strength.  Did you forget to connect an antenna to the module (only applies to u.FL version)?

dvarrel, 2022-12-23:
this is the module :
[image]

with a ap rssi -22dB, it did not connect anymore

Jeff-Jordan, 2023-01-01:
it's an ESP32 D0WD-V3 problem (maybe RTOS related).
[Look here](https://github.com/espressif/arduino-esp32/issues/7562), there the issue is reported as well.... and I was able to track down the problem to the encryption password.
It seems to happen, when the password string only contains numbers.

dvarrel, 2023-01-01:
i try with open-AP , also changing the key with alphanumeric, just alpha, just numeric...
also try with wifiMulti example

:-(

mrengineer7777, 2023-01-03:
As noted in #7562, the Wemos modules have a hardware issue that can be fixed by modifying the board.  If that resolves your problem, then please close this issue.

closed 2025-01-16 by Parsaabasi