espressif/arduino-esp32 #7607: ESP32-C3 (LILYGO T-01C3) UART trouble

evlo, 2022-12-20:
### Board

LILYGO T-01C3

### Device Description

esp01 form factor replacement with esp32-c3

### Hardware Configuration

nothing connected, but using esphome

### Version

v2.0.5

### IDE Name

ESPhome (not an ide)

### Operating System

win11, mint21.1

### Flash frequency

40MHz i guess

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm trying to use esphome, i dunno even how to use arduino framework standalone and I'm having issue when i put in uart component that device does not boot https://github.com/esphome/issues/issues/3924

maybe related https://github.com/espressif/arduino-esp32/pull/7496?

### Sketch

```cpp
it is just esphome stuff
```

### Debug Message

```plain
WARNING Decoded 0x4202d1ac: uart_ll_get_txfifo_len at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32c3/include/hal/uart_ll.h:329 (discriminator 1)
 (inlined by) panic_print_char at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:79 (discriminator 1)
```

### Other Steps to Reproduce

in september I was able to build using
```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

but now i'm unable to make workin build with uart component using arduino framework in esphome, even with older esphome, again esphome stuff, i do not really understand the framework relation, but it seems related

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-12-21:
You need to provide a example Arduino sketch which fails.
The issue is highly caused from the esphome build system.

SuGlider, 2022-12-28:
It is interesting once `uart_ll_get_txfifo_len()` is an inline function that returns just a number from a simple addition.
It seems to be some sort of linking issue. 
Maybe cleaning the whole building folder and building it over again could help?!

VojtechBartoska, 2023-02-16:
Hello, closing this issues as expired due to no answer. If needed, you can reopen this.

closed 2023-02-16 by VojtechBartoska