espressif/arduino-esp32 #7614: Automatically close File once it is out-of-scope?

Zhu-jiatong, 2022-12-22:
### Related area

File system

### Hardware specification

ESP32

### Is your feature request related to a problem?

Please do correct me if I am wrong, but I noticed that one has to manually call `close()` on every `fs::File` instance created. I wonder if it is possible to make `fs::File` instances automatically call `close()` or do something equivalent upon going out-of-scope.
This would reduce redundancy in code, and this does seem like a reasonable feature to be added.

### Describe the solution you'd like

It appears that the `fs::File` class does not currently have a destructor. I would suggest adding a destructor like this:
`~File() { close(); }`

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

igrr, 2022-12-22:
That should already be the case: File keeps a reference counted pointer (shared_ptr) to FileImpl interface class, and FileImpl implementation closes the file in the destructor: https://github.com/espressif/arduino-esp32/blob/bb8c8559b62a0d2e888d51bf12bd50cc070f0b4b/libraries/FS/src/vfs_api.cpp#L343

Do you observe some behavior which indicates that the file isn't actually being closed?

Zhu-jiatong, 2022-12-22:
I see, then it must be some other issues in my code. I tested this using a simple recursively print directory programme and things appeared fine. Thank you.

Best regards,

Zhu

From: Ivan Grokhotkov
Sent: Thursday, December 22, 2022 2:46 PM
To: espressif/arduino-esp32
Cc: Zhu-jiatong; Author
Subject: Re: [espressif/arduino-esp32] Automatically close File once it isout-of-scope? (Issue #7614)

That should already be the case: File keeps a reference counted pointer (shared_ptr) to FileImpl interface class, and FileImpl implementation closes the file in the destructor: https://github.com/espressif/arduino-esp32/blob/bb8c8559b62a0d2e888d51bf12bd50cc070f0b4b/libraries/FS/src/vfs_api.cpp#L343
Do you observe some behavior which indicates that the file isn't actually being closed?

everslick, 2022-12-22:
@Zhu-jiatong pls, close this issue then.

closed 2022-12-22 by Zhu-jiatong