espressif/arduino-esp32 #7615: exists() is giving an error

Adrianotiger, 2022-12-22:
### Board

ESP32-S3

### Device Description

ESP32-S3 16MB Dev-Kit

### Hardware Configuration

Nothing

### Version

v2.0.5

### IDE Name

Arduino IDE 2.0.4

### Operating System

Windows 10

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am using the LittleFS library.
When I try to check if a file exists, this function is executed:
https://github.com/espressif/arduino-esp32/blob/bb8c8559b62a0d2e888d51bf12bd50cc070f0b4b/libraries/LittleFS/src/LittleFS.cpp#L46
```
bool LittleFSImpl::exists(const char* path)
{
    File f = open(path, "r",false);
    return (f == true);
}
```

As it try to open a file, I get this error:
[E][vfs_api.cpp:104] open(): /littlefs/g00000/c001 does not exist, no permits for creation

From this line:
https://github.com/espressif/arduino-esp32/blob/bb8c8559b62a0d2e888d51bf12bd50cc070f0b4b/libraries/FS/src/vfs_api.cpp#L105

I think this error should not be logged, if we just check if a file exists.

### Sketch

```cpp
char path[20];
createPath(path);  
if(!LittleFS.exists(path))
{
  LittleFS.mkdir(path);
}
```

### Debug Message

```plain
[E][vfs_api.cpp:104] open(): /littlefs/g00000/c001 does not exist, no permits for creation
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-12-22:
I agree. The message should be at info, debug, or verbose log level.  However it's not that simple.  exists() actually calls open() on the file (and then closes it) with read only permission, no create flag.  If your case you are looking at a directory.

Workaround: use `LittleFS.mkdir(path)` instead of exists().  It will create the directory if it doesn't exist.  Otherwise it just returns true.  See `bool VFSImpl::mkdir(const char *fpath)` in vfs_api.cpp.

Adrianotiger, 2022-12-22:
I believe you that's not that simple :)

But even as info or verbose, this is a wrong info: "XXX does not exist, no permits for creation". As I don't want create a file at this point of the code.

Rewrite the `open` function inside the `exists` functions is not a solution.

Maybe with a new argument with `__builtin_FUNCTION()` as parameter but it is probably also bad, for just a debug output.

Maybe opening the file with the `e`xists mode?

Anx2k, 2022-12-27:
I encountered this same issue, and instead of using their exists function I just use the normal sys/stat.h stat() function - it will return whether a file/directory exists or not, and won't print that error.  This actually works for SD cards as well as a bonus.

OttoKlaasen, 2023-02-12:
I have the same issue when i do a LittleFS.exist on an empty directory
It will show an error in the serial.
The code itself works fine.

Code snippet example:

      if (LittleFS.exists(FILE_PHOTO)) {
           //Returns true if a file with a given path exists; false otherwise.
           if(LittleFS.remove(FILE_PHOTO)) {
              //Deletes the file given its absolute path. Returns true if the file was deleted successfully.
               Serial.printf("FILE DELETED: %s\n", FILE_PHOTO);
           } else {
                Serial.println("ERROR WHILE DELETE FILE");
       }
      }
      delay(1000);   // Wait for it
      // Open dir folder
      dir = LittleFS.open("/");
      // Cycle all the content
      printDirectory(dir);
      Serial.println("Close dir connection.");
      dir.close();
      
      // next will call an error: [ 30214][E][vfs_api.cpp:105] open(): /littlefs/photo.jpg does not exist, no permits for creation
      // I will ignore the error as the code seems to work.
      bool FileExist = LittleFS.exists(FILE_PHOTO);

      if(!FileExist){
          Serial.println("FILE DOES NOT EXIST, so we can CREATE and WRITE to file..............");      
          file = LittleFS.open(FILE_PHOTO, FILE_WRITE);

BlueAndi, 2023-04-16:
In my use case I avoid checking first with exists() and if yes, use open() to load it because of performance. I would recommend to only show it in verbose mode.

hsaturn, 2023-05-27:
> I agree. The message should be at info, debug, or verbose log level. However it's not that simple. exists() actually calls open() on the file (and then closes it) with read only permission, no create flag. In your case you are looking at a directory, so the solution is slightly different.
> 
> Workaround: use `LittleFS.mkdir(path)` instead of exists(). It will create the directory if it doesn't exist. Otherwise it just returns true. See `bool VFSImpl::mkdir(const char *fpath)` in vfs_api.cpp.

This is not an answer. The library *MUST* not log in that case. I have the same problem, and I'm using the com port.
I tried many solutions to disable log with no success. And even with success, this would have been a workaround.
My roomba do not like at all receiving garbage through the serial port when I check for file existency !!!
LittleFS has to be fixed.
Technically, 
- adding a parameter with default to open() is a solution
- using stat instead of open in exists is a solution
- reversing logic: open uses exists then opens instead of exists uses open then closes is a solution
- using a 'special mode' (nullptr, '*' etc) is one more solution
- other solutions for sure exists
Don't say that it's not that simple. It is !

PR #10217 (2024-08-30): fix(littlefs): Use VFSImpl::exists() to avoid false error log
## Description of Change
Use exists() from VFSImpl base class and avoid using open() to check whether the file exists, because open() will log an error in case a file doesn't exists which is confusing, but might be relevant in case a developer uses open() directly and expects that the file is there.

## Related links
Closes #7615
closed 2024-08-30 by me-no-dev