espressif/arduino-esp32 #7616: rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M) arduino ide 1.8.19 and esp32 core 2.0.5  not declared in this scope

tradexsrl, 2022-12-22:
### Board

esp32 dev module

### Device Description

lilygo t-pcie

### Hardware Configuration

hx711

### Version

v2.0.5

### IDE Name

arduino ide

### Operating System

windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi i'm trying to use hx711 load cell amplifier with esp32 module and i read on internet that i need to slow down the cpu freq to 80Mhz
i'm using arduino ide 1.8.19 and esp32 core 2.0.5 but inserting #include "soc/rtc.h" in the code and declaring rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M); in the main gives me that rtc_clk_freq_set not declared in this scope.
googleing i saw many tutotrials doing the same , i don't understand why i get this problem
when compiling i do not see inclunding rtc.h.....

### Sketch

```cpp
#include "soc/rtc.h"
//Load Cell
#include "HX711.h"
float units;
RTC_DATA_ATTR float tara = 625;

// 1. HX711 circuit wiring
const int LOADCELL_DOUT_PIN = DOUT_PIN;
const int LOADCELL_SCK_PIN = SCK_PIN;

HX711 scale;

//#define zero_factor
RTC_DATA_ATTR float calibration_factor = 500; // this calibration factor must be adjusted according to your load cell
float units;
RTC_DATA_ATTR float tara = 625;
void setup()
{
rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
#if DEBUG_TO_SERIAL == 1
  scale.set_scale(calibration_factor); //Adjust to this calibration factor
  long zero_factor = scale.read_average(); //Get a baseline reading
  SerialMon.print("Zero factor: "); //This can be used to remove the need to tare the scale.
  SerialMon.println(zero_factor);
#endif
  scale.power_down(); //disattivo HX711
  rtc_clk_cpu_freq_set(RTC_CPU_FREQ_240M);
}
void loop()
{
  rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
  scale.set_scale(calibration_factor);
#if DEBUG_TO_SERIAL == 1
  SerialMon.print("Scale Reading");
#endif
  if (scale.wait_ready_timeout(1000)) {
    //scale.set_offset(zero_factor);
    units = scale.get_units(5), 1;
    units = units - tara;
    if (units < 0)
    {
      units = 0.00;
    }
  }

#if DEBUG_TO_SERIAL == 1
  SerialMon.print(" Weight: ");
  SerialMon.print(units);
  SerialMon.print(" grams");
  SerialMon.print(" calibration_factor: ");
  SerialMon.print(calibration_factor);
  SerialMon.println();
#endif
  scale.power_down();
  rtc_clk_cpu_freq_set(RTC_CPU_FREQ_240M);
}
```

### Debug Message

```plain
In function 'void setup()':
smart-trap:353:3: error: 'rtc_clk_cpu_freq_set' was not declared in this scope
   rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
   ^~~~~~~~~~~~~~~~~~~~
note: suggested alternative: 'rtc_clk_apb_freq_get'
   rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
   ^~~~~~~~~~~~~~~~~~~~
   rtc_clk_apb_freq_get
In function 'void loop()':
smart-trap:623:3: error: 'rtc_clk_cpu_freq_set' was not declared in this scope
   rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
   ^~~~~~~~~~~~~~~~~~~~
note: suggested alternative: 'rtc_clk_apb_freq_get'
   rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
   ^~~~~~~~~~~~~~~~~~~~
   rtc_clk_apb_freq_get
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-12-22:
Try setCpuFrequencyMhz() in esp32-hal-cpu.h

tradexsrl, 2022-12-23:
Hi , thank's for the reply , i set #include "esp32-hal-cpu.h" and used setCpuFrequencyMhz(RTC_CPU_FREQ_80M) but gives me rtc_cpu_freq_80m not declared in this scope
inspecting file esp32-hal-cpu.h i saw value of 80 and 240 , do i need to insert this value or for example 8000000L - 24000000L as i found in other tutorial?

tradexsrl, 2022-12-23:
a note: why all tutorials on internet related to hx711 and esp32 with arduino ide succesfully use rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
and it doesn't work for me?
i do not think that everyone do it wrong...

lbernstone, 2022-12-23:
You are missing the include for soc/rtc.h which would expose this function.  The example code was probably written years ago, before the setCpuFrequencyMhz function was added to arduino-esp32. The new function manages all of the underlying timers, not just the rtc multiplier. Just based on the rest of the code, it also looks like they were using deep sleep, and wanted to control the speed during sleep as well.
Please close the issue if the fix works.
Don't trust everything you read on the internet.

tradexsrl, 2022-12-23:
No if you look at the begining of the sketch there is #include "soc/rtc.h"
but anyway ok so the new one to be used  is seCpuFrequencyMhz

closed 2022-12-23 by tradexsrl

lbernstone, 2022-12-23:
The API has changed, and in any case, rtc.h includes all sorts of warnings that these functions are not to be directly used.

isasetiawan, 2023-07-12:
Hi @tradexsrl based on arduino-esp32 version 2.0.9 package, I found that this snippet work to me
```ino
#include <Arduino.h>
#include "soc/rtc.h"
#include "HX711.h"

const int LOADCELL_DOUT_PIN = 16;
const int LOADCELL_SCK_PIN = 4;

HX711 scale;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  rtc_cpu_freq_config_t config;
  rtc_clk_cpu_freq_get_config(&config);
  rtc_clk_cpu_freq_to_config(RTC_CPU_FREQ_80M, &config);
  rtc_clk_cpu_freq_set_config_fast(&config);

  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
}
```
It seem the arduino-esp32 API changed into these functions
* `rtc_clk_cpu_freq_get_config`
* `rtc_clk_cpu_freq_to_config`
* `rtc_clk_cpu_freq_set_config_fast`

Instead of `rtc_clk_cpu_freq_set`

Hopefully. this will help

LindelwaSfiso, 2025-03-10:
On esp32 board version 3.0.7 + Esp32S Dev Board, this worked for me 

```c++
#include <Arduino.h>
#include "soc/rtc.h"
#include "HX711.h"

const int LOADCELL_DOUT_PIN = 16;
const int LOADCELL_SCK_PIN = 4;

HX711 scale;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  rtc_cpu_freq_config_t config;
  rtc_clk_cpu_freq_get_config(&config);
  rtc_clk_cpu_freq_mhz_to_config(RTC_XTAL_FREQ_40M, &config);
  rtc_clk_cpu_freq_set_config_fast(&config);

  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
}
```