espressif/arduino-esp32 #7617: get.py script doesn't support on MacOs with M1 chip

meirke, 2022-12-22:
### Board

ESP32-C3-MINI-1

### Device Description

ESP32-C3-DevKit-RUST-1

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS-13.1-arm64-arm-64bit

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

on installation when run get.py I get :
System: Darwin, Bits: 64, Info: macOS-13.1-arm64-arm-64bit
Platform: x86_64-apple-darwin
when I try to compile I get :
Bad CPU type in executable
Compilation error: exit status 126

### Sketch

```cpp
void setup(){
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ;
  }

  Serial.println("Hello Arduino!");
}

void loop(){
}
```

### Debug Message

```plain
Bad CPU type in executable
Compilation error: exit status 126
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

meirke, 2022-12-22:
[    arduino_platform_names = {'Darwin'  : {32 : 'i386-apple-darwin',   64 : 'x86_64-apple-darwin'},](https://github.com/espressif/arduino-esp32/blob/b63f947037bece7a499faf0043636dd8531b885e/tools/get.py#L185)

me-no-dev, 2022-12-23:
Interesting... I am on M1 and have no issues with get.py or compilation

me-no-dev, 2022-12-23:
Is it possible that you have turned off Rosetta in some way?

meirke, 2022-12-23:
Ohh I see It happen because I tried to run it on a native Arduino IDE, and now I installed the intel version under the rosette and it's working.
Thank you

closed 2022-12-23 by meirke