espressif/arduino-esp32 #7618: BLE Beacon Scanner example might not handle negative temperatures?

Humancell, 2022-12-23:
### Board

Olimex ESP32 PoE ISO

### Device Description

PoE PCB using the ESP32 WROOM 32UE

### Hardware Configuration

Nothing special

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

All

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_Beacon_Scanner/BLE_Beacon_Scanner.ino#L118

When low temperatures occur, and go below 0 C, the calculations do not seem to be correct.  It appears that using these 16-bit values in a 32-bit INT doesn't detect/use twos complement to handle the negative numbers properly.

In my final code I moved to a "signed short" which seems to fix the issue.

### Sketch

```cpp
signed short value;

          // extract the temperature
          value = (cServiceData[TEMP_MSB] << 8) + cServiceData[TEMP_LSB];
```

### Debug Message

```plain
No debug ... just wrong calculated values.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-12-28:
Your sketch doesn't match the example.  Note the (int) cast on each payload byte.  Also we don't have your hardware.  It would be very helpful to know the actual bytes received in ServiceData[] as well as the actual temperature.

SuGlider, 2022-12-29:
This seems to be just a casting matter.
`uint32_t` with just 16 bits in the LSB will be always positive.

Casting it to whatever struct or type is an application subject, not related to the BLE Beacon Scanner example.

SuGlider, 2022-12-29:
I'll add this fix. Thanks.

SuGlider, 2023-02-02:
@PilnyTomas 

The temperature is represented in **signed** 8.8 fixed-point notation and measured in Celsius.
https://github.com/google/eddystone/blob/master/eddystone-tlm/tlm-plain.md 
When Temperature is not supported BLE will return 0x8000, which means -128 °C.

SuGlider, 2023-02-03:
As we talked, the change shall be in 
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_Beacon_Scanner/BLE_Beacon_Scanner.ino#L118-L120

I think that it just has to use `signed short`, like 
``` cpp
          // BIG ENDIAN payload in signed 8.8 fixed-ppint notation. Unit is Celsius. 
          int16_t temp_payload = payLoad[16] + (payLoad[15] << 8);
          if (payLoad[15] == 0x80 && payLoad[16] == 0) {
            Serial.printf("This device does not support measuring temperature.\n");
          } else {
            float calcTemp = temp_payload / 256.0f;
            Serial.printf("Reported temperature from data: %.2fC\n", calcTemp);
          }
```

But it shall be verified by testing some cases, like below:

For instance:

Signed Short Value -32768 = 0x8000 shall be calculated as -128 °C
Signed Short Value -9024 = 0xDCC0 shall be calculated as -35.25 °C
Signed Short Value 9024 = 0x2340 shall be calculated as +35.25 °C
Signed Short Value 384 = 0x0180 shall be calculated as +1.50 °C
Signed Short Value -1024 = 0xFC00 shall be calculated as -4.00 °C

SuGlider, 2023-02-03:
@PilnyTomas - The exactly same issu also happens with the code in
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEEddystoneTLM.cpp

This also needs to be fixed.

PilnyTomas, 2023-02-03:
@Humancell could you please try the change in #7791 ?
I don't have the devices to test this.

PR #7791 (2023-02-20): Fix for negative temp in Eddystone TLM; solving #7618
This PR changes the conversion of 8.8 to float and vise versa to accommodate negative values.

Fix #7618 
Fix #7858
closed 2023-02-20 by me-no-dev