espressif/arduino-esp32 #7620: Error during Complining while including <SD.h> file

rsramki, 2022-12-24:
### Board

ESP32 PICO-D4

### Device Description

I am using ESP32 PICO DevKit1 for checking on SDCARD.

### Hardware Configuration

#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define SCK  22
#define MISO  19
#define MOSI  21
#define CS  5

SPIClass spi = SPIClass(VSPI);

  spi.begin(SCK, MISO, MOSI, CS);

  if (!SD.begin(CS,spi,80000000)) {
    Serial.println("Card Mount Failed");
    return;
  }
  

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Checked with Version 2.0.5 and 2.0.6 having same issue so far.

Error While compiling code after including SD.H header file. C:\Users\XXXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:24:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.

### Sketch

```cpp
Sample SD CARD program after add SPI is working file. While doing new program and include "SD.H" file getting below error while running

******  Sample Program SD CARD modified **** WORKING FINE,, 
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define SCK  22
#define MISO  19
#define MOSI  21
#define CS  5

SPIClass spi = SPIClass(VSPI);
**********************
I have created a new program and included the same giving error message, See below
     C:\Users\XXXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:24:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.    ^
```

### Debug Message

```plain
C:\Users\XXXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:24:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.

![ESP32-SDScreenshot](https://user-images.githubusercontent.com/32425707/209425362-824df550-f259-4d80-865c-c7ef265c99d4.jpg)

After making changes to  memory.h instead of memory

following are additional errors.
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:27:1: error: unknown type name 'namespace'
 namespace fs
 ^~~~~~~~~
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:28:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:121:1: error: unknown type name 'using'; did you mean 'ulong'?
 using fs::FS;
 ^~~~~
 ulong
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:121:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 using fs::FS;
         ^
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:122:1: error: unknown type name 'using'; did you mean 'ulong'?
 using fs::File;
 ^~~~~
 ulong
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:122:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 using fs::File;
         ^
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:123:1: error: unknown type name 'using'; did you mean 'ulong'?
 using fs::SeekMode;
 ^~~~~
 ulong
C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/FS.h:123:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 using fs::SeekMode;
         ^
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rsramki, 2022-12-24:
[image]

me-no-dev, 2022-12-24:
Is it possible that you are including SD.h inside a C file and not CPP?

rsramki, 2022-12-24:
> Is it possible that you are including SD.h inside a C file and not CPP?

How to do this?

mrengineer7777, 2023-01-03:
> > Is it possible that you are including SD.h inside a C file and not CPP?
> 
> How to do this?

What is the name of your file that references SD.h?

Jason2866, 2023-04-19:
Closing since no feedback.

closed 2023-04-19 by Jason2866