espressif/arduino-esp32 #7621: Arduino ESP32 Framework V2.0.5 Changes TWAI Errata Config Options

abombay, 2022-12-24:
### Board

ESP32­-WROOM­-32E/ESP32-DevkitC

### Device Description

Custom PCB or Dev Module.

### Hardware Configuration

CAN Bus transceiver connected to pins 4 and 5.

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

macOS 12.0.1

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have had issues with a few devices failing to recover from bus off state. I located the cause to be the absence of CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC set in the Arduino framework. I spent quite a bit of time converting over to ESP-IDF to allow the extra flexibility of setting various sdkconfig (tools/sdk/esp32/sdkconfig) options. I copied the latest arduino sdkconfig to get started but when I went to change the TWAI errata options I found they were already set. I thought I was losing my mind as I had checked before and they were not set. I went back through the commits and found "ESP-IDF v4.4.2 and build time elf2bin for the bootloaders (#7259)".

I see that 4 errata fixes were enabled in that commit without any mention of them in release notes or issues. I am happy they are changed, I think it's the right thing, however, I have some questions.

1. Was there a particular reason these were just set now in this commit?

2. Are these options going to stay set, can I rely on them or is they going to change in the future.

3. Is there a way that I can verify these options are set at compile time to warn me incase I miss checking the source of each update.

Thanks!

The sdkconfig commit in question: [https://github.com/espressif/arduino-es ... d333cd4bb7](https://github.com/espressif/arduino-esp32/commit/da66b2b665624dff8b7793904669f6fe0bc8be1a#diff-e841640ecb84f39a6641d517f764d240368658a1c4797a7a09ba0cd333cd4bb7)

```
#
# TWAI configuration
#
# CONFIG_TWAI_ISR_IN_IRAM is not set
CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC=y
CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=y
CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID=y
CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT=y
# end of TWAI configuration
```

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-12-30:
This was actually enabled via https://github.com/espressif/esp32-arduino-lib-builder/commit/7015cc11f514b317b958ab00ca182e5a321143e4 which builds the pre-built binary bits that end up in this repository.

It should be safe to assume these are enabled by default going forward, however, you can also add checks for these being defined in your code using something like:

```
#include "sdkconfig.h"

#if !defined(CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC) || ...
#error CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC is required to be enabled....
#endif
```

This way you have a build time failure if these options ever change from what you expect.

abombay, 2022-12-30:
Ahh, this is why I was not able to find it. I did figure out how to add a check, just wasn't sure if platformio was finding the correct sdkconfig.h as it was prebuilt, thanks!

closed 2022-12-30 by abombay