espressif/arduino-esp32 #7625: AP not Found after adrduino framework upgrade

aragornale, 2022-12-24:
### Board

ESP32 Dev Module

### Device Description

Esp32 Wroom connected to an custom board, it worked before upgrade

### Hardware Configuration

Display 20x4 i2C, working fine, some solid state reles, Neopixel led 

### Version

v2.0.5

### IDE Name

Platformio in VSCode

### Operating System

Windows 10

### Flash frequency

default

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hi There. I´ve recently upgraded my Arduino Framework in Platformio/VSCode from 1.x to 2.0.5 and my 2 years development proyect stopped to connect to wifi. After 1 day of trying and trying, i´ve compiled and uploaded the wifi client connect example provided from examples. Still coudn´t connect to my AP, saying AP_NOT_FOUND reason, and of course my AP is working fine cause i have another 3 iot devices connected to it. Everything went fine until i´ve upgraded my Arduino Framework. Please, anyone knows any suggestion for this issue. Best reggards.

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

const char* ssid     = "iot";
const char* password = "*******";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop()
{
    delay(5000);
}
```

### Debug Message

```plain
In this example just only get infinites ".". With my full code, with corresponding handlers i get, "station Disconnected" reason: NO_AP_FOUND
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

aragornale, 2022-12-24:
i´ve posted this issue in Platformio forum too, and someone else suggested to try platform = espressif32@3.5.0 in platofrmio.ini. So, this downgraded the arduino framework, and with the same sketch, I could connect succesfully. The Wifi Library working is 1.0; the wifi 2.0 is the problem. But I wish to use latest libraries on everything, cause I´m looking for bugs in other stuff (AsynWebServer Panic sometimes when randomly loads a webpage)

lbernstone, 2022-12-24:
Set the core debugging to verbose and post that log.

aragornale, 2022-12-24:
.....[ 39399][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: iot, BSSID: 00:00:00:00:00:00, Reason: 201
[ 39400][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 39406][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 39413][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 39422][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.....[ 41852][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: iot, BSSID: 00:00:00:00:00:00, Reason: 201
[ 41853][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 41860][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 41866][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 41875][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0

lbernstone, 2022-12-24:
Most likely it is using WPA, which is no longer supported by default (it can be decrypted). Set it with [WiFi.setMinSecurity](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#setMinSecurity)(WIFI_AUTH_WPA_PSK). 
If not, make sure you can see the network with [WiFiScan](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino)

aragornale, 2022-12-24:
you could be in the right way... for compatibility troubles with older devices, i´ve setted my AP security in WPA, and MAC autentication from AP side. By the way this function member setMinSecurity doesnt belong to any member in library, but i´ve seen inside de .cpp implementacion an struct with this configuracion... i´m gonna deep dive in this implementacion

lbernstone, 2022-12-24:
https://github.com/espressif/arduino-esp32/blame/master/libraries/WiFi/src/WiFiSTA.h#L68

aragornale, 2022-12-24:
I was in espressif32@3.5.0 that doesnt have  setMinSecurity implemented.
The good news, i´ve setted my AP in WPA2, and it connected!! My AP it´s a Mikrotik router and has de hability to set multiple types of AP´s with diffent type of encryption
Thanks very much!! merry christmas.

everslick, 2022-12-28:
@aragornale pls, close the issue, if it is resolved for you.

WebDust21, 2024-07-12:
This bug seems to have resurfaced in Arduino-ESP32 v3.0.1 for some strange reason.  (I never had issues with this in Arduino-ESP32 v2.0.xx.)  

- Chip in question is an ESP32-WROOM-32E on a custom PCB (3A-rated 3.3v buck converter is literally 10mm from the ESP32 power pins, with 2x 22uF + 0.1uF decoupling caps--so it's not a lack of power to the ESP32).

Basically, I'm (more often than not) experiencing an instant "Reason: 201 - NO_AP_FOUND" log upon calling "WiFi.begin" to an SSID other than the ESP32 is currently connected to.

(Worth noting that the ESP32 _always_ connects properly to saved SSID creds at power on.  First time connecting to a WiFi network: no problem.  Subsequent times...usually fail.)

Code in question is called from a captive portal page accepting user input for SSID + password, as follows:
```
// Try to connect to the new network.
//NOTE: The SSID arg is prefixed with a single character to indicate "open" or "locked" network: we must skip it.
String newSSID = localServer.arg("ss").substring(1);
String newPass = localServer.arg("ps");
log_d("Attempting connection with SSID \"%s\", password \"%s\"", newSSID.c_str(), newPass.c_str());

wl_status_t wConnRes = WiFi.begin(newSSID, newPass); log_d("WiFi Begin result: %d", wConnRes);
```

Here's a verbose debug log from a **successful** network switch:
```
[ 59675][D][network_thread.cpp:416] lsvCONN(): Attempting connection with SSID "[...]", password "[...]"
[ 59737][W][STA.cpp:362] connect(): STA currently connected. Disconnecting...
[ 59878][V][STA.cpp:214] _onStaEvent(): STA Disconnected: SSID: , BSSID: b4:4b:d6:2e:8b:30, Reason: 8
[ 59934][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 14 - STA_DISCONNECTED
[ 59981][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 14 - STA_DISCONNECTED
[ 60027][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
[ 60105][D][network_thread.cpp:418] lsvCONN(): WiFi Begin result: 6
[ 60945][V][STA.cpp:204] _onStaEvent(): STA Connected: SSID: [...], BSSID: b6:4b:d6:1e:8b:30, Channel: 1, Auth: WPA2_PSK
[ 61013][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 13 - STA_CONNECTED
[ 61058][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
[ 64211][V][NetworkInterface.cpp:66] _onIpEvent(): sta Got New IP: 192.168.9.242 MASK: 255.255.255.0 GW: 192.168.9.1
[ 64273][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 16 - STA_GOT_IP
[ 64317][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 16 - STA_GOT_IP
[ 64360][V][STA.cpp:169] _onStaArduinoEvent(): STA IP: 192.168.9.242, MASK: 255.255.255.0, GW: 192.168.9.1
[ 64417][D][network_thread.cpp:151] operator()(): WiFi STA IP received, requesting SNTP fix.
```

Interestingly, it works with 100% reliability when a **verbose** log is requested.  However, with a _debug_ log, it fails more than 90% of the time with an output log of this:

```
[ 44067][D][network_thread.cpp:416] lsvCONN(): Attempting connection with SSID "[...]", password "[...]"
[ 44078][W][STA.cpp:362] connect(): STA currently connected. Disconnecting...
[ 44131][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
[ 44220][D][network_thread.cpp:418] lsvCONN(): WiFi Begin result: 6
[ 44622][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[ 44629][D][STA.cpp:153] _onStaArduinoEvent(): WiFi Reconnect Running
[ 44635][W][STA.cpp:533] disconnect(): STA already disconnected.
[ 45046][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
```
(note that the "WiFi Begin result" can be almost anything; it does not seem to have any bearing on the outcome.  Reran the test again, and it returned "1")

Worth noting that the main loop code tries to rerun "WiFi.begin()" on the (Preferences-saved) SSID + password every 30 seconds (no, it's blocked from running during the captive portal connection)--and after the initial connection failure, these _also_ fail with the same error:
```
[ 74689][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[105390][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[136058][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[166742][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[197417][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[228089][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[258773][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[289433][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[320107][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[350806][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[381200][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
...
```

While "WiFi.begin()" is constantly failing, I can easily run a WiFi scan (from the captive portal), and the scan results are correct.  And the captive portal webserver is running without issues.

It seems to me that something in the Arduino-ESP32 WiFi substructure needs to wait a bit longer before concluding "NO_AP_FOUND", due to the following findings:

1. It works with 100% reliability when a verbose debug log is enabled (tested at both 19200 and 115200 baud) -> the serial printing consumes a tiny bit of time.
2. The "Reason: 201 - NO_AP_FOUND" message appears almost immediately after disconnection from the network it was connected to, i.e. the WiFi library is jumping to a conclusion before the actual ESP32 radio has time to finish the WiFi disconnection.

Thoughts?  Suggestions?  Solutions?

me-no-dev, 2024-07-12:
@WebDust21 please open a new issue. Provide minimal example code and we will look into it. Also please try with 3.0.2 or even better, master branch.

WebDust21, 2024-07-12:
Arduino-ESP32 v3.0.2 did not solve the issue.  I will create a minimally reproducible example, and open a new issue.
Thanks!

WebDust21, 2024-07-13:
update: I have not been able to create a minimally reproducible example that showcases the problem (it always works properly)...although I can reproduce the issue with 100% reliability in the project context.  Yes, I know that is not helpful!

But I did find a solution...

Starting in Arduino-ESP32 2.0.6, running a WiFi scan ("WiFi.scanNetworks()") would consistently take about 13 seconds to respond--which would cause phones/laptops to lose a connection to the WiFi AP.  Quite problematic for a captive portal network manager implementation.
By trial and error, I found that by specifying a "passive" scan and 125mS/channel (i.e. "WiFi.scanNetworks(false, false, true, 125);"), the scan would complete in a far more reasonable 3-4 seconds.

Well, it turns out that use of a passive WiFi scan (in the context of my project) on Arduino-ESP32 v3.0.x seems to be the cause for "WiFi.begin()" calls being unable to access any WiFi network (in the context of my project) until the ESP32 is restarted.  

Fortunately, the WiFi scan issue has been fixed in Arduino-ESP32 v3.0.x, meaning that I can revert back to a normal "WiFi.scanNetworks()" call (3-4 seconds) and solve both problems.

Trying a passive WiFi scan in a minimal captive portal example does not replicate the issue, however.

VojtechBartoska, 2024-07-15:
hello @WebDust21, seems it si really specific to your project/code. Or do you have any idea/proposal for code changes in this repo.

I am aware we are not able to do much if we will not be able to reproduce it.

WebDust21, 2024-07-15:
> I am aware we are not able to do much if we will not be able to reproduce it.

I completely understand.

I've been able to solve the issue in my project (by using active WiFi scan instead of passive).  As I've thus far been unable to successfully create a "minimal" example to replicate the issue (it always works correctly!), I think it best to let this rest.