espressif/arduino-esp32 #7626: HTTPS OTA example Failed to verify peer certificate

gomez9656, 2022-12-24:
### Board

ESP32 Dev Module

### Device Description

ESP32 board powered by PC USB

### Hardware Configuration

Out of the box HPPTS OTA example for ESP32 in Arduino https://github.com/espressif/arduinoesp32/blob/master/libraries/Update/examples/HTTPS_OTA_Update/HTTPS_OTA_Update.ino  e

### Version

latest master (checkout manually)

### IDE Name

PlatformIO with Arduino framework

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello there!

I'm working with the HTTPS_OTA_Update example in my ESP32 board but I get the following error:
[image]

The HTTPS server is running in git bash with the OTA ESP-IDF example instructions: https://github.com/espressif/esp-idf/tree/master/examples/system/ota

I tested this example with HTTP + python server and it works.  THE AWS S3 example doesn't work because the code is for HTTP, and now the AWS S3 bucket is HTTPS.

I also tested the example using echo "" | openssl s_client -showcerts -connect raw.githubusercontent.com:443 | sed -n "1,/Root/d; /BEGIN/,/END/p" | openssl x509 -outform PEM >ca_cert.pem and storing the firmware in my github: https://github.com/gomez9656/aaaaaa/blob/master/firmware.bin

This option gives me the error: 
[image]

Overall. when using a web server I get a certificate error, and when using the local HTTPS server I got the first error.

What should I be looking for here? 

I really appreciate all the help and thanks in advance :) + merry Xmas 24th December 👍 

### Sketch

```cpp
Same example as https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/HTTPS_OTA_Update/HTTPS_OTA_Update.ino

I have also used ESP-IDF example: https://github.com/espressif/esp-idf/tree/master/examples/system/ota/advanced_https_ota
```

### Debug Message

```plain
When using PlatformIO + HTTPS local server in git bash
[... 506 lines cut ...]
␛[0m
␛[0;32mI (9622) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 3453 (-0xfffff283)
␛[0m
␛[0;32mI (9632) mbedtls: ssl_msg.c:2089 <= fetch input
␛[0m
␛[0;32mI (9672) mbedtls: ssl_msg.c:1144 => decrypt buf
␛[0m
␛[0;32mI (9712) mbedtls: ssl_msg.c:1741 <= decrypt buf
␛[0m
␛[0;32mI (9742) mbedtls: ssl_msg.c:4015 <= read record
␛[0m
␛[0;32mI (9742) mbedtls: ssl_msg.c:5693 <= read
␛[0m
␛[0;32mI (9742) HTTP_CLIENT: Body received in fetch header state, 0x3ffb9649, 467␛[0m
␛[0;32mI (9752) esp_https_ota: Starting OTA...␛[0m
␛[0;32mI (9752) esp_https_ota: Writing to partition subtype 16 at offset 0x110000␛[0m
␛[0;32mI (9762) advanced_https_ota_example: Running firmware version: 1␛[0m
␛[0;31mE (9772) esp_https_ota: Mismatch chip id, expected 0, found 5␛[0m
␛[0;31mE (9772) advanced_https_ota_example: Complete data was not received.␛[0m
␛[0;31mE (9792) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed␛[0m
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

gomez9656, 2022-12-27:
Hello again!

After long hours of debugging sessions, I finally solved the issue.

I had to set *config.skip_cert_common_name_check* variable to false using *HttpsOTA.begin(url, server_certificate, false)*

This variable was initialized to true when the class was initiated 
[image]

I double-checked with the ESP-IDF advanced_https_ota example and the behavior is the same. If I enable *example configuration -> skip server certificate CN fielcheck* then OTA doesn't work.
If I leave it unchecked, then it works. 

Altough the code is working for me, my final question is:

Why is *config.skip_cert_common_name_check*  initialized to true? shouldn't be false by default?

Thanks a lot :)

lvincek, 2023-11-11:
Thanks for this post.  I had success whenever I used my own self-signed.  I used the same logic with my Cloudflare certs, with no luck.  Your post just save me.

closed 2024-02-20 by VojtechBartoska