espressif/arduino-esp32 #7633: I2CSslave problem

crousekimo, 2022-12-27:
### Board

ESP32

### Device Description

nodemcu-32s

### Hardware Configuration

SDA: GPIO 21
SCL: GPIO 22

### Version

latest master (checkout manually)

### IDE Name

Arduino ide

### Operating System

window 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I try to make esp32 a i2c slave device.
This i2c device can help microbit online.
However, when I send i2c message to esp32, ask esp32 to send these message to a https server.
I get the following error messages.
May I know what these messages are?
Can I close these error messages?

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (i2c_slave_task) 
Core  1 register dump:
PC      : 0x40131dc6  PS      : 0x00060336  A0      : 0x80132cf5  A1      : 0x3ffd2580  
A2      : 0x36363636  A3      : 0x3ffe1fd0  A4      : 0x36363636  A5      : 0x3ffd2660  
A6      : 0x00000010  A7      : 0x00003600  A8      : 0x00000036  A9      : 0x3ffd28a0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffe0440  A13     : 0x3ffc3e94  
A14     : 0x3ffd2ac0  A15     : 0x3f41a134  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40131d88  LEND    : 0x40131dcc  LCOUNT  : 0x0000000b  

Backtrace:0x40131e27:0x3ffd25f00x40132d56:0x3ffd2950 0x40132dcb:0x3ffd2970 0x40132e05:0x3ffd2990 0x4012b0f0:0x3ffd29b0 0x4012b255:0x3ffd29d0 0x4012ad5b:0x3ffd2a40 0x401265b9:0x3ffd2a60 0x4012715c:0x3ffd2ad0 0x40127c37:0x3ffd2cc0 0x40127f89:0x3ffd2e20 0x40127fb0:0�N��D)�!!!�����!��

### Sketch

```cpp
My Sketch can be compiled and uploaded to esp32.
```

### Debug Message

```plain
No debug message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-12-27:
You can use the [exception decoder](https://github.com/me-no-dev/EspExceptionDecoder) to see where in your code the error is occuring.  Stack canary watchdog errors mean that you are running out of stack memory.  Use heap memory for large allocations.

crousekimo, 2022-12-28:
I try to use PSRAM. However, I still get the same errors.
However, these errors only occur when I send messages to https server.
MQTT http protocol are fine.
I think it needs to increase i2c_slave_task memory.
Do you know how to do it?

mrengineer7777, 2023-12-15:
Stale.  Not enough info to reproduce. Closing.

closed 2023-12-15 by mrengineer7777