espressif/arduino-esp32 #7634: SPIFFS filesystem problem

franz-66, 2022-12-27:
### Board

LILYGO TTGO T-Call V1.4 SIM800L

### Device Description

LILYGO TTGO T-Call V1.4 SIM800L on breadboard

### Hardware Configuration

ND

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Incorrect file size after creation with ESP32 FS tool.  The problem only occurs with core version 2.0.6.

[image]

### Sketch

```cpp
ND
```

### Debug Message

```plain
ND
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-01-02:
we added a new partition in order to have ESP-Insights working. That resulted in 64K less FS partition. We need to update all labels still

Speedy2k, 2023-02-17:
> we added a new partition in order to have ESP-Insights working. That resulted in 64K less FS partition. We need to update all labels still

Is this the reason why when we update from 2.0.5 to 2.0.6 it always reformat my SPIFFS partition ?? If yes is there a way to avoid that because it errase all saved data doing that!!

Thanx a lot!

me-no-dev, 2023-02-18:
@Speedy2k the one thing that comes to mind is to have a custom partitions.csv in your sketch folder that does not have that extra partition and extends the SPIFFS to the previous value (more or less copy the partition scheme from 2.0.5)

handmade0octopus, 2023-10-12:
@me-no-dev I encountered errors about core dump partition. I have a question - what if I OTA update my boards that doesn't have this partition? Is it needed or can I continue without using it (use older partition format)? What are benefits? I hope it won't cause any issues once we move to ESP-IDF 5.X based Arduino core?

lbernstone, 2023-10-12:
It is used by [esp-insights](https://github.com/espressif/esp-insights). If you aren't using that, then you don't need the crash dump partition. OTA updates will work just fine, the size of the firmware partitions has not changed.