espressif/arduino-esp32 #7637: missing null terminator for string

Franck78, 2022-12-28:
### Board

ESP32-S3

### Device Description

homemade

### Hardware Configuration

homemade

### Version

latest master (checkout manually)

### IDE Name

ARduino IDE

### Operating System

linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello
v2.0.6

this is the output of the demo program using BLE_client

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_client/BLE_client.ino

```
We are now connected to the BLE Server.
Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 2
data: AT⸮⸮⸮?⸮$

```
The data length is correct, I recognize just an 'AT' command from the HC06.
As you can see, there is some garbage added, meaning the zero is not where is should be.

https://github.com/espressif/arduino-esp32/blob/b63f947037bece7a499faf0043636dd8531b885e/libraries/BLE/examples/BLE_client/BLE_client.ino#L32

### Sketch

```cpp
bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
```

### Debug Message

```plain
x
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-12-28:
Are you sure that HC-06 can be used as BLE?!
As far as I know HC-06 is Bluetooth v2.0 (classic) only.
Therefore it can't be used as BLE device.

SuGlider, 2022-12-28:
The S3 can only connect using BLE with another BLE device. 
You may consider using another ESP32/C3/S3 to connect, or maybe, a BLE module such as the HM-10, instead of HC-06.

Franck78, 2022-12-28:
well, I swear to you I'm using my HC06 with the app you pointed me (ble client ino)

May be the bug is line 32 if we say that the 'bluetooth' transports an array of byte and a size for that array.

SuGlider, 2022-12-29:
Considering that your HC06 works with BLE, I can tell you that BLE doesn't understand NULL terminated strings.

The sender will send a payload with a defined length. 
The receiver side will read the length and then get the payload.

If you want to `Serial.print(char *)` the payload, you shall add the `\0' at `pData[length]`.

Franck78, 2022-12-29:
> If you want to Serial.print(char *) the payload, you shall add the '\0' at pData[length].

it happens it is _your_ app in need of this zero ;)

SuGlider, 2022-12-29:
I'll add that to the example. Thanks.

closed 2023-02-06 by me-no-dev