espressif/arduino-esp32 #7640: OTA update fails: boot_comm: mismatch chip ID

papakpmartin, 2022-12-30:
### Board

ESP32-S2FH4 custom board

### Device Description

Custom board with a couple of I2C devices. Board working great. Local firmware loading working great.

### Hardware Configuration

* GPIO 33 & 34 used for first I2C bus
* GPIO 35 & 36 used for second I2C bus (not in currently use)

### Version

v2.0.4

### IDE Name

VS Code + Platformio

### Operating System

macOS 12.5.1

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

460800

### Description

I can "Build" and "Upload and Monitor" normally. But when I try to OTA update the _exact same firmware_ that I loaded locally, the OTA update fails. The core issue appears to be this:

```
12:49:05.393 > E (67786) boot_comm: mismatch chip ID, expected 2, found 3460
12:49:05.398 > [ 67607][E][HTTPUpdate.cpp:442] runUpdate(): Update.end failed! (Could Not Activate The Firmware)
```

I have used this exact OTA update process for a while on the ESP8266.

I get through `HTTPUpdateStartCB` and `HTTPUpdateProgressCB`, then go to the `HTTPUpdateErrorCB`.

### Sketch

```cpp
Serial.println("OTA: Starting httpUpdate.update()");
t_httpUpdate_return ret = httpUpdate.update(
  client,
  "http://server.local:21880/firmware.bin",
  s.c_str() // the current firmware version
);
```

### Debug Message

```plain
### Here's console from when I upload the firmware for the ESP32 locally:

*  Executing task: platformio run --target upload --target monitor --environment esp32-s2-saola-1 

Processing esp32-s2-saola-1 (platform: espressif32; board: esp32-s2-saola-1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (5.1.1) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20004.220825 (2.0.4) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> <removed bitly url>
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- EEPROM @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- HTTPUpdate @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/esp32-s2-saola-1/firmware.elf
Checking size .pio/build/esp32-s2-saola-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.7% (used 35156 bytes from 327680 bytes)
Flash: [======    ]  56.6% (used 741546 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbmodemflip_Claidend1
Uploading .pio/build/esp32-s2-saola-1/firmware.bin
esptool.py v3.3
Serial port /dev/cu.usbmodemflip_Claidend1
Connecting....
Chip is ESP32-S2FH4
Features: WiFi, Embedded Flash 4MB, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 58:cf:79:00:00:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000c5fff...
Flash params set to 0x022f
Compressed 14592 bytes to 10123...
Writing at 0x00001000... (100 %)
Wrote 14592 bytes (10123 compressed) at 0x00001000 in 0.5 seconds (effective 251.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 451.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 689.7 kbit/s)...
Hash of data verified.
Compressed 741936 bytes to 482685...
Writing at 0x00010000... (3 %)
   <snip>
Writing at 0x000c288b... (100 %)
Wrote 741936 bytes (482685 compressed) at 0x00010000 in 12.7 seconds (effective 468.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
================ [SUCCESS] Took 17.46 seconds ================

### And here's what I see when I try to load that same firmware.bin to the ESP32 via OTA:

12:48:59.408 > FIRMWARE: Desired version different than current? 7 != 255
12:48:59.413 > FIRMWARE: Beginning OTA update process
12:48:59.417 > WiFi: Status: 3
12:48:59.418 > OTA: Attaching handlers
12:48:59.420 > OTA: Getting current firmware version
12:48:59.424 > OTA: Starting httpUpdate.update()
12:48:59.746 > OTA: HTTP update process started
12:48:59.847 > OTA: HTTP update process at 0 of 343632 bytes...
   <snip>
12:49:05.388 > OTA: HTTP update process at 343632 of 343632 bytes...
12:49:05.393 > E (67786) boot_comm: mismatch chip ID, expected 2, found 3460
12:49:05.398 > [ 67607][E][HTTPUpdate.cpp:442] runUpdate(): Update.end failed! (Could Not Activate The Firmware)
12:49:05.406 > 
12:49:05.406 > [ 67616][E][HTTPUpdate.cpp:356] handleUpdate(): Update failed
12:49:05.412 > 
12:49:05.415 > HTTP_UPDATE_FAILED Error (9): Update error: Could Not Activate The Firmware
12:49:05.419 > WiFi: Status: 3
12:49:05.421 > FIRMWARE: OTA update process completed
12:49:05.424 > FIRMWARE: Firmware block completed
```

### Other Steps to Reproduce

Here's what the server is sending:

```
> curl -v http://server.local:21880/firmware.bin -O
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0*   Trying 10.52.176.48:21880...
* Connected to server.local (10.52.176.48) port 21880 (#0)
> GET /firmware.bin HTTP/1.1
> Host: server.local:21880
> User-Agent: curl/7.79.1
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Access-Control-Allow-Origin: *
< X-Powered-By: Express
< Vary: Origin
< Access-Control-Allow-Credentials: true
< Content-Length: 343632
< Content-Type: application/octet-stream
< ETag: W/"53e50-TBiWs4kMeOJ6LOm6Ndta+MEFeuU"
< Date: Thu, 29 Dec 2022 16:47:43 GMT
< Connection: keep-alive
< Keep-Alive: timeout=5
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

papakpmartin, 2022-12-30:
Updating above, but I've run the same code on a ESP32-S2-DevKitM-1 with _exactly_ the same result. Just wanted to rule out any presumptions about it being because of a custom board.

papakpmartin, 2022-12-30:
UPDATE: I updated to Espressif 32 v5.3.0 and am still seeing the same thing. Here are the packages in use now:

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
```

Also tried building with `board_build.partitions = min_spiffs.csv` in case it was a space issue; same result.

schafon, 2022-12-30:
> UPDATE: I updated to Espressif 32 v5.3.0 and am still seeing the same thing. Here are the packages in use now:
>
> ```
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
>  - [...]

Hi, do you have a board with 8mb flash?
I wonder if It worked on my esp32s3 because it has 8mb

papakpmartin, 2022-12-31:
I don't have one but am probably going to grab one soon. I wondered the same, which is why I tried monkeying with the partitions.

papakpmartin, 2023-01-10:
OK! I got a ESP32-S3-DevKitC-1 and the OTA update code **_does_** work on it. So...

* ESP32-S2, simple HTTP OTA updating is not working
* ESP32-S3, simple HTTP OTA updating works

I'll upload sample code in a moment.

papakpmartin, 2023-01-10:
Here is code that **does work on ESP32-S3** and **does not work on ESP32-S2**:

```
#include <string>
#include <Arduino.h>
#include "credentials.h"
#include <WiFi.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>

WiFiClient client;

boolean wifi_connect() {
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println(F("WIFI: Connecting..."));
  }
  WiFi.persistent(true);
  Serial.println(F("WIFI: Connected to the Wi-Fi network"));
  return true;
}

void update_started() {
  Serial.println("OTA: HTTP update process started");
}

void update_finished() {
  WiFi.disconnect();
  Serial.println("LOOP: Delaying 5s so Wi-Fi reconnect is more reliable");
  delay(5000);
  Serial.println("OTA: HTTP update process finished");
}

void update_progress(int cur, int total) {
  Serial.printf("OTA: HTTP update process at %d of %d bytes...\n", cur, total);
}

void update_error(int err) {
  Serial.printf("OTA: HTTP update fatal error code %d\n", err);
}

void perform_firmware_update() {
    httpUpdate.setLedPin(LED_BUILTIN, LOW);

    Serial.println("OTA: Attaching handlers");
    httpUpdate.onStart(update_started);
    httpUpdate.onEnd(update_finished);
    httpUpdate.onProgress(update_progress);
    httpUpdate.onError(update_error);

    Serial.println("OTA: Getting current firmware version");
    String s = "xxx";

    Serial.println("OTA: Starting httpUpdate.update()");
    t_httpUpdate_return ret = httpUpdate.update(
      client,
      "http://server.local:21880/firmware.bin",
      s.c_str()
    );
    httpUpdate.setLedPin(LED_BUILTIN, HIGH);

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
}

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.println("SETUP: Serial setup complete");
  Serial.println("SETUP: Complete");
}

void loop() {
  Serial.println("LOOP: Starting loop");
  wifi_connect();
  delay(1000);
  perform_firmware_update();
  WiFi.disconnect();
}
```

This code was compiled and stored on the server so that OTA would basically result in running the exact same code again.

On ESP32-S3 it's functioning perfectly.

On ESP32-S2 I'm seeing this (essentially identical to my original problem report above):

```
13:35:46.884 > OTA: HTTP update process at 742224 of 742224 bytes...
13:35:46.890 > E (179749) boot_comm: mismatch chip ID, expected 2, found 9
13:35:46.895 > [180212][E][HTTPUpdate.cpp:442] runUpdate(): Update.end failed! (Could Not Activate The Firmware)
13:35:46.906 > 
13:35:46.906 > [180221][E][HTTPUpdate.cpp:356] handleUpdate(): Update failed
13:35:46.912 > 
13:35:46.912 > HTTP_UPDATE_FAILED Error (9): Update error: Could Not Activate The Firmware
13:35:46.919 > LOOP: Starting loop
```

miloserdev, 2023-01-12:
In my case i need to add encryption type to POST packet
Just like `enctype="multipart/form-data"`

papakpmartin, 2023-01-12:
@miloserdev where do you add that? I'm using `httpUpdate.update()` and I don't think that let's me set headers (and I'm not even sure it's a HTTP POST).

papakpmartin, 2023-01-17:
@hmalpani might [your recommendation here](https://github.com/espressif/esp-idf/issues/10467#issuecomment-1368673039) also apply to the problem I'm seeing?

miloserdev, 2023-01-18:
> @miloserdev where do you add that? I'm using `httpUpdate.update()` and I don't think that let's me set headers (and I'm not even sure it's a HTTP POST).

You can use esp_ota instead of httpUpdate.
See this [example](https://github.com/miloserdev/experiments/blob/master/esp32_relay/esp32_relay_http.ino)

hmalpani, 2023-01-18:
> @hmalpani might [your recommendation here](https://github.com/espressif/esp-idf/issues/10467#issuecomment-1368673039) also apply to the problem I'm seeing?

Hello @papakpmartin 
The linked fix was added to fix GCM hardware. I am not sure whether this fix is applicable here as you are not using pre_encrypted_ota images or `mbedtls` at any stage of the OTA update.
What I can recommend to you is to try disabling the CONFIG_MBEDTLS_HARDWARE_GCM in menuconfig. If after disabling it, you don't get the error, the fix might be applicable in this case.

closed 2023-04-18 by papakpmartin

feng612266, 2024-11-01:
> Hi all! Super dumb, but leaving this here in case it helps others...
> 
> The original issue was that I was unexpectedly seeing `mismatch chip ID` errors.
> 
> The errors were correct. **I had wrong firmware being served from the server.** It really was for the wrong chip.
> 
> What I _should_ have done was download that server file locally and tried loading it locally. That would have taken OTA out of the picture and shown that it was bad firmware for that chip.
> 
> Sorry to have been a bother!

hello ,i have this same error,how can get right chip ID,my board is esp-32-s3 thankyou