espressif/arduino-esp32 #7645: C++17 support?

Zhu-jiatong, 2022-12-31:
### Board

ESP32 Dev module

### Device Description

ESP32 Dev module
SD card with SD card module

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

VSCode with Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am trying to use `std::filesystem`, the `insert_or_assign` method of `std::tuple`, and structured binding in my project, but it appears that C++17 is not supported, as shown below.
[image]

Is there a way by which I can enable C++17? 
Are there any existing incompatibilities/issues concerning the use of C++17?
When will C++17 be supported 'out-of-the-box' without the need for manual configuration?

Thank you.

### Sketch

```cpp
Use any functionalities newly introduced in C++17
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2022-12-31:
ESP-IDF has switched to C++20 as the default language standard in the recent v5.0 release ([details](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html)). Arduino-esp32 project will get this change when it upgrades from IDF v4.4.x to IDF v5.x, most likely in Q1-Q2 this year.

In the meantime you might be able to add `compiler.cpp.extra_flags=-std=gnu++17` to your [`platform.local.txt`](https://arduino.github.io/arduino-cli/0.21/platform-specification/#platformlocaltxt) file. No guarantees this will work, though, since we aren't testing this particular configuration.

Zhu-jiatong, 2022-12-31:
> ESP-IDF has switched to C++20 as the default language standard in the recent v5.0 release ([details](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html)). [...]

Thanks for your information. 

The mentioned method appears not to be working as expected. 
I do not have the [platform.local.txt](https://arduino.github.io/arduino-cli/0.21/platform-specification/#platformlocaltxt) file in the directory, so I tried passing the flag using the [`arduino.json` file generated by VSCode](https://github.com/microsoft/vscode-arduino#options) by adding the following entry in the file:
```
    "buildPreferences": [
        [
            "compiler.cpp.extra_flags",
            "-std=gnu++17"
        ]
    ]
```
the result is shown in in the image below:
[image]
The version is still unchanged.

Zhu-jiatong, 2022-12-31:
However, the size sketch does change from 
`Sketch uses 909637 bytes (46%) of program storage space...Global variables use 46668 bytes (14%) of dynamic memory`
to
`Sketch uses 909701 bytes (46%) of program storage space...Global variables use 46660 bytes (14%) of dynamic memory`

Jason2866, 2022-12-31:
Platform.txt is not used in Platformio

Zhu-jiatong, 2022-12-31:
> Platform.txt is not used in Platformio

@Jason2866 I am not using PlatformIO, I am using [Arduino extension](https://github.com/microsoft/vscode-arduino.git) with VSCode.

hansmbakker, 2023-01-28:
Work is being done in https://github.com/espressif/arduino-esp32/pull/7733

closed 2023-01-29 by Zhu-jiatong