espressif/arduino-esp32 #7647: Compiler creating debug.cfg, debug_custom.json, and esp32.svd files in my sketch folder under version 2.0.6

HomeSpan, 2022-12-31:
### Board

Any board

### Device Description

Occurs on all Espressif development boards

### Hardware Configuration

Nothing

### Version

other

### IDE Name

Arduino 1.8.19

### Operating System

MacOs 13.1

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

In the latest release (2.0.6), whenever you compile a sketch, the compiler is dropping three new files into the sketch folder:  *debug.cfg*, *debug_custom.json*, and *esp32.svd*.  These files have never appeared when compiling with version 2.0.5 or earlier.  Is there a way of turning the automatic output of these files?

### Sketch

```cpp
Any sketch.
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

Compile any sketch with Arduino IDE 1.8.19 using the latest Arduino-ESP32 version 2.0.6 and the compiler drops three new files, presumably used for debugging/testing, in to the sketch folder. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

svdrummer, 2023-01-01:
I have just been asking this question on the Arduino forum. I have started getting these files being added. 
debug.cfg
debug_custom.json
esp32.svd

me-no-dev, 2023-01-02:
they are required for JTAG debugging with Arduino IDE 2.0.3+

IeiuniumLux, 2023-01-03:
@me-no-dev, The above 3 files are generated with the classic Arduino 1.8.19 IDE. Why does it now require those files too?

me-no-dev, 2023-01-03:
@IeiuniumLux it does not, but we can not detect which version is the IDE in order to copy them only on the new one

HomeSpan, 2023-01-03:
@me-no-dev, Can the output be an option that is selected from the tool menu, similar to setting the "Core Debug Level" or "Erase all Flash Before Sketch Upload"?  Perhaps something like "Output JTAG files" where the default is "No" so that the files would never be output in version 1 of the Arduino IDE and would be optionally output in version 2?

Separately, is there anything in either "platform.txt" or "programmers.txt" that we can edit to disable the JTAG output?

me-no-dev, 2023-01-03:
@HomeSpan I'm not sure if it's possible to be controlled by an option, but could be tried. You can remove [those lines](https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L186-L196) to disable copy to sketch folder

tobozo, 2023-01-03:
does it have to be the sketch folder? 

looks like an additional entry in .gitignore files will be required for the meantime

[image]

me-no-dev, 2023-01-03:
@tobozo it has to be the sketch folder, because that is where the IDE is looking for `debug_custom.json` and is also the current folder for gdb/openocd (and where they can find the other files). Maybe this is a temporary thing and more flexibility will be added to the IDE in the future. I am looking into only doing it when debug is enabled also. I will add the files to gitignore in a bit.

tobozo, 2023-01-03:
@me-no-dev I've edited the platform.txt as per your recommendation.

If suggesting a different place is possible, it would certainly make sense to find those debug files in the same folder as the ELF file (near the build.options.json), where some debug material already exists.

me-no-dev, 2023-01-03:
@tobozo we have no control over that part or I would have used the build folder as well.
Anyway, I created [this PR](https://github.com/espressif/arduino-esp32/pull/7655) that adds gitignore and copies the files only if JTAG adapter is selected (or particular build variable is set to 1)

@IeiuniumLux @HomeSpan PTAL

tobozo, 2023-01-03:
can those debug tools eventually be chained with Arduino 1.8.x or is it only functional with IDE 2.0.3+ ?

me-no-dev, 2023-01-03:
@tobozo since they are using a particular extension that the new IDE includes, it can not be made to work with earlier versions. 2.0.x versions before 2.0.3 also have it, but it's older version and does not work correctly.

tobozo, 2023-01-03:
@me-no-dev thanks for clarifying

what about using `{runtime.ide.version}` to add precision?

I don't know what's the `{runtime.ide.version}` value for IDE 2.0.3+ but for Arduino-1.8.19 it's `10819` so in my situation this works:

`[[ {runtime.ide.version} -le 10819 || {build.copy_jtag_files} -eq 0 ]]`

tobozo, 2023-01-03:
oh crap, so "fixing" it will break something else?

looks like it's directly conflicting with [arduino-cli platform specs](https://arduino.github.io/arduino-cli/0.29/platform-specification/)

```
{runtime.ide.version}: the version number of the Arduino IDE as a number (this 
uses two digits per version number component, and removes the points and 
leading zeroes, so Arduino IDE 1.8.3 becomes 01.08.03 which becomes 
runtime.ide.version=10803). When using Arduino development software other 
than the Arduino IDE, this is set to a meaningless version number.

(....)

Compatibility note: Versions before Arduino IDE 1.6.0 only used one digit per 
version number component in {runtime.ide.version} (so 1.5.9 was 159, not 10509).
```

me-no-dev, 2023-01-03:
> what about using `{runtime.ide.version}` to add precision?

Would have been nice to be able to filter menu items this way as well, but... cant :)
Anyway, current solution will disable it for all boards that do not have menu item enabling it and only Espressif's boards do have it (default is off). Should make it fine for everyone. I doubt we need more precision than that and the new IDE should be the one we support from now on anyway. It's much better and offers more options and features.

closed 2023-01-06 by me-no-dev

tobozo, 2023-01-20:
@dlarue that would sure save [some github space](https://github.com/search?q=filename%3Aesp32.svd&type=code)

dlarue, 2023-01-20:
For example, this is how the STM32 devs allow Arduino IDE developers to control the build system:
[image]

me-no-dev, 2023-01-20:
We added a menu for selecting if JTAG is enabled or not. If not, it will not copy the files. It will come out soon in 2.0.7. Optimization flags are something else and should not mix them together.

dlarue, 2023-01-20:
if just doing JTAG or !JTAG then hopefully you're leveraging what Arduino IDE devs put in place on the Sketch menu with the "Optimize for Debugging" flag.

Yes, compiler optimizations are another item and even I can do that by editing a couple of files and backing them up for when they get replaced by uprades like 2.0.6 -> 2.0.7.