espressif/arduino-esp32 #7652: Multiple client devices to connect to the ESP32 micro server

CYE-GitENG, 2023-01-03:
### Related area

Wifi

### Hardware specification

ESP32-S3-DevkitC-1 v1.6

### Is your feature request related to a problem?

I have a question, if there are multiple client devices to connect to the ESP32 micro server, will there be an example?

### Describe the solution you'd like

multiple client devices to connect to the ESP32 micro server.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-01-03:
You need to explain what you mean.  What protocol are you talking about?  There is generally no need for an example, because most libraries/services will accept multiple clients without any issues.

CYE-GitENG, 2023-01-03:
Sorry, I may not express clearly, I want to use the HTTP communication protocol to set up a micro server on ESP32-S3, so that other clients can connect to ESP32-S3, because I can only connect with one client at a time, I want multiple clients to be able to connect to the ESP32-S3 micro server at the same time.

lbernstone, 2023-01-03:
WebServer can only service one client per loop cycle (server.handleClient()). If you use a small delay in that loop, you will be able to service a lot of clients.  If your application requires a lot of processing, such that the loop happens infrequently, then using [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) will allow you to service more clients. However, this is a low power device running at 80MHz, expecting it to give you performance comparable to a PC is unrealistic.

CYE-GitENG, 2023-01-03:
Thank you very much, this advice is very helpful for me.

closed 2023-01-04 by CYE-GitENG