espressif/arduino-esp32 #7653: At the same time, wifi mode and Ble mode can exist

savagerex, 2023-01-03:
### Board

ESP32 S3

### Device Description

ESP32-S3-DevKitC-1

### Hardware Configuration

NONE

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

At the same time, wifi mode and Ble mode can exist  ???

### Sketch

```cpp
NONE
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-01-11:
At a first glance, the answer would be YES, based on this document:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/coexist.html

SuGlider, 2023-01-11:
It seems to be enabled in the Arduino ESP32-S3 sdkconfig.
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32s3/sdkconfig#L1602

savagerex, 2023-01-11:
OK!!  Thank you!!!

closed 2023-01-27 by SuGlider