espressif/arduino-esp32 #7659: Improvements of Examples

VojtechBartoska, 2023-01-04:


dsyleixa, 2023-01-08:
just out of interest: what are the MT examples for?
Finally we have std:thread 
`std::thread thread_0 (myfunction);`

and std::mutex 
`std::mutex my_mutex;`

etc...?

dsyleixa, 2023-01-08:
just out of interest: what are the MT examples for?
Finally we have std:thread 
`std::thread thread_0 (myfunction);`

and std::mutex 
`std::mutex my_mutex;`

and so on, which are absolutely common...?

dsyleixa, 2023-01-08:
just out of interest: what are the MT examples for?
Finally we have std:thread 
`std::thread thread_0 (myfunction);`

and std::mutex 
`std::mutex my_mutex;`

 and so on, which are absolutely common providing an absolutely catchy syntax, plus even vTaskPrioritySet() (which admittedly is not quite POSIX-like though)...?
https://github.com/dsyleixa/Arduino/blob/master/ESP32_Test/std_thread_007/std_thread_007.ino
https://github.com/dsyleixa/Arduino/blob/master/ESP32_Test/esp32_stdmutex_002/esp32_stdmutex_002.ino

dsyleixa, 2023-01-08:
just out of interest: what are the hard-to-understand examples good for?
Finally we have std:thread 
`std::thread thread_0 (myfunction);`

and std::mutex 
`std::mutex my_mutex;`

 and so on, which are absolutely common providing an absolutely catchy syntax, plus even vTaskPrioritySet() (which admittedly is not quite POSIX-like though)...?
https://github.com/dsyleixa/Arduino/blob/master/ESP32_Test/std_thread_007/std_thread_007.ino
https://github.com/dsyleixa/Arduino/blob/master/ESP32_Test/esp32_stdmutex_003/esp32_stdmutex_003.ino

PilnyTomas, 2023-01-09:
@dsyleixa
My examples illustrate how FreeRTOS (on which all ESP32s run) does the multithreading natively.
Why use the existing and well-tested solution? I'll let that answer for you.

dsyleixa, 2023-01-09:
aren't std::thread and std::mutex well-tested too?

dsyleixa, 2023-01-09:
aren't std::thread and std::mutex well-tested too?
And finally this is the arduino-ESP32 section IIUC (please CMIIW)

dsyleixa, 2023-01-09:
aren't std::thread and std::mutex well-tested as well?
And finally this is the arduino-ESP32 section IIUC (please CMIIW)

(but BTW, some more std:: lib and POIX-like pthread examples to queuing, semaphores, and atomic would be highly appreciated, too!)

Xylopyrographer, 2023-01-10:
What does the green and orange shading in the [spreadsheet](https://docs.google.com/spreadsheets/d/1Opq2uVwrJL7QuXLr04ymHeXNzQ90rbh_Z76-8BiHdxk/edit?usp=sharing) represent?

VojtechBartoska, 2023-01-10:
@Xylopyrographer Green - examples are okay. Orange - examples need some action with higher priority.

PilnyTomas, 2023-01-13:
@dsyleixa 
> aren't std::thread and std::mutex well-tested as well?

Sure they are, but they are not supported by FreeRTOS. If you believe that implementing `std::thread` and `std::mutex` to work on FreeRTOS is an easy job we will welcome your PR!

> And finally this is the arduino-ESP32 section IIUC (please CMIIW)

Yes, you "UC" - what is your point? As it was pointed out elsewhere - Arduino ESP32 runs on ESP-IDF which runs on FreeRTOS - they are inseparable.

dsyleixa, 2023-01-13:
@PilnyTomas 
my point was that I wouldn't expect weird RTOS MT code in the arduino-esp32 section when std::thread etc are available, because of simplicity, convenience, and user-friendlyness.

dsyleixa, 2023-01-13:
@PilnyTomas 
my point was that I wouldn't expect weird RTOS MT code in the arduino-esp32 section when std::thread etc are available, because of simplicity, convenience, and user-friendlyness. 
And IIUC, std::thread is already powered by RTOS underneath, and they are already working like a charm as has been demonstrated here: https://github.com/espressif/arduino-esp32/issues/7659#issuecomment-1374835495

dsyleixa, 2023-01-13:
@PilnyTomas 
my point was that I wouldn't expect weird RTOS MT code in the arduino-esp32 section when std::thread etc are available, because of simplicity, convenience, and user-friendlyness. 
(So probably yout RTOS-MT will better fit to the esp32-IDF sction instead.)
And IIUC, std::thread is already powered by RTOS underneath, and they are already working like a charm as has been demonstrated here: https://github.com/espressif/arduino-esp32/issues/7659#issuecomment-1374835495

dsyleixa, 2023-01-13:
@PilnyTomas 
re: "but they are not supported by FreeRTOS"
I have to disagree, std::thread is already powered by RTOS underneath, and they are already working like a charm as has been demonstrated here: https://github.com/espressif/arduino-esp32/issues/7659#issuecomment-1374835495
Somy point was that I wouldn't expect weird RTOS MT code in the arduino-esp32 section when std::thread etc are available, because of simplicity, convenience, and user-friendlyness. 
(So probably your RTOS-MT code will better fit to the esp32-IDF section instead.)

dsyleixa, 2023-01-13:
@PilnyTomas 
re: "but they are not supported by FreeRTOS"
I have to disagree, std::thread is already powered by RTOS underneath, and they are already working like a charm as has been demonstrated here: https://github.com/espressif/arduino-esp32/issues/7659#issuecomment-1374835495
So my point was that I wouldn't expect RTOS MT code in the arduino-esp32 section when std::thread etc are available, because of simplicity, convenience, and user-friendlyness. 
(So probably your RTOS-MT code will better fit to the esp32-IDF section instead.)

dsyleixa, 2023-01-13:
@PilnyTomas 
re: "but they are not supported by FreeRTOS"
I have to disagree, std::thread is already powered by RTOS underneath, and they are already working like a charm as has been demonstrated here (no PR needed at all!): 
https://github.com/espressif/arduino-esp32/issues/7659#issuecomment-1374835495
So my point was that I wouldn't expect RTOS MT code in the arduino-esp32 section when std::thread etc are available, because of simplicity, convenience, and user-friendlyness. 
(So probably your RTOS-MT code will better fit to the low-level esp32-IDF section instead.)

PilnyTomas, 2023-01-13:
@dsyleixa If it is supported we will be happy to review your PR with examples.
You can follow the example of my pseudo library and create a new one named for example "pthread" and include your examples there.

dsyleixa, 2023-01-13:
@PilnyTomas 
I don't issue PRs, Ido not even know how that works, but my examples work as shown above, and if you want you may use them.

mathertel, 2023-05-29:
Please don't forget:

There is another example in https://github.com/espressif/arduino-esp32/pull/7687
* improving static files responses using a http eTag header
* extensive documentation

VojtechBartoska, 2024-05-15:
a lot of examples have been improved and we are continuously working on this. 

This ticket fulfil it purpose to do the investigation and start the improvements so I am closing it as done.

closed 2024-05-15 by VojtechBartoska