espressif/arduino-esp32 #7663: tcp_process error every 1-2 or 3 days.

zekageri, 2023-01-05:
### Board

ESP32-Wrover-E

### Device Description

I'm using ETH_LAN8720 Ethernet for networking.

### Hardware Configuration

Using SPI, Ext ram, I2C, Ethernet.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Every 1-2 or max 3 days i have a crash. 
My device is in a remote location so it was really hard to debug and get the crash log since ESP32 does not SUPPORT crash save to FLASH by default. It was just really a lucky time i got the crash log.

I'm using ESPAsyncWebServer and AsyncTCP.
My application runs flawlessly but crashes without ram loss and any notification.

### Sketch

```cpp
// Sketch uses ETH.h mostly.
// Crash happens in the following funcions:

int8_t AsyncServer::_s_accept(void * arg, tcp_pcb * pcb, int8_t err){
    return reinterpret_cast<AsyncServer*>(arg)->_accept(pcb, err);
}

//runs on LwIP thread
int8_t AsyncServer::_accept(tcp_pcb* pcb, int8_t err){
    //ets_printf("+A: 0x%08x\n", pcb);
    if(_connect_cb){
        AsyncClient *c = new AsyncClient(pcb);
        if(c){
            c->setNoDelay(_noDelay);
            return _tcp_accept(this, c);
        }
    }
    if(tcp_close(pcb) != ERR_OK){
        tcp_abort(pcb);
    }
    log_e("FAIL");
    return ERR_OK;
}

//Used to switch out from LwIP thread
static int8_t _tcp_accept(void * arg, AsyncClient * client) {
    lwip_event_packet_t * e = (lwip_event_packet_t *)malloc(sizeof(lwip_event_packet_t));
    if(!e) return ERR_MEM;
    e->event = LWIP_TCP_ACCEPT;
    e->arg = arg;
    e->accept.client = client;
    if (!_prepend_async_event(&e)) {
        free((void*)(e));
    }
    return ERR_OK;
}
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401a68d3  PS      : 0x00060230  A0      : 0x801a696b  A1      : 0x3ffb42f0  
A2      : 0x00000000  A3      : 0x3ffdfbec  A4      : 0x00000000  A5      : 0x3ffcbfc0  
A6      : 0x3ffcbfd8  A7      : 0x00060023  A8      : 0x8014499c  A9      : 0x3ffb42e0  
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000014  A13     : 0x0000ffff  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4008b014  LEND    : 0x4008b02a  LCOUNT  : 0xffffffff  

Backtrace: 0x401a68d0:0x3ffb42f0 0x401a6968:0x3ffb4320 0x40138675:0x3ffb4340 0x4013d6f2:0x3ffb43b0 0x40141766:0x3ffb43e0 0x4013220d:0x3ffb4400

  #0  0x401a68d0:0x3ffb42f0 in AsyncServer::_accept(tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1322
  #1  0x401a6968:0x3ffb4320 in AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1361
  #2  0x40138675:0x3ffb4340 in tcp_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:945
      (inlined by) tcp_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:438
  #3  0x4013d6f2:0x3ffb43b0 in ip4_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:800
  #4  0x40141766:0x3ffb43e0 in ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:186
  #5  0x4013220d:0x3ffb4400 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:180
      (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: 225259e45e21a87e

Rebooting...
ets Jul 29 2019 12:21:46
```

### Other Steps to Reproduce

I don't even know how i produced this.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

BlueAndi, 2023-01-05:
It reads a variable from a illegal address: 0x0000001c
The 0x0000001c sounds like accessing e.g. a structure, but the base address is NULL.

BlueAndi, 2023-01-05:
Can you add a NULL check in _s_accept() for the arg?

zekageri, 2023-01-15:
It turned out that i used the native AsyncTCP library with a tweaked version of AsyncWebServer. If i replace the AsyncTCP library with a tweaked one which was written for the tweaked AsyncWebserver it works fine. The problem is with the AsyncTCP and AsyncWebServer libraries. Since they are not maintained anymore i will close this issue. 

Thank you for all the help.

closed 2023-01-15 by zekageri