espressif/arduino-esp32 #7671: Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

meakashrao, 2023-01-06:
### Board

ESP32 WROOM 32 Module

### Device Description

ESP32 240MHz, 320KB RAM, 4MB Flash

### Hardware Configuration

No Hardware Interfaces

### Version

v2.0.6

### IDE Name

Platform IO

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying to create a dynamic timer using millis which runs different functions when timerperiod elapses

### Sketch

```cpp
My Source Files are:

1) Main.cpp
#include <Arduino.h>
#include "Timer.h"

Timer timer(7);

void setup() {
Serial.begin(9600);

timer.datumTime = millis();
timer.nextTime=(float)((millis() - timer.datumTime) / 1000L);
// Set the update functions for each variable
timer.updateFunctions[0] = (void*)updateWeight;
timer.updateFunctions[1] = (void*)updateMotor;
timer.updateFunctions[2] = (void*)updateTemperature;
timer.updateFunctions[3] = (void*)updateTimer;
timer.updateFunctions[4] = (void*)updateJar;
timer.updateFunctions[5] = (void*)updateMessage;
timer.updateFunctions[6] = (void*)updateStatus;

// Set the timeDeltaConditions values for each variable
timer.timeDeltaConditions[0] = timeDeltaWeight;
timer.timeDeltaConditions[1] = timeDeltaMotor;
timer.timeDeltaConditions[2] = timeDeltaTemperature;
timer.timeDeltaConditions[3] = timeDeltaTimer;
timer.timeDeltaConditions[4] = timeDeltaJar;
timer.timeDeltaConditions[5] = timeDeltaMessage;
timer.timeDeltaConditions[6] = timeDeltaStatus;

}

void loop() {

timer.updateStatusSentence();
}

2) Timer.cpp
#include "Timer.h"
#include "Arduino.h"

float timeDeltaWeight = 0.05;
float timeDeltaMotor = 0.5;
float timeDeltaTemperature = 1.0;
float timeDeltaTimer = 0.5;
float timeDeltaJar = 0.1;
float timeDeltaMessage = 1.0;
float timeDeltaStatus = 0.5;

void updateWeight();
void updateMotor();
void updateTemperature();
void updateTimer();
void updateJar();
void updateMessage();
void updateStatus();

Timer::Timer(int numVariables) {
this->numVariables = numVariables;

// Allocate memory for the updateCycleIteration, variables, and updateFunctions arrays
updateCycleIteration = new int[numVariables];
variables = new float[numVariables];
updateFunctions = new void*[numVariables];

for (int i = 0; i < numVariables; i++) {
updateCycleIteration = 1;
}
}

void Timer::updateStatusSentence()
{
currentTime = ((float)((millis() - datumTime) / 1L) / 1000.0);

timeDelta = abs(currentTime - nextTime);

Serial.print("cT:"); Serial.println(currentTime);
Serial.print("nT:"); Serial.println(nextTime);
Serial.print("TD: "); Serial.println(timeDelta);
float maxDelta = 0;

for (int i = 0; i < numVariables; i++) {
maxDelta = max(maxDelta, timeDeltaConditions);
}

for (int i = 0; i < numVariables; i++) {
updateStatusHelper(i);
}

if (timeDelta > maxDelta)
{
nextTime = currentTime;
for (int i = 0; i < numVariables; i++) {
updateCycleIteration = 1;
}
}
}

void Timer::updateStatusHelper(int variableIndex) {
if (timeDelta >= timeDeltaConditions[variableIndex] * updateCycleIteration[variableIndex]) {
void (*updateFunction)() = (void (*)())updateFunctions[variableIndex];
updateFunction();
updateCycleIteration[variableIndex]++;
}
}

void updateWeight(){
Serial.println("updateWeight");
}

void updateMotor(){
Serial.println("updateMotor");
}

void updateTemperature(){
Serial.println("updateTemperature");
}

void updateTimer(){
Serial.println("updateTimer");
}

void updateJar(){
Serial.println("updateJar");
}

void updateStatus(){
Serial.println("updateStatus");
}

void updateMessage(){
Serial.println("updateMessage");
}

3) Timer.h
#ifndef TIMERBASEDCLASS_H
#define TIMERBASEDCLASS_H

void updateWeight();
void updateMotor();
void updateTemperature();
void updateTimer();
void updateJar();
void updateMessage();
void updateStatus();

extern float timeDeltaWeight;
extern float timeDeltaMotor ;
extern float timeDeltaTemperature;
extern float timeDeltaTimer;
extern float timeDeltaJar;
extern float timeDeltaMessage;
extern float timeDeltaStatus;

class Timer
{
public:
float currentTime;
float nextTime;
float timeDelta;
unsigned long datumTime;
int *updateCycleIteration; // Dynamically allocated array to store iteration values for each variable
float *variables; // Dynamically allocated array to store the values of each variable
void **updateFunctions; // Dynamically allocated array of function pointers to store the update functions for each variable
int numVariables; // Number of variables and update functions to use
float *timeDeltaConditions;
Timer(int numVariables);
void updateStatusSentence();

private:
void updateStatusHelper(int variableIndex);
};

#endif // TIMERBASEDCLASS_H
```

### Debug Message

```plain
17:42:18.260 > Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.
17:42:18.347 >
17:42:18.350 > Core 1 register dump:
17:42:18.375 > PC : 0x400d1335 PS : 0x00060830 A0 : 0x800d26f9 A1 : 0x3ffb27f0
17:42:18.470 > A2 : 0x00000000 A3 : 0x00000000 A4 : 0x00000014 A5 : 0x00000004
17:42:18.563 > A6 : 0x3ffb8874 A7 : 0x80000001 A8 : 0x3ffbdb7c A9 : 0x400d1298
17:42:18.657 > A10 : 0x00000000 A11 : 0x00000000 A12 : 0x0800001c A13 : 0x00000003
17:42:18.751 > A14 : 0x00000001 A15 : 0x00000000 SAR : 0x0000000a EXCCAUSE: 0x0000001d
17:42:18.846 > EXCVADDR: 0x00000000 LBEG : 0x4008401d LEND : 0x40084025 LCOUNT : 0x00000027
17:42:18.938 >
17:42:18.940 >
17:42:18.942 > Backtrace:0x400d1332:0x3ffb27f00x400d26f6:0x3ffb2820
17:42:19.000 >
17:42:19.002 >
17:42:19.004 >
17:42:19.006 >
17:42:19.008 > ELF file SHA256: 0000000000000000
```

### Other Steps to Reproduce

NO

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-06:
Using "float" is going to be heavy/slow.  Recommend using "unsigned long", which is the output from millis().
Any time you dynamically cast a variable you should make it it's not null.  Specifically "updateFunction".
Use the "esp32_exception_decoder" to find out what line caused the crash in your code.  For PIO add `monitor_filters = time, colorize, esp32_exception_decoder` to your platformio.ini file.

meakashrao, 2023-01-07:
Thank you for the help, this helps me to an extent. 

Regards,
Akash Rao

closed 2023-01-07 by meakashrao