espressif/arduino-esp32 #7673: ledCwrite, nothing happens on the output.

Findus76, 2023-01-06:
### Board

ESP-WROOM-32, custom board

### Device Description

Custom board.
4 pcs pwm outputs, 0-10V out, digital in, analog in, I2C, onewire.

### Hardware Configuration

PWM connected to base of fet transistor.
0-10V, averaged in files and amplified with OP amplifier.

### Version

v2.0.6

### IDE Name

Platformio IO

### Operating System

Windows7

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

The unit has been running flawlessly for almost 2 years. Upgraded the kernel first to 2.0.5 then to 2.0.6. Since then the PWM and 0-10V outputs do not work. Have measured with an oscilloscope, the outputs are constantly at 0V.
Have tried
-To read back the status of the pins if they are still output, they are.
-To read back (ledcread) which reports the previous set value... Ie start value 0 I set 100 and I read back 0, I write new value 60 and I read back 100.
-To put several ledcwrite after each other. Does not make a difference on the pin, however, it can read back the data correctly. Not always but sometimes.
-Moving initiation of ledcsetup and ledcattatch from constructor to a begin function no difference.
- Replaced ledcwrite with digitaWrite. Now I can change the pin state.

### Sketch

```cpp
Part of constructor
NodeFunction::NodeFunction() :oneWire(OneWireBUS)//,I2Cdevice(),Sensor_BME280()
{
	sensors.setOneWire(&oneWire);
	LastMillisTime = millis();
	
	pinMode(InputCh1, INPUT);
	pinMode(InputCh2, INPUT);
	pinMode(InputCh3, INPUT);
	pinMode(InputCh4, INPUT);
	ledcSetup(PWMCh1Channel, PWMfreq, PWMresolution);
	ledcAttachPin(PWMCh1, PWMCh1Channel);
	pinMode(PWMCh1, OUTPUT);
	ledcSetup(PWMCh2Channel, PWMfreq, PWMresolution);
	ledcAttachPin(PWMCh2, PWMCh2Channel);
	pinMode(PWMCh2, OUTPUT);
	ledcSetup(PWMCh3Channel, PWMfreq, PWMresolution);
	ledcAttachPin(PWMCh3, PWMCh3Channel);
	pinMode(PWMCh3, OUTPUT);
	ledcSetup(PWMCh4Channel, PWMfreq, PWMresolution);
	ledcAttachPin(PWMCh4, PWMCh4Channel);
	pinMode(PWMCh4, OUTPUT);
	ledcSetup(AnalogOutCh1Channel, PWMfreq, PWMresolution);
	ledcAttachPin(AnalogOutCh1, AnalogOutCh1Channel);
	pinMode(AnalogOutCh1, OUTPUT);
	ledcSetup(AnalogOutCh2Channel, PWMfreq, PWMresolution);
	ledcAttachPin(AnalogOutCh2, AnalogOutCh2Channel);
	pinMode(AnalogOutCh2, OUTPUT);

	LEDValue = 0;
	pinMode(OnLED, OUTPUT);
	PushButtonCh1 = false;

Part of settings of pin
if(ValueParam >= 0 && ValueParam <= 100)
			{
				AnalogOutCh1_value = ValueParam;
				ledcWrite(AnalogOutCh1Channel, CalculatRegisterValue(AnalogOutCh1_value));
				ledcWrite(AnalogOutCh1Channel, CalculatRegisterValue(AnalogOutCh1_value));
				ledcWrite(AnalogOutCh1Channel, CalculatRegisterValue(AnalogOutCh1_value));
				ledcWrite(AnalogOutCh1Channel, CalculatRegisterValue(AnalogOutCh1_value));
				Serial.printf("Incomming set value: %d, Readback LedCValue: %d, LedC freq: %d\n\r", CalculatRegisterValue(AnalogOutCh1_value), ledcRead(AnalogOutCh1Channel), ledcReadFreq(AnalogOutCh1Channel));
				getPinMode(InputCh1);
				getPinMode(InputCh2);
				getPinMode(InputCh3);
				getPinMode(InputCh4);
				getPinMode(PWMCh1);
				getPinMode(PWMCh2);
				getPinMode(PWMCh3);
				getPinMode(PWMCh4);
				getPinMode(AnalogOutCh1);
				getPinMode(AnalogOutCh2);
			}
```

### Debug Message

```plain
---------  IDE log ---------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ArduinoJson @ 6.20.0
|-- WiFi @ 2.0.0
|-- OneWire @ 2.3.7

---------  Board log ---------

Incomming SET JSON type
Incomming set value: 153, Readback LedCValue: 0, LedC freq: 0
Pin 32 is Input
Pin 33 is Input
Pin 34 is Input
Pin 35 is Input
Pin 14 is Output
Pin 18 is Output
Pin 19 is Output
Pin 21 is Output
Pin 12 is Output
Pin 13 is Output
Send Value list over MQTT
Value send:{"Type":"Value list","Function list":{"Analog out":{"Ch1":{"Value":15}}}}
JSON data
Filesize: 644
Check Ok
Incomming SET JSON type
Incomming set value: 317, Readback LedCValue: 153, LedC freq: 5000
Pin 32 is Input
Pin 33 is Input
Pin 34 is Input
Pin 35 is Input
Pin 14 is Output
Pin 18 is Output
Pin 19 is Output
Pin 21 is Output
Pin 12 is Output
Pin 13 is Output
Send Value list over MQTT
Value send:{"Type":"Value list","Function list":{"Analog out":{"Ch1":{"Value":31}}}}
JSON data
Filesize: 644
Check Ok
Incomming SET JSON type
Incomming set value: 491, Readback LedCValue: 317, LedC freq: 5000
Pin 32 is Input
Pin 33 is Input
Pin 34 is Input
Pin 35 is Input
Pin 14 is Output
Pin 18 is Output
Pin 19 is Output
Pin 21 is Output
Pin 12 is Output
Pin 13 is Output
Send Value list over MQTT
Value send:{"Type":"Value list","Function list":{"Analog out":{"Ch1":{"Value":48}}}}
Read sensor
Temp: 26.950001
Pressure: 101922.492188
Humidity: 24.541016, Old: 24.713867
JSON data
Filesize: 644
Check Ok
Incomming SET JSON type
Incomming set value: 634, Readback LedCValue: 491, LedC freq: 5000
Pin 32 is Input
Pin 33 is Input
Pin 34 is Input
Pin 35 is Input
Pin 14 is Output
Pin 18 is Output
Pin 19 is Output
Pin 21 is Output
Pin 12 is Output
Pin 13 is Output
Send Value list over MQTT
Value send:{"Type":"Value list","Function list":{"Analog out":{"Ch1":{"Value":62}}}}
JSON data
Filesize: 644
Check Ok
Incomming SET JSON type
Incomming set value: 398, Readback LedCValue: 634, LedC freq: 5000
Pin 32 is Input
Pin 33 is Input
Pin 34 is Input
Pin 35 is Input
Pin 14 is Output
Pin 18 is Output
Pin 19 is Output
Pin 21 is Output
Pin 12 is Output
Pin 13 is Output
Send Value list over MQTT
```

### Other Steps to Reproduce

case #5306 feels similar.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-01-06:
ledcAttachPin is going to set the pin mode (it is going to use the ledc peripheral, rather than digital out).  Don't set pinMode after you attach it.

Findus76, 2023-01-07:
Many thanks. That was the mistake. Now it works.
I was unsure when I created the code about 2 years ago, it became a bit of "double security" and has worked.

closed 2023-01-07 by Findus76

lbernstone, 2023-01-07:
Undefined behavior sometimes works, but is subject to change when the APIs change.  Following the [examples in the repo](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/ledcWrite_RGB/ledcWrite_RGB.ino) should give you better results than random (old) internet code...