espressif/arduino-esp32 #7675: BluetoothSerial: register_callback compile error

paulhamsh, 2023-01-08:
### Board

ESP32 Dev Module

### Device Description

Not relevant because not uploading - compile issue

### Hardware Configuration

N/A

### Version

v2.0.6

### IDE Name

Arduino 2.0.3

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I get this error in v2.0.6 - but not in v2..0.5 or any version below that.
What is really weird is that I can't see any difference in the src files for register_callback.
```
C:\Users\Paul\Desktop\sketch_jan7a\sketch_jan7a.ino: In function 'void setup()':
C:\Users\Paul\Desktop\sketch_jan7a\sketch_jan7a.ino:24:38: error: no matching function for call to 'BluetoothSerial::register_callback(void (&)(esp_spp_cb_event_t, esp_spp_cb_param_t*))'
   SerialBT.register_callback(callback);
                                      ^
In file included from C:\Users\Paul\Desktop\sketch_jan7a\sketch_jan7a.ino:7:
C:\Users\Paul\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BluetoothSerial\src/BluetoothSerial.h:56:19: note: candidate: 'esp_err_t BluetoothSerial::register_callback(void (**)(esp_spp_cb_event_t, esp_spp_cb_param_t*))'
         esp_err_t register_callback(esp_spp_cb_t * callback);
                   ^~~~~~~~~~~~~~~~~
C:\Users\Paul\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BluetoothSerial\src/BluetoothSerial.h:56:19: note:   no known conversion for argument 1 from 'void(esp_spp_cb_event_t, esp_spp_cb_param_t*)' to 'void (**)(esp_spp_cb_event_t, esp_spp_cb_param_t*)'

exit status 1

Compilation error: no matching function for call to 'BluetoothSerial::register_callback(void (&)(esp_spp_cb_event_t, esp_spp_cb_param_t*))'
```

### Sketch

```cpp
// Works in 2.0.2
// Works in 2.0.3
// Works in 2.0.4
// Works in 2.0.5
// Doesn't work in 2.0.6

#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;
 
void callback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param){
  if(event == ESP_SPP_SRV_OPEN_EVT){
    Serial.println("Client Connected");
  }
 
  if(event == ESP_SPP_CLOSE_EVT ){
    Serial.println("Client disconnected");
  }
}
 
void setup() {
  Serial.begin(115200);
 
  SerialBT.register_callback(callback);
 
  if(!SerialBT.begin("ESP32")){
    Serial.println("An error occurred initializing Bluetooth");
  }else{
    Serial.println("Bluetooth initialized");
  }
}
 
void loop() {}
```
```

### Debug Message

```plain
N/A - compile error
```

### Other Steps to Reproduce

Just load sketch and try to complile it

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

fheday, 2023-01-09:
I can confirm I am having the same problem after an update of platformIO.
I also had no change in the source code.

fheday, 2023-01-09:
actjally I checked the code works with espressif32@5.2.0, but not with espressif32@5.3.0.

It seems that the arduino component does not works with the latest version.

me-no-dev, 2023-01-09:
the signature of the callback has changed in ESP-IDF. Linked above is a PR that fixes this on our side :)

closed 2023-01-12 by me-no-dev