espressif/arduino-esp32 #7677: UDP handling leads to Kernel Panic in release 2.0.6

Deadeye5589, 2023-01-08:
### Board

ESP-WROOM-32

### Device Description

ESP-WROOM-32 configured as ESP32 Dev Module in Arduino IDE

### Hardware Configuration

No external components

### Version

v2.0.6

### IDE Name

Arduino IDE 2.0.3

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Updated the ESP32 Board Manager from 1.0.6 to 2.0.6.
Now my previously working sketch crashes with a Kernel Panic. After some debugging I was able to narrow down the problem to be located in the used NTP functions and more specific somewhere in the UPD handling, hence I believe the problem should be handled here and not in the NTP library. 

Here is the basic NTP programm I have used for my ESP32 programm.
https://usercontent.one/wp/blog.thesen.eu/wp-content/uploads/2017/08/ESP_NTP_LocalTime_Demo.zip?media=1672860717
If you try to compile it with the 2.0.6 release it should crash. Please be aware that this example was intended to be used with ESP8266 processors. Therefore the header files must be adjusted accordingly to match the ESP32. With release 1.0.6 this code is working for me and has been implemented sucessfully in my own projects. 

If have added some Serial prints to his code to narrow down at what line of code the Kernal Panic happens

### Sketch

```cpp
unsigned long getNTPTimestamp()
{
  unsigned long ulSecs2000;
  
  Serial.println("NTP UDP Start");
  udp.begin(ntpPort);
  Serial.println("NTP Send package");
  sendNTPpacket(timeServer); // send an NTP packet to a time server
  Serial.println("Wait for response");
  delay(1000);    // wait to see if a reply is available
  int cb=0, repeat=0; 
  while(!cb && repeat<20)  // try for 2 sec
  {
    cb = udp.parsePacket();
    delay(100);
    repeat++;
  }

  if(!cb)
  {
    ulSecs2000=0;
  }
  else
  {
    udp.read(packetBuffer, NTP_PACKET_SIZE); // read the packet into the buffer
    
    //the timestamp starts at byte 40 of the received packet and is four bytes,
    // or two words, long. First, esxtract the two words:
    unsigned long highWord = word(packetBuffer[40], packetBuffer[41]);
    unsigned long lowWord = word(packetBuffer[42], packetBuffer[43]);
    
    // combine the four bytes (two words) into a long integer
    // this is NTP time (seconds since Jan 1 1900):
    ulSecs2000  = highWord << 16 | lowWord;
    ulSecs2000 -= 2208988800UL; // go from 1900 to 1970
    ulSecs2000 -= 946684800UL; // go from 1970 to 2000
  }    
  return(ulSecs2000);
}

// send an NTP request to the time server at the given address
unsigned long sendNTPpacket(IPAddress& address)
{
  // set all bytes in the buffer to 0
  memset(packetBuffer, 0, NTP_PACKET_SIZE);
  // Initialize values needed to form NTP request
  // (see URL above for details on the packets)
  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
  packetBuffer[1] = 0;     // Stratum, or type of clock
  packetBuffer[2] = 6;     // Polling Interval
  packetBuffer[3] = 0xEC;  // Peer Clock Precision
  
  // 8 bytes of zero for Root Delay & Root Dispersion
  packetBuffer[12]  = 49;
  packetBuffer[13]  = 0x4E;
  packetBuffer[14]  = 49;
  packetBuffer[15]  = 52;

  Serial.println("Buffers filled");
  // all NTP fields have been given values, now
  // you can send a packet requesting a timestamp:
  udp.beginPacket(address, 123); //NTP requests are to port 123
  Serial.println("UDP Begin Packet");
  udp.write(packetBuffer, NTP_PACKET_SIZE);
  Serial.println("UDP Send package");
  udp.endPacket();
  Serial.println("UDP finished");
}
```

### Debug Message

```plain
Setup sync with NTP service. (This is located inside the main program)
NTP UDP Start (From here on the added lines of debug messages to the given program)
NTP Send package
Buffers filled
UDP Begin Packet
UDP Send package
UDP finished

//dozenz more loops of the same serial messages here

NTP UDP Start
NTP Send package
Buffers filled
UDP Begin Packet
UDP Send package
UDP finished
NTP UDP Start
NTP Send package
Buffers filled
UDP Begin Packet
UDP Send package
UDP finished
NTP UDP Start
NTP Send package
Buffers filled
UDP Begin Packet
UDP Send package
UDP finished
NTP UDP Start
NTP Send package
Buffers filled
UDP Begin Packet
UDP Send package
UDP finished
NTP UDP Start
NTP Send package
Buffers filled
UDP Begin Packet
UDP Send package
UDP finished
NTP UDP Start
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x4009065b  PS      : 0x00060b36  A0      : 0x8008fcc3  A1      : 0x3ffb08c0  
A2      : 0x3ffbf5d8  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060b23  
A6      : 0x00060b23  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00060b23  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x00060d23  
A14     : 0x007bf5d8  A15     : 0x003fffff  SAR     : 0x0000001c  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40089860  LEND    : 0x40089876  LCOUNT  : 0xffffffff  

Backtrace: 0x40090658:0x3ffb08c0 0x4008fcc0:0x3ffb0900 0x4008da5a:0x3ffb0920 0x4008df36:0x3ffb0940 0x4008e0a7:0x3ffb0980 0x400839c9:0x3ffb09a0 0x4016302e:0x3ffb09c0 0x4013f004:0x3ffb09e0 0x4013c0fd:0x3ffb0a00 0x400f8dc3:0x3ffb0a20 0x4015e59d:0x3ffb0a40 0x4010b220:0x3ffb0a60 0x400fd9c2:0x3ffb0a80 0x4010dddd:0x3ffb0aa0 0x4010d359:0x3ffb0ac0 0x4010d2c0:0x3ffb0af0 0x4010c074:0x3ffb0b10 0x400f9b97:0x3ffb0b30 0x400f9bf3:0x3ffb0b50 0x400fb055:0x3ffb0b70 0x400f1359:0x3ffb0bc0 0x4000bd4b:0x3ffb0be0 0x40001781:0x3ffb0c00 0x400d5e07:0x3ffb0c20 0x400d5e83:0x3ffb0c50 0x400d5f29:0x3ffb0c90 0x400d347f:0x3ffb0cc0 0x400d3465:0x3ffb0ce0 0x400d348e:0x3ffb0d10 0x400d3465:0x3ffb0d30 0x400d348e:0x3ffb0d60 0x400d3465:0x3ffb0d80 0x400d348e:0x3ffb0db0 0x400d3465:0x3ffb0dd0 0x400d348e:0x3ffb0e00 0x400d3465:0x3ffb0e20 0x400d348e:0x3ffb0e50 0x400d3465:0x3ffb0e70 0x400d348e:0x3ffb0ea0 0x400d3465:0x3ffb0ec0 0x400d348e:0x3ffb0ef0 0x400d3465:0x3ffb0f10 0x400d348e:0x3ffb0f40 0x400d3465:0x3ffb0f60 0x400d348e:0x3ffb0f90 0x400d3465:0x3ffb0fb0 0x400d348e:0x3ffb0fe0 0x400d3465:0x3ffb1000 0x400d348e:0x3ffb1030 0x400d3465:0x3ffb1050 0x400d348e:0x3ffb1080 0x400d3465:0x3ffb10a0 0x400d348e:0x3ffb10d0 0x400d3465:0x3ffb10f0 0x400d348e:0x3ffb1120 0x400d3465:0x3ffb1140 0x400d348e:0x3ffb1170 0x400d3465:0x3ffb1190 0x400d348e:0x3ffb11c0 0x400d3465:0x3ffb11e0 0x400d348e:0x3ffb1210 0x400d3465:0x3ffb1230 0x400d348e:0x3ffb1260 0x400d3465:0x3ffb1280 0x400d348e:0x3ffb12b0 0x400d3465:0x3ffb12d0 0x400d348e:0x3ffb1300 0x400d3465:0x3ffb1320 0x400d348e:0x3ffb1350 0x400d3465:0x3ffb1370 0x400d348e:0x3ffb13a0 0x400d3465:0x3ffb13c0 0x400d348e:0x3ffb13f0 0x400d3465:0x3ffb1410 0x400d348e:0x3ffb1440 0x400d3465:0x3ffb1460 0x400d348e:0x3ffb1490 0x400d3465:0x3ffb14b0 0x400d348e:0x3ffb14e0 0x400d3465:0x3ffb1500 0x400d348e:0x3ffb1530 0x400d3465:0x3ffb1550 0x400d348e:0x3ffb1580 0x400d3465:0x3ffb15a0 0x400d348e:0x3ffb15d0 0x400d3465:0x3ffb15f0 0x400d348e:0x3ffb1620 0x400d3465:0x3ffb1640 0x400d348e:0x3ffb1670 0x400d3465:0x3ffb1690 0x400d348e:0x3ffb16c0 0x400d3465:0x3ffb16e0 0x400d348e:0x3ffb1710 0x400d3465:0x3ffb1730 0x400d348e:0x3ffb1760 0x400d3465:0x3ffb1780 0x400d348e:0x3ffb17b0 0x400d3465:0x3ffb17d0 0x400d348e:0x3ffb1800 0x400d3465:0x3ffb1820 0x400d348e:0x3ffb1850 |<-CONTINUES

ELF file SHA256: 5721aa3b4c8d7313

Rebooting...
```

### Other Steps to Reproduce

As you can see, instead of running through the code once, for some reason at the end of the sendNTPpacket function call the programm does not return to the correct line of code from the outer function getNTPTimestamp, but instead reenters the getNTPTimestamp function from the beginning.

I have tried to add a yield() instruction after the  udp.endPacket() function call within the sendNTPpacket function. In this case all of the serial print debugg messages will only be printed out once but still the kernel panic happens without ever reaching the Serial.println("Wait for response"); instruction within the getNTPTimestamp function. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-01-09:
You are running out of stack space in code that you haven't provided.  Most likely something about not closing your udp object.  There is not any good reason to persist a udp session through the whole program, keep it local.
UDP and NTP work well for me. https://github.com/lbernstone/rtc_calibration/blob/main/rtc_calib.ino

Deadeye5589, 2023-01-09:
Hello lbernstone,

thank you for your reply. The complete code was provided in the link of the description. Nevertheless your hint towards stack space was the right solution. By relocating the UDP object into the function it is possible to compile this software with ESP32 librarys greater than 1.0.6 successfully again. Still there seems to be a much higher stack usage with the ESP32 lib starting at version 2.0.0. Otherwise I would not see why the program should suddenly run out of stack where it was fine before.

closed 2023-01-11 by Deadeye5589