espressif/arduino-esp32 #7679: Arduino Freezes After Flash

neo305, 2023-01-08:
### Board

all boards ESP32-S2 and ESP8266

### Device Description

FeatherS2

### Hardware Configuration

No

### Version

v2.0.2

### IDE Name

Arduino 1.8.19

### Operating System

Windows 11

### Flash frequency

240Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

after every flash Arduino freezes and I have to open task manager and force Java to close so Arduino will shutdown and close. and I basically have everything mirrored on my laptop and don't have this problem on it.

### Sketch

```cpp
Arduino: 1.8.19 (Windows 10), Board: "UM FeatherS2, Enabled, Disabled, Disabled, Enabled, 16M Flash (2MB APP/12.5MB FATFS), 240MHz (WiFi), 16MB (128Mb), 921600, None, Enabled"

Sketch uses 880626 bytes (41%) of program storage space. Maximum is 2097152 bytes.
Global variables use 46652 bytes (14%) of dynamic memory, leaving 281028 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3.2
Serial port COM3
Connecting...
Chip is ESP32-S2
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:73:16:12
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 37.1s
Compressed 14592 bytes to 10150...
Writing at 0x00001000... (20 %)
Writing at 0x00001c5a... (40 %)
Writing at 0x0000279d... (60 %)
Writing at 0x000032ac... (80 %)
Writing at 0x00003c5b... (100 %)
Wrote 14592 bytes (10150 compressed) at 0x00001000 in 0.3 seconds (effective 393.3 kbit/s)...
Hash of data verified.
[... 355 lines cut ...]
->Writing at 0x00cb7f17... (99 %)
->Writing at 0x00ebbf17... (100 %)
->Wrote 12513280 bytes (2733338 compressed) at 0x00411000 in 80.0 seconds (effective 1250.7 kbit/s)...
->Hash of data verified.
->
->Leaving...
->WARNING: ESP32-S2 chip was placed into download mode using GPIO0.
->esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
->To suppress this note, set --after option to 'no_reset'.
FatFS Upload failed!

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### Debug Message

```plain
is there somewhere else I can get it from? cuz Arduino is frozen and it won't let me copy the log from within it.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-01-08:
v2.0.2 is not supported anymore. Please try actual v2.0.6

lbernstone, 2023-01-08:
This looks to be an error in the Arduino IDE, not arduino-esp32.  I would recommend you open an issue at [their site](https://github.com/arduino/arduino-ide/issues).  They are going to want you to run arduino from the command line- I don't have an exact command for you to do that in Windows, it is `~/arduino-1.8.19/arduino` in linux.  I would assume there is a batch file in there for windows.

closed 2023-04-19 by Jason2866