espressif/arduino-esp32 #7689: Slower ADC Sampling rate when having firmware version 1.0.5 or 1.0.6 compared to version 1.0.4

AkthemRehab, 2023-01-09:
### Board

ESP32 Dev Module

### Device Description

ESP WROOM 32

### Hardware Configuration

GPIO 35 is connected to lm35 temperature sensor.

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The sampling rate from any ADC1 pin is much less than expected. The sampling rate acheived is 105KHz when using analogread and version 1.0.4 of the ESP32 firmware installed from arduino IDE board manager.

When version 1.0.6 or 1.0.5 is used the sampling rate drops to ~18.5KHz.

Refer to this video and comments for better explanation of the issue and workaround solution: https://www.youtube.com/watch?v=Mgh2WblO5_c&list=PLnPcJ7PLgOmThiBQlR8dsZcq9BQcYF3k3&index=1&t=188s

I am afraid that, with time, version 1.0.4 is deprecated and we get stuck with the slower sampling rate. Could you place rectify this in future versions?

### Sketch

```cpp
#define Temp_In_Pin 35

int analogValue;
unsigned long newTime;

void setup() {
  Serial.begin(115200);
}

void loop() {
  newTime = micros();

  // Do 1 million reads and record time taken
  for(int i = 0; i < 1000000; i++) {
    analogValue = analogRead(Temp_In_Pin);
  }
  
  float conversionTime = (micros() - newTime) / 1000000.0;

  Serial.print("Conversion time: ");
  Serial.print(conversionTime);
  Serial.println(" uS");

  Serial.print("Max sampling frequenc: ");
  Serial.print((1.0 / conversionTime) * 1000000);
  Serial.println(" Hz");
}
```

### Debug Message

```plain
No errors returned. Issue is with slower sampling rate.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-09:
Hi @AkthemRehab.  arduino-esp32 v1.x.x has been unsupported for some time.  Please use 2.0.6.

AkthemRehab, 2023-01-09:
> **[mrengineer7777](/mrengineer7777) ** c

Hi @mrengineer7777 , Actually I selected other because I did not know what the Version question was referring to. If it is referring to the esp32 versions in the boards manager, then I can only see up to 1.0.6 and the issue I am facing is with versions 1.0.5 and 1.0.6 contrary to 1.0.4 which is working with much higher sampling rate.

mrengineer7777, 2023-01-09:
@AkthemRehab Unfortunately I can't help with the ArduinoIDE as I switched to VSCode with PlatformIO.  However from other issue reports I'm sure the 2.x.x versions are available in the ArduinoIDE.  I hear there's a newer version of the IDE.  You might have to use that one.

P-R-O-C-H-Y, 2023-01-10:
Hi @AkthemRehab, if you don't see 2.0.X versions in the Arduino IDE, please go through the [Installation Guide](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide), for sure you are using old URL for esp32-core inside Arduino IDE preferences. After updating the URL, you should be able to download newest versions from board manager.

AkthemRehab, 2023-01-10:
> Hi @AkthemRehab, if you don't see 2.0.X versions in the Arduino IDE, please go through the [Installation [...]

Hi @P-R-O-C-H-Y, you are absolutely correct. I changed the url and found the latest 2.0.6 version of Arduino-esp32. Thank you!

AkthemRehab, 2023-01-10:
I installed version 2.0.6 and tested the code again. The sampling rate dropped even more than versions 1.0.5 and 1.0.6! Now the sampling rate is ~11,270 Hz.

lbernstone, 2023-01-10:
ADC has been changed to use the IDF API, rather than directly writing to registers.  This is necessary to support the newer variants, so it is not going back.  However, the code is open source, and you can certainly take the [code from 1.0.4](https://github.com/espressif/arduino-esp32/blob/1.0.4/cores/esp32/esp32-hal-adc.c) and incorporate it into your programs to use the older functions on an ESP32.

P-R-O-C-H-Y, 2023-01-17:
Hi @AkthemRehab, it is as @lbernstone commented. In order to have support for never chips, we changed a lot peripherals to use ESP-IDF APIs. So I'm closing this and adding label "won't fix", as there is nothing to do about.

AkthemRehab, 2023-01-17:
Thank you @P-R-O-C-H-Y , @lbernstone , and @mrengineer7777 for your help and replies. I am working with version 1.0.4 at the moment to utilize the faster sampling rate possible with that version. Perhaps, at a later stage I will work with @lbernstone proposition of using 1.0.4 adc code in the more recent versions (2.0.6 and beyond).