espressif/arduino-esp32 #7690: TinyS3 BLE does not stay paired

anthony-simetri, 2023-01-09:
### Board

UM TinyS3

### Device Description

Regular board

### Hardware Configuration

Nothing additional

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10, Android 12

### Flash frequency

?

### PSRAM enabled

yes

### Upload speed

921600

### Description

Pairing with an Android tablet, the S3 attempts to pair and, shows a connection was made from the BLEServerCallbacks instance before pressing the Pair button on the Android side, and then shows up as paired for a moment before showing it's not paired again and going back into the list of unpaired devices.

Pairing on Windows, the S3 is showing a series of connect and disconnect messages from the callback instance. While it stays in the pairing list, it never successfully pairs.

### Sketch

```cpp
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;

#define SERVICE_UUID        "b0738b60-bf42-11ec-9d64-0242ac120002"
#define CHARACTERISTIC_UUID "6c31faac-c269-11ec-9d64-0242ac120002"

class MyServerCallbacks : public BLEServerCallbacks
{
    public:
        void onConnect(BLEServer* pServer)
        {
            Serial.println("New client connected!");
        }

        void onDisconnect(BLEServer* pServer)
        {
            Serial.println("Client disconnected");
        }
};

void setup()
{
    Serial.begin(115200);

    // Create the BLE Device
    Serial.println("Starting Arduino BLE Server application...");
    BLEDevice::init("S3");

    // Create the BLE Server
    pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyServerCallbacks());

    // Create the BLE Service
    BLEService *pService = pServer->createService(SERVICE_UUID);

    // Create a BLE Characteristic
    pCharacteristic = pService->createCharacteristic(
                          CHARACTERISTIC_UUID,
                          BLECharacteristic::PROPERTY_READ   |
                          BLECharacteristic::PROPERTY_NOTIFY
                      );

    // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
    // Create a BLE Descriptor for configuring notify
    pCharacteristic->addDescriptor(new BLE2902());

    // Start the service
    pService->start();

    // Start advertising
    BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
    pAdvertising->addServiceUUID(SERVICE_UUID);
    pAdvertising->setScanResponse(false);
    pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter
    BLEDevice::startAdvertising();
    Serial.println("Waiting for a client connection");
}

void loop()
{

}
```

### Debug Message

```plain
No debug message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-01-11:
There is no need to pair BLE for this example.
Try using the [nRF Connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&gl=US&pli=1) in your Android Tablet and then scan for the "S3" BLE device.
Press `connect` and read the characteristic created in the example code.

anthony-simetri, 2023-01-11:
> There is no need to pair BLE for this example. Try using the [nRF Connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&gl=US&pli=1) in your Android Tablet and then scan for the "S3" BLE device. Press `connect` and read the characteristic created in the example code.

The reason we are using pairing is to be able to select among identical devices which one should be connected to in an app as part of requirements.
Using a TinyS3, it does show that it connects using nRF Connect app. I've paired with two different kinds of Android devices and Windows and different TinyS3 boards but nothing seems to work. It's usually a connect and then it will eventually says disconnect after giving up on the pairing process. Idk if there is some subtle difference between the ESP32-S3 and TinyS3. I do not have a regular ESP32-S3 to test with.

SuGlider, 2023-01-12:
In the nRF Connect App, there is a TAB saying "Bonded". This is the list of paired devices.
ESP32-S3 uses BLE 5.0
ESP32 uses BLE 4.2

Could this be an issue with the version of the Android and BLE?
I can see that older Android devices work fine with ESP32 and BLE 4.

BLE pairing shall be just a process for authentication (which is not used in the example sketch) and adding it to the bonded device list.

anthony-simetri, 2023-01-12:
> In the nRF Connect App, there is a TAB saying "Bonded". This is the list of paired devices. ESP32-S3 uses BLE 5.0 ESP32 uses BLE 4.2
>
> Could this be an issue with the version of the Android and BLE? [...]

When I bond in app, it does say "BONDING..." for a moment before giving up, but nothing ever appears in the Bonded list. The original TinyPICO has no issue at all with its BLE 4.2. I tried with an Android 7.0 device, but it does not succeed either. Granted, that's fairly on the old end by this point.

closed 2025-01-16 by Parsaabasi