espressif/arduino-esp32 #7692: taking long time for ping response

sushilf, 2023-01-10:
### Board

ESP32-WROVER-B(M213DH3264PH3Q0)

### Device Description

We're using ESP32-WROVER-B(M213DH3264PH3Q0) module in one of our project named Dosecar which is smart battery operated robotic car moving around our constrained environment. 

### Hardware Configuration

[esp_32.zip](https://github.com/espressif/arduino-esp32/files/10381264/esp_32.zip)
[image]

we are using one uart port of esp32 for communication with stm32 mcu. 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

linux ubuntu 20.04

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

We're facing issue of high packet loss/telnet issue which is not replicable but occurs randomly. We're not able to find the cause. We're using Arduino IDE for firmware development. 

### Sketch

```cpp
there is no specific portion which replicate the issue of esp32. we attach the whole code in hardware configuration block.
```

### Debug Message

```plain
No error message.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-01-10:
Try `WiFi.setSleep(false);`

sushilf, 2023-02-02:
i tried this one. but it is not working for me.

sushilf, 2023-02-02:
i added following three lines before WiFi.begin().

1) WiFi.mode (WIFI_STA);
2)esp_wifi_set_ps(WIFI_PS_NONE);
3)WiFi.setSleep(false);

closed 2025-01-16 by Parsaabasi