espressif/arduino-esp32 #7700: LittleFS debug messages polluting Serial(0).

everslick, 2023-01-11:
### Board

n.a.

### Device Description

n.a.

### Hardware Configuration

n.a.

### Version

latest master (checkout manually)

### IDE Name

n.a.

### Operating System

n.a.

### Flash frequency

n.a.

### PSRAM enabled

yes

### Upload speed

n.a.

### Description

It seems as if LittleFS is compiled with debug macros enabled so it will pollute the serial output of any program. I do not think it can be switched off at runtime.

### Sketch

```cpp
n.a.
```

### Debug Message

```plain
./components/esp_littlefs/src/littlefs/lfs.c:2246:debug: Relocating {0x2f, 0x2e} -> {0x79f, 0x2f}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-11:
Possibly wrong repository.  This is IDF functionality.  Are you seeing LFS debug messages in Arduino projects?

That message is printed if LFS_DEBUG is defined in lfs_util.h.  The only thing this repository might be able to control is defining LFS_NO_DEBUG.

everslick, 2023-01-11:
> Are you seeing LFS debug messages in Arduino projects?

yes, that's the point.

I was confused by the `./components/esp_littlefs` path as well and looked into IDF source first to report the issue there, but could not find it anythere in the repo. I assume it must be in some code Arduino CORE pulls in from elsewhere.

As this component is pre-compiled into **tools/sdk/esp32/lib/libesp_littlefs.a** it is something to be fixed in the build of that library I guess.

mrengineer7777, 2023-01-11:
The .a files (and config options) are built here: https://github.com/espressif/esp32-arduino-lib-builder

Hmmm.  I have a local copy of IDF 4.4.3 on my computer, but I can't find the "components\esp_littlefs" in the IDF repository.  I did notice that SPIFFS (which I use) uses ESP_LOGE() macros for debug statements vs printf in LittleFS.  I believe the Arduino compile turns those off ESP_LOGE(), which is why they aren't an issue.  This looks like an issue that IDF should solve: https://github.com/espressif/esp-idf.  Odd though that LittleFS has disappeared....

mrengineer7777, 2023-01-11:
This implies LFS may be a separate component now: https://github.com/espressif/esp-idf/pull/5469

lbernstone, 2023-01-11:
LittleFS is added as an additional component in lib-builder. Open a similar issue there so they can track it.

mrengineer7777, 2023-01-11:
https://github.com/joltwallet/esp_littlefs/tree/master/src
https://github.com/littlefs-project/littlefs/blob/40dba4a556e0d81dfbe64301a6aa4e18ceca896c/lfs.c#L2244

everslick, 2023-01-11:
@mrengineer7777 thanks for archeological work! ;-) I opened an issue here: https://github.com/espressif/esp32-arduino-lib-builder/issues/103

Jason2866, 2023-09-11:
Fixed in repo LittleFS

closed 2023-09-11 by Jason2866