espressif/arduino-esp32 #7706: HardwareSerial.h 2.0.6 not work with code 2.0.5

t4k3, 2023-01-12:
### Board

esp32 C3 Dev Module

### Device Description

esp32 and nextion 3.2" uart 

### Hardware Configuration

esp32 and nextion 3.2 disaply via uart 38400 baud

### Version

v2.0.6

### IDE Name

arduino 2.0.3

### Operating System

both mac and windows

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115

### Description

using the library HardwareSerial.h in 2.0.5 everything is fine.
updated to 2.0.6 serial stop working.
necessary downgrade to works

### Sketch

```cpp
void DisplayCTRL::Init(HardwareSerial* serialPort)
{
  this->SerialPort = serialPort;
}
```

### Debug Message

```plain
no message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-12:
@t4k3 Interesting but you don't give us anything to replicate the issue.  Can you setup a basic project that demonstrates the issue?  What do you mean by "stop working"?

t4k3, 2023-01-12:
We control a Nextion display with esp32 with 2.0.5 everything is ok with 2.0.6 serial communication doesn’t work probably parcel the string is different, I really can’t help you more the only thing I know is downgrading communication si ok

The code is this

#include "DisplayCTRL.h"

DisplayCTRL::DisplayCTRL():
SerialPort(0),
Page(-1),
States(0),
StatesTimer(0),
PageStates(0)
{

}

void DisplayCTRL::Init(HardwareSerial* serialPort)
{
  this->SerialPort = serialPort;
}

bool DisplayCTRL::DisplayReady()
{
  if(this->SerialPort == 0)
  {
    return false;
  }
  return true;
}

void DisplayCTRL::Exec()
{
  if(this->SerialPort == 0)
  {
    return;
  }
  char ToSend[64];
  uint8_t BytesReceived[1024];
  uint16_t ReceivedBytesSize = 0;
  while((ReceivedBytesSize = this->SerialPort->available()) > 0)
  {
    ReceivedBytesSize = ReceivedBytesSize > 1024 ? 1024 : ReceivedBytesSize;
    this->SerialPort->readBytes(BytesReceived, ReceivedBytesSize);
    if(ReceivedBytesSize >= 4)
    {
      // Check bytes validity
      if((BytesReceived[ReceivedBytesSize - 3] == 0xFF) &&
        (BytesReceived[ReceivedBytesSize - 2] == 0xFF) &&
        (BytesReceived[ReceivedBytesSize - 1] == 0xFF))
      {
        // correct info received
        switch(BytesReceived[0])
        {
          case 0x00:
          if(ReceivedBytesSize == 4)
          {
            // Invalid instruction
          }
          else
          {
            // Nextion startup
            if((BytesReceived[1] == 0) && (BytesReceived[2] == 0))
            {
              // Correct startup
            }
            this->States = 2;
            this->Page = 1;
            this->PageStates = 0;
          }
          break;
[... 733 lines cut ...]
    return MIN_SPEED;

    case MANUAL_PAGE:
    return ManualDownSetValue;
    
    case FLOAT_PAGE:
    return FloatDownSetValue;
    
    case SPIN_PAGE:
    return SpinDownSetValue;
    
    case SET_PAGE:
    return SetDownSetValue;
    
    case REVERSE_PAGE:
    return ReverseDownSetValue;

    case BLE_PAGE:
    return MIN_SPEED;
  }
}

Rossano Bertocco
***@***.***
+39 347 9574746

mrengineer7777, 2023-01-12:
That is not a simple example.  We can't debug your code for you.  What specific error are you seeing?

t4k3, 2023-01-12:
No I'm not asking debug
This is the code doesn't work with new version

Adrianotiger, 2023-01-23:
If I read it correctly, some error handlings was added in the last version. Like Buffer overflows.
Maybe in the version 2.0.5 you are just loosing some data but it works.
Now, with the version 2.0.6 you are notified when an error on the UART occurs.

t4k3, 2023-01-23:
Ok, thanks.

SuGlider, 2023-01-26:
@t4k3 - The issue seems to be related to the `Stream` function `readBytes()`.
This function has been overridden in Arduino 2.0.6 / `HardwareSerial` class with the intention to make the reading faster.

What is exactly the issue you are seeing?

SuGlider, 2023-01-26:
@t4k3 - I just tested a simplification of the code with `DisplayCTRL` and it seems to work fine with 2.0.5 and 2.0.6.
Please provide more information about how `Serial` is initialized. If possible, please provide a complete sketch with `setup()` and `loop()` for testing.

When starting `Serial`, try setting the RX Buffer Size, like this:
``` cpp
  Serial.setRxBufferSize(2048);  // always before Serial.begin()
  Serial.begin(115200);
``` 

The testing sketch I have used is:
``` cpp
class DisplayCTRL {
  public:
    HardwareSerial *SerialPort;

    DisplayCTRL();
    void Init(HardwareSerial* serialPort);
    void Testing();
};

DisplayCTRL::DisplayCTRL():
  SerialPort(0)
{
  log_d("Runnnig DisplayCTRL()");
}

void DisplayCTRL::Testing() {
  log_d("Runnnig Testing() Available = %d", SerialPort->available());
  if (this->SerialPort == 0)
  {
    log_e("Serial Port is NULL");
    return;
  }
  uint8_t BytesReceived[1024];
  uint16_t ReceivedBytesSize = 0;

  while ((ReceivedBytesSize = this->SerialPort->available()) > 0)
  {
    log_i("Available() ==> %d", ReceivedBytesSize);
    ReceivedBytesSize = ReceivedBytesSize > 1024 ? 1024 : ReceivedBytesSize;
    this->SerialPort->readBytes(BytesReceived, ReceivedBytesSize);
    this->SerialPort->printf("Got %d bytes: \"", ReceivedBytesSize);
    for (size_t i = 0; i < ReceivedBytesSize; i++) {
      this->SerialPort->write(BytesReceived[i]);
    }
    this->SerialPort->println("\"");
    log_i("readBytes() ==> %d", ReceivedBytesSize);
  }
}

void DisplayCTRL::Init(HardwareSerial* serialPort)
{
  log_d("Runnnig Init()");

  this->SerialPort = serialPort;
}

DisplayCTRL me;

void setup() {
  log_i("Setup() started.");
  Serial.setRxBufferSize(2048);
  Serial.begin(115200);
  while (!Serial) delay(5);
  Serial.println("SERIAL is READY... \n Starting.  \nUse Serial Monitor to send bytes to this sketch.\n");
  me.Init(&Serial);
}

void loop() {
  me.Testing();
  delay(500);
  log_i("LOOP...");
}

```

t4k3, 2023-01-27:
I’m not able to describe 
With 2.0.5 serial communication works with 2.0.6 doesn’t works

t4k3, 2023-01-27:
I try definite

Serial.setRxBufferSize(512);  
  Serial1.setRxBufferSize(512); 

 

But code with 2.0.6 not works with 2.0.5 works I’m not able to give more information

SuGlider, 2023-01-27:
@t4k3 -
The code provided so far is incomplete in order to try to understand the issue.
Based on the attempt to reproduce the issue [using the code I have assembled based on the snippet code provided](https://github.com/espressif/arduino-esp32/issues/7706#issuecomment-1404521604), I can't reproduce any issue.

Given that the issue is related to a 3rd party Library and product, I have no means to reproduce it here.
Therefore, I can't help any further. 

> I try definite Serial.setRxBufferSize(512); Serial1.setRxBufferSize(512); But code with 2.0.6 not works with 2.0.5 works I’m not able to give more information

I can see that the code provided in the issue uses a buffer with 1024 bytes, when processing the data, as shown below.
``` cpp
    ReceivedBytesSize = ReceivedBytesSize > 1024 ? 1024 : ReceivedBytesSize;
```
A last attempt would be to increase from 512 to 2048 `Serial.setRxBufferSize(2048);` - this line shall be placed before `Serial.begin(baud rate)`.

Gustavomurta, 2023-02-14:
Please what is the USB/Serial chip on your ESP32 C3 board? I'm having problems with the serial of my ESP32 C3 Wemos that uses the CH340 chip. I'm researching the problem.

t4k3, 2023-02-14:
I’m using ESP£”-C£-DevKIT but also a custom board we program with cable ftdi

Gustavomurta, 2023-02-14:
USB/Serial FTDI. Which chip? See the number above the chip.

t4k3, 2023-02-14:
Sorry I can’t see
The cable is 
FTDI FT232RL SH-U09G da USB-a TTL  DSD TECH

But I don’t think is the problem

Gustavomurta, 2023-02-14:
A very strange symptom is happening on the ESP32 C3 serial! I put a logic analyzer on Serial's TXD and RXD pins. Data is being sent! When I open the Arduino console, the ESP32 stops sending data. I'm searching.

Gustavomurta, 2023-02-15:
I achieve serial communication using the **Coolterm** program.
https://learn.sparkfun.com/tutorials/terminal-basics/coolterm-windows-mac-linux

**Serial port Setup :** 
Initial Line States when Port opens: 
**DTR OFF** and **RTS OFF**

These lines were resetting the board. And blocking communication.

**Arduino Console:** 
ESP32 Chip model = ESP32-C3 Rev 3
This chip has 1 cores
Chip ID: 12683120
ESP32 Chip model = ESP32-C3 Rev 3
This chip has 1 cores
Chip ID: 12683120

SuGlider, 2023-02-15:
DTR and RTS are used by ESP32-xx UART to reset the SoC or to set it to BOOT mode.

[image]

SuGlider, 2023-02-15:
About the C3:
https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html

SuGlider, 2023-02-15:
Thanks for observing it, @Gustavomurta!
@t4k3 - Do you thinkn it may be related to this issue?

SuGlider, 2023-02-15:
@t4k3 

> I’m not able to describe With 2.0.5 serial communication works with 2.0.6 doesn’t works

Based on the sketch used in https://github.com/espressif/arduino-esp32/issues/7706#issuecomment-1404521604, I can't replicate this issue.

In order to investigate it, I need the simpler piece of code that can reproduce the issue.

closed 2023-03-30 by mrengineer7777