espressif/arduino-esp32 #7707: SD cards not mounting

tomsmilton, 2023-01-12:
### Board

ESP32 Dev Board

### Device Description

ESP32 Dev kit. 
Micro SD card breakout board (3.3V).

### Hardware Configuration

GPIO 5 connected to CS
GPIO 23 connected to MOSI
GPIO 18 connected to CLK
GPIO 19 connected to MISO 

### Version

v2.0.6

### IDE Name

PlatformIO (VS Code)

### Operating System

Ubuntu

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

_Most_ SD cards will not connect. The sketch compiles and runs, but when trying to mount SD cards produces an error.

Using the SD library communicating by SPI _most_ SD cards do not mount. I have five identical (same size, type, brand etc.) micro SD cards, one of them always connects and runs perfectly. The other four do not mount at all. 

They all mount to my computer and have all been formatted as FAT32. 

I have tried directly wiring up the SD cards and I get the same behaviour (four don't work and one does). 

I have tried powering off a power supply to ensure that isn't the issue but the behaviour remains the same. 

The error message that I get is:

`ff_sd_initialize(): APP_OP_COND failed: 1`

This is different to the error that I get when the SD cards are not connected which is: 

`ff_sd_initialize(): GO_IDLE_STATE failed`

### Sketch

```cpp
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void setup()
{
  Serial.begin(115200);

}

void loop()
{
  if (!SD.begin())
  {
    Serial.println("Card Mount Failed");
  }
  else
  {
    Serial.println("Worked");
  }
  delay(2000);

}
```

### Debug Message

```plain
[ 72613][W][sd_diskio.cpp:550] ff_sd_initialize(): APP_OP_COND failed: 1
[ 72613][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
```

### Other Steps to Reproduce

I have also reproduced this soldering the pins of a micro SD card adaptor to the ESP32 with the correct pull-ups. The behaviour is identical. 
I have tried two different dev boards and the behaviour persists. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-01-12:
@P-R-O-C-H-Y can you please help with triage? Thanks

everslick, 2023-01-23:
I don't have this particular issue, that SD cards fail to mount, but I see other SD related problems like writes to files returning 0 (and no data was written), but also data that silently gets dropped. I cannot say more ATM, unfortunately. Just wanted to raise my suspicion, that something is off with SD card support in the current core, in case someone else sees similar issues.

Catzy44, 2023-01-25:
Im experiencing this issue too.
Code uploaded via arduino IDE 1.8.19 works perfectly but via PlatformIO downloaded today not working at all, same error.
Is there any temporary fix?
<3

nsieg, 2023-01-29:
I can confirm this behavior in my setup:

```INI
[env:esp32dev]
platform = espressif32
board = esp-wrover-kit
```

An older **SanDisk Extreme 16GB** will mount without errors, while a new **SanDisk Ultra 32GB** leads to:

```
[ 1065][E][sd_diskio.cpp:807] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
```
There is no difference in my case regarding the version used: it fails with 2.0.6 as well as 1.0.6 using Arduino IDE 1.8.19

P-R-O-C-H-Y, 2023-01-29:
@nsieg If I understand correctly, the issues with the new Sanddisk SD card accurs even using Arduino IDE? So it's not related to platformIO? Thanks

nsieg, 2023-01-30:
Yes, I am primarily using PlatformIO. I have tried different versions here (manually specifying `platform` and `platform-packages`), all leading to the error. To make sure, I tried the same using Arduino IDE (current and 1.8.19) but it remains the same with both IDEs.

Catzy44, 2023-01-30:
> Im experiencing this issue too. Code uploaded via arduino IDE 1.8.19 works perfectly but via PlatformIO downloaded today not working at all, same error. Is there any temporary fix? <3 [ 1166][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00 [ 1167][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work [ 1471][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00

Okay so i resolved my problem by manually specifying SPI pins, like this:

int SD_CS_PIN = 19;
SPI.begin(18, 36, 26, SD_CS_PIN);
SPI.setDataMode(SPI_MODE0);
SD.begin(SD_CS_PIN);

Skywalkerf34, 2023-04-23:
Same problem here, the card mount correctly 2 or 3 times then give this error message. If I remove the SD card then pull back the problem dissapear for 2 or 3 run then back again with same error
using 4G TF card formated with fat32

sivar2311, 2023-04-30:
I am also experiencing the same problem. Whether I try a purchased SD card module or a simple self-soldered adapter. 
The used SD card is a SanDisk Ultra Plus 32GB (the smallest card I could find). The card was of course formatted with Fat32 @ 32kb.

lbernstone, 2023-05-01:
> Okay so i resolved my problem by manually specifying SPI pins, like this:
> 
> int SD_CS_PIN = 19; SPI.begin(18, 36, 26, SD_CS_PIN); SPI.setDataMode(SPI_MODE0); SD.begin(SD_CS_PIN);
> 
> But It's still unclear to me why I need to manually specify SPI pins only on PlatformIO. (I'm using M5-STAMP-PICO module)

The M5stack-stamp-s3 (?) does not have spi pins defined in the variant file.  You can submit a PR if you would like to fix it.
https://github.com/espressif/arduino-esp32/blob/master/variants/m5stack_stamp_s3/pins_arduino.h
See https://github.com/espressif/arduino-esp32/blob/master/variants/esp32/pins_arduino.h#L20-L23 for the template

shaun-mzj, 2023-05-14:
We are experiencing the same issue, however it is very specific to the SanDisk Extreme 32 GB card. Other cards are mounting fine and multiple people with different devices are experiencing the same issue from our testing. We are using Arduino and the SD card test sketch for testing the issue and it's present in all versions we have tried.

Is there a solution for mounting SanDisk 32GB cards? Below is the output from the SD card test sketch

```07:21:51.868 -> [E][SD.cpp:32] begin(): After begin
07:21:51.900 -> [W][sd_diskio.cpp:169] sdCommand(): no token received
07:21:51.997 -> [W][sd_diskio.cpp:169] sdCommand(): no token received
07:21:52.094 -> [W][sd_diskio.cpp:169] sdCommand(): no token received
07:21:52.193 -> [E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
07:21:52.193 -> [W][sd_diskio.cpp:505] ff_sd_initialize(): GO_IDLE_STATE failed
07:21:52.226 -> [E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
07:21:52.226 -> [W][sd_diskio.cpp:169] sdCommand(): no token received
07:21:52.323 -> [W][sd_diskio.cpp:169] sdCommand(): no token received
07:21:52.422 -> [W][sd_diskio.cpp:169] sdCommand(): no token received
07:21:52.519 -> [E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
07:21:52.519 -> [E][SD.cpp:43] begin(): Unable to mount
07:21:52.519 -> Card Mount Failed```

And to clarify a different make 32GB card works in the same device without issue

Sell24, 2023-06-10:
Hardware: LilyGo TTGO T-SIM7600NA ESP32
Platform: Espressif 32 v6.3.1
Board: Esp32dev
PlatformIO IDE: 3.2.0

I cannot seem to get the SD card to work at all. Originally I had purchased a SanDisk 32GB and in looking at the previous posts to this issue, I went out and purchased Lexar SD cards to see if I could use that instead.

Both were formatted with FAT32, however I am still getting this issue:

```
16:00:58.069 > [   974][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
16:00:58.569 > [  1474][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
```
I looked at the build output and it appears that I am using the latest code

```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 ```

The code for sdSelectCard() in sd_diskio.cpp is as follows:
```c++
bool sdSelectCard(uint8_t pdrv)
{
    ardu_sdcard_t * card = s_cards[pdrv];
    digitalWrite(card->ssPin, LOW);
    bool s = sdWait(pdrv, 500);
    if (!s) {
        log_e("Select Failed");
        digitalWrite(card->ssPin, HIGH);
        return false;
    }
    return true;
}
```

I also tried the Arduino IDE v2.1.0 examples, but they also didn't work.

I realize that this issue is still open but I was just wondering if there was something specific/obvious I was possibly missing in my setup to make the Lexar SD (non SanDisk) card work.

DavidStacer, 2023-06-23:
I'm encountering similar issues trying to get various SD Cards to work.  Sometime a card will work for a while then not mount.  I had some success in running a Windows Scan disk on the card to correct errors. Sometimes that seems to help. 

The most consistent success I've had is formatting the card using an SD Card formatter from https://www.sdcard.org/ using the overwrite format. I do this when the card refuses to mount.  Fixed 3 of the 4 cards that were having the mounting problem.

fOmey, 2023-09-11:
> [  4498][W][sd_diskio.cpp:547] ff_sd_initialize(): APP_OP_COND failed: 1
> [  4498][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work

Error is showing up using a brand new sandisk 32GB microsd, my other card works perfectly fine.. very odd.

I made an image of my original card, wrote it to my new one.. same error. So I don't believe it's related to partitions.

malbrook, 2023-09-18:
We are also having a similar problem and seems especially related to Sandisk cards, cards bought over a year ago work fine, cards bought recently do not work, cards from Kingston Canvas Select marked as SDCS do work, haven't tried the newer SDCS2 cards, cards from Lexar seem OK. Card from Sandisk manufactured China, works OK, similar card from Sandisk manufactured in Tiawan does not work. So my current thinking is that there are changes in manufacture that are responsible and not the current libraries per se. Possibly a speed issue or timing issue.

All cards are Micro SD 32GB, SDHC, Class 10, UHS-I, all less than 100MB/s transfer speed and tested on M5Stack and custom designed ESP32 Wroom board,  all with 40MHz cpu clock.

svebert, 2023-09-18:
I had the exact same problem and most probably it was a fault of the MicroSD Card Adapter Board. I think, I did break the adapter board by using 5V instead of 3,3V in one of my first trials. Actually this did not destroy my SD Card, only the adapter board. Luckily I always buy at least 3 pieces, even if I only need 1...
Maybe this helps some people, here

aklein24, 2023-11-14:
> > [  4498][W][sd_diskio.cpp:547] ff_sd_initialize(): APP_OP_COND failed: 1
> > [  4498][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
> 
> Error is showing up using a brand new sandisk 32GB microsd, my other card works perfectly fine (also a sandisk 32GB microsd).. very odd.
> 
> I made an image of my original card, wrote it to my new one.. same error. So I don't believe it's related to partitions.
> 
> I'm using the arduino-esp32 library.
> 
> I've attached the output of `sudo udevadm info -a -n /dev/sde` hopefully this helps?
> 
> [working.log](https://github.com/espressif/arduino-esp32/files/12574594/working.log) [not_working.log](https://github.com/espressif/arduino-esp32/files/12574595/not_working.log)
> 
> Edit: Kioxia (16GB) branded sd cards seem to be working fine.

I'm having same problem with 16GB and 32GB sandisk SDHC cards. I had no problems 2 years ago, deployed 6 devices with ESP32 and SD cards without issues. Now I cannot get any to work. Not sure whether it's the SD cards or the library updates, but SD cards with ESP32 no longer function for me.

Doubletop12, 2023-11-22:
Same for me Sandisk Ultra 16GB however a old Sandisk 16GB works, I can read it but not write however that is a card problem I can't format it even on my PC

For the Sandisk Ultra 16GB

`[   335][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   336][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   642][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card Mount Failed`

I have tried @Catzy44 fix above without any luck and the MISO Pullup fix. Absolutly no response from the card when the mount is attempted at SD.Begin(SD_CS);

See attached
[image]

eismeraldo, 2024-01-14:
I have the same problem with different ESP32_S3 boards and the SPI SD Card Module:
[SPI SD](https://www.aliexpress.com/item/32877331401.html?spm=a2g0o.order_list.order_list_main.60.21ef1802yrqoKj
)
With the old boards like ESP32-D0WDQ6 everything works perfectly.
However, when I use this module, the ESP32-S3 boards also work.
[SDIO/SIP](https://www.aliexpress.com/item/1005006189454034.html?spm=a2g0o.productlist.main.11.7fa9505cMLPHAj&algo_pvid=b9280f4f-ac27-45a1-9747-35c05e5fc278&algo_exp_id=b9280f4f-ac27-45a1-9747-35c05e5fc278-5&pdp_npi=4%40dis%21CHF%211.47%210.46%21%21%2112.12%213.76%21%402103247917052561163513767e9053%2112000036196047516%21sea%21CH%21131359193%21&curPageLogUid=x4rqBPJPiMPH&utparam-url=scene%3Asearch%7Cquery_from%3A)
I use PlatformIO
If the ESP-S3 is operated on the serial interface, the SPI SD card cannot be handled.
[ 94][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[ 106][I][main.cpp:71] showStep(): **** 0 'Init SPI'
[ 106][I][main.cpp:71] showStep(): **** 1 'SD.begin'
[ 108][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 210][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 310][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 410][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 410][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[ 411][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 420][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 525][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 626][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 727][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 727][E][main.cpp:144] setup(): Initialization failed!

The Logic 2 Analyzer from saleae shows that the clock signal is not output correctly.
[image]

If the power is supplied via a laboratory power supply, the SPI SD card is written correctly and the Logic 2 image looks correct.
[image]

The old ESP32-D0WDQ6 work perfectly.
My conclusion:
I have tried various S3 boards (Lolin S3 Mini, S3 Original from Espresif, ESP32-S3-WROOM-1) with SPI SD card modules and they all have the same error. If they are powered via USB they will not work correctly, but if they are powered with a lab power supply everything is ok.
The old boards like ESP32-D0WDQ6 work perfectly with the SPI SD card module.
The S3 series has a problem with the current software or even a serious hardware problem related to the SPI SD card module.

My Source Code:
`/**
 * @brief Write and read file to and from the SD card using SPI and SD library
 *
 *  - Boards: ESP32-D0WDQ6_Dev_Module, ESP32-S3-WROOM-1
 *  - MicroSD Shield with SPI Interface
 *
 * How to use: Uncomment
 *  - BOARD_.... definition in this file
 *  - env: and board= in platformio.ini
 *
 * Connections:
 *  - BOARD_DOIT_DEFAULT_GPIO are the same as BOARD_DOIT_DEFINED_DEFAULT_GPIO
 *  - Otherwise connect the SPI to the GPIOs according to definitions
 *
 */

//-----------to uncomment-----------------
// #define BOARD_DOIT_DEFAULT_GPIO
// #define BOARD_DOIT_DEFINED_DEFAULT_GPIO
#define BOARD_ESP32_S3_DEFINED_GPIO
// #define BOARD_ESP32_S3_JTAG_DEFINED_GPIO
//----------------------------------------

#ifdef BOARD_DOIT_DEFINED_DEFAULT_GPIO
#define PIN_NUM_MOSI GPIO_NUM_13         // MOSI (White)
#define PIN_NUM_MISO GPIO_NUM_12         // MISO (Brown) (Pul up)
#define PIN_NUM_CLK GPIO_NUM_14          // SCK  (Rosa / Pink)
#define PIN_NUM_CS GPIO_NUM_15           // SD_CS (Orange)
#define PIN_NUM_SD_ACTIVITY GPIO_NUM_27  // Activity (Yellow) To show activity on the SD Card like read, write
#define PIN_STEP_NR_1 GPIO_NUM_26        // 2^0 (Green)
#define PIN_STEP_NR_2 GPIO_NUM_25        // 2^1 (Blue)
#define PIN_STEP_NR_4 GPIO_NUM_32        // 2^2 (Violet)
#endif
#ifdef BOARD_ESP32_S3_DEFINED_GPIO
[... 114 lines cut ...]
    writeToSD("/test.txt", "123ABCabc1122");
    readFromSD("/test.txt");
}

void loop() {}`

My platformio.ini
`
; [env:esp32doit-devkit-v1]
[env:esp32-s3-devkitc-1]
; [env:lolin_s3_mini]
; board = esp32doit-devkit-v1
; board = esp32-s3-devkitc-1
board = esp32-s3-devkitc-1-n16r8v
; board = lolin_s3_mini

platform = espressif32
framework = arduino
monitor_speed = 115200
monitor_filters = log2file, default
monitor_dtr = 1

build_flags = 
    -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    -DARDUINO_USB_CDC_ON_BOOT=1`

alesricar, 2024-02-26:
I've had exactly the same error as @eismeraldo. However in my case (I use ESP32 WROOM 32D dev board) I found out pin 23 can't be used for MOSI for some reason. After some frustration I found out that MOSI is still 0V. Using pin 21 fixed the issue. There is no need for me to use external power supply. USB power works ok for me too.

HoxBox, 2024-05-13:
I took all of the advice on this thread, and found a working solution... you're not going to like it...

1. Had to initialize different board-specific pins:

    /* Override pins */
    int SD_CS_PIN = 5;
    SPI.begin(18, 19, 21, SD_CS_PIN); // sck, miso, mosi, ss(cs)
    SPI.setDataMode(SPI_MODE0);

    if(!SD.begin(SD_CS_PIN)){
        Serial.println("Card Mount Failed");
        return;
    }

2. Had to change pin 23 to pin 21 to avoid problems with stock mosi pin.
3. Had to use a formatted SDHC card 32gb and under (in my case it's a cheap Lexar v10 I purchased for one of my pi systems).
4. When all that didn't work, I broke out a variable power supply and plugged it into my Freenove ESP32 Wroom Breakout Board's external power input and swapped the VCC pin over to the powered 5v rail (using a chip with auto down conversion so not going to fry it) then poof, like magic, it worked.

So seems it's all part of a larger issue, perhaps the newer chips are super sensitive to the power voltage, I'll have to test the VCC pin with and without the external power supply and update here when my new Multimeter arrives next week.

Curious if using a 3.7v battery would also fix this, got those on order as well for another project.

eismeraldo, 2024-05-15:
Problem solved.
Adapter created according to this information:
[](https://mischianti.org/how-to-use-sd-card-with-esp32-2/)
Now everything works perfectly.
I measured the ports with the oscilloscope and they all have a high voltage of 3.2V - 3.3V. It's not clear to me why these boards don't work when powered via USB with 5V from ESP32.
[image]

HoxBox, 2024-05-15:
Funny since mine only work when powered by the 5v and don't when on the 3.3v, I have a feeling it's probably board specific to do with the amps on each rail, I've only really got it working on my Freenove ESP32 but when I have time this weekend I plan to test on my entire library of ESP32's, perhaps some PI's, and I should have some AT based chips by then to try as well (if they're compatible even, I'm not sure).

Doubletop12, 2024-05-15:
We may have gone a bit off track here. The original problem was that some cards work and some don't in particular Sandisk cards. So if you have a solution you need to say what card you were using. You may have just been lucky to choose a card that works.

eismeraldo, 2024-05-15:
You're right, it might have been better to open a new thread, I apologize.
I tested the following types and all could be written and read successfully:

- eekoo MicroSD 128MB
- eekoo MicroSD 128MB
- MicroSD 256MB
- MicroSD 2GB Taiwan
- SanDisk ultra 16GB
- SanDisk ultra 64GB

qiweimao, 2024-05-26:
I ditched this module and switched to the 3V3 card adapter, but still have stability issues. Just like y'all mentioned, very well be power supply replated. My problem only went away after I directly wire the module with the 3V3 and GND pins. No cheap breadboard circuit in between!

I have documented my setup here: https://github.com/qiweimao/ESP32-Datalogger/issues/8#issuecomment-2127983769

RishabRao, 2024-07-03:
This adaptor works when I power the esp32 via USB and use the 5 VIN pin as a power source for the module. It appears that you need 5V for this module to work.

dizcza, 2024-07-12:
Just adding my five cents to the discussion.

If you share the SPI bus with a TFT screen or LoRa, the SD card must be initialized AFTER all other SPI devices are initialized.

rapax00, 2024-10-18:
> I've had exactly the same error as @eismeraldo. However in my case (I use ESP32 WROOM 32D dev board) I found out pin 23 can't be used for MOSI for some reason. After some frustration I found out [...]

I solved with it, thanks @alesricar.

It's strange that before it worked normally with MOSI on pin 23 but at one point when I uploaded the code back to esp32 it didn't work with MOSI on pin 23.

Finally I read this and solved change MOSI to pin 21.

My board is ESP32 DEV KIT.

P-R-O-C-H-Y, 2025-01-09:
Hi, I am closing this issue as there was no activity for a long time, seems to be solved and I have not occurred any new issues with SD on latest versions. If the issue persists feel free to reopen or open a new issue. Thanks

closed 2025-01-09 by P-R-O-C-H-Y

fOmey, 2025-01-09:
Confirmed resolved in latest versions.

fOmey, 2025-01-20:
@P-R-O-C-H-Y can you please reopen this issue?

gadingnst, 2025-01-28:
same module here, but I gotta an error:

```
Initializing SD card...
[  2559][W][sd_diskio.cpp:109] sdWait(): Wait Failed
[  2559][W][sd_diskio.cpp:485] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  2561][W][sd_diskio.cpp:489] ff_sd_initialize(): GO_IDLE_STATE failed
[  2567][E][sd_diskio.cpp:761] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3076][W][sd_diskio.cpp:109] sdWait(): Wait Failed
[  3076][E][sd_diskio.cpp:128] sdSelectCard(): Select Failed
```

TarqGaur, 2025-03-06:
my issue for lilygo t4 S3 inbuilt sd card when i am using this

audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(100);
  audio.connecttoFS(SD, "/a.wav");

[  1940][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x0d
[  1946][E][sd_diskio.cpp:624] ff_sd_status(): Check status failed
[  2253][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2560][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00

14790897, 2025-09-14:
same error:The physical drive cannot work