espressif/arduino-esp32 #7708: WebServer.handleClient() longer time execution on ESP32-C3 compares to ESP8266 (930 us vs 10us)

Owiec94, 2023-01-12:
### Board

ESP32-C3

### Device Description

ESP32-C3 DevKitM-1

### Hardware Configuration

GPIO 19 as a test for Logic Analyzer software

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10 x64

### Flash frequency

mode:DOUT, clock div:1

### PSRAM enabled

yes

### Upload speed

ota

### Description

Im using WebServer.handleClient() to manage GET commands with parameters in URL. This function on ESP32 takes 90 times longer compares to ESP8266.
My method of testing:
[image]
with Logic analyzer.

platform = espressif32
board = esp32-c3-devkitm-1

test with few handlers (webServer.on("/setParams"......):
933us
[image]

test without any handlers, just webServer.init():
933us
[image]

platform = espressif8266
board = esp07
test with few handlers:
11.88us
[image]

test without any handlers, just init:
[image]
11.88us

ESP32-C3 needs 933us compares to 11us on ESP8266 which is 90x timer slower than older chips. In my solution time is critical and I had to slow down other peripherals (to prevent overflow buffers), which affects essential on performance.

### Sketch

```cpp
platformio.ini
[env:esp32dev]
platform = espressif32
board = esp32-c3-devkitm-1
board_build.flash_mode = dout
board_build.f_cpu = 160000000L
board_build.filesystem = littlefs
board_build.mcu = esp32c3
board_build.variant = esp32c3
framework = arduino
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

This long time execution doesnt matter if WiFi is connected or not, also doesnt matter if we called before webServer.init() or not.

Both chips are clocked 160MHz.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Owiec94, 2023-01-12:
> @Owiec94 - Would you mind posting an example Sketch that demonstrates this issue? It would help us to analyse it.

Sure, I've pasted basic code without my stuff which still generating long time execution - tested before paste here. If You need any additional info feel free to write, I'll do my best. Hope that helps quickly optimize that function. Thank You

Owiec94, 2023-02-08:
@SuGlider any chances to fix it? Should I provide any info to help solve that problem?

SuGlider, 2023-02-09:
Yes, I'll work on it soon!

scottchiefbaker, 2023-03-09:
I ran in to this same issue and posted an easily [recreatable](https://www.reddit.com/r/esp32/comments/11mzyri/webserverh_handleclient_slows_my_code_down_by_3737/) (is that a word) example on Reddit. This issue affects ALL ESP32 boards so the title may need to be updated.

With `WebServer.h` in it's current form it is impossible to get your main `loop()` to run more than 1000 times per second. This is extremely limiting for most applications.

|  Board   | `handleClient()` on | `handleClient()` off |
| -------- | ------------------- | -------------------- |
| ESP8266  | 118156/s            | 142672/s             |
| ESP32    | 1001/s              | 37409/s              |
| ESP32-S2 | 994/s               | 289981/s             |

The issue only crops up when `handleClient()` is called. I believe the issue is the `delay(1)` that's inside of `handleClient()` [here](https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/libraries/WebServer/src/WebServer.cpp#L290).

Whoever implemented this planned for this, because it can be disabled with `server.enableDelay(false)`. This is an _undocumented_ feature, and **really** hard to track down. Having `delay` inside of `handleClient()` should not be the default, because most people will not expect that adding a web server to their project will drop performance so much.

scottchiefbaker, 2023-03-09:
Attached is the script I use to test this: [blink-web.ino.gz](https://github.com/espressif/arduino-esp32/files/10936185/blink-web.ino.gz).

FWIW while simply adding `server.enableDelay(false)` **does** improve performance significantly, the performance level is still _significantly_ lower than `ESP8266WebServer.h`. 

|  Board   | Iterations | Percent of ESP8266 |
| -------- | ---------- | ------------------ |
| ESP8266  | 118156     | 100%               |
| ESP32    | 32293      | 27.3%              |
| ESP32-S2 | 57370      | 48.5%              |

Step 1 should be removing that arbitrary delay (or making `enableDelay(true)` the default). 
Step 2 should be to dive deeper in to `WebServer.h` to figure out why it's **so** much slower than `ESP8266WebServer.h` even with more powerful ESP32 hardware.

scottchiefbaker, 2023-03-27:
The issue as I see it is that a person could have a project that has a working requirement of 20kHz, they add in `WebServer.h` and now their project is limited to 1kHz. There is little to no explanation about this limitation, nor is it on par with other webserver libraries. `ESP8266WebServer.h` has an effective 116kHz and the `ESPAsyncWebServer.h` has speed in the 175kHz range.

Wouldn't it make more sense to `yield()` than `delay(1)`? That would free up resources for the other spawned threads without limiting loop speed.

Jason2866, 2025-03-12:
Closing since not an issue. Implementation is different for esp8266 and esp32

closed 2025-03-12 by Jason2866

scottchiefbaker, 2025-03-12:
I understand the implementations are different, but I think a good target goal would at least be similar performance. This is pretty limiting for **MOST** applications. Closing it as WONTFIX doesn't help anyone.

me-no-dev, 2025-03-12:
@scottchiefbaker they are very different in implementation and can never get even close. There are multiple levels of things between RAW LwIP (what the 8266 is based on) and sockets (what ESP32 is based on) and also there is FreeRTOS on ESP32. Something with similar performance is the AsyncWebServer that works on both and does not even have `handleClient()`