espressif/arduino-esp32 #7710: include esp-dsp header in a sketch

parlotto, 2023-01-13:
### Board

ESP32 dev module

### Device Description

esp32 dev module alone

### Hardware Configuration

nothing connected

### Version

v2.0.6

### IDE Name

Arduino IDE , platformIO

### Operating System

linux

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

How to correctly include esp-dsp header files in a sktech ?
They are in the directory tools/sdk/esp32/include/esp-dsp/modules/iir/include/... but I can't find a way to correctly include
them in a sketch.
Pointing the absolute location on my disk seams to work but It is not a portable solution.

### Sketch

```cpp
#include "/home/philippe/.arduino15/packages/esp32/hardware/esp32/2.0.6/tools/sdk/esp32/include/esp-dsp/modules/iir/include/dsps_biquad_gen.h" 

void setup() {
  Serial.begin(115200);
  float coeffs[15]={0},f=0.4,qFactor=4;
dsps_biquad_gen_lpf_f32(coeffs, f,  qFactor);

for (int i=0;i<15;i++){
  Serial.printf("%f \n",coeffs[i]);
}
}

void loop() {
  

}
```

### Debug Message

```plain
# with the ugly include line I have this error
/home/philippe/Arduino/sketch_jan12a/sketch_jan12a.ino: In function 'void setup()':
sketch_jan12a:6:1: error: 'dsps_biquad_gen_lpf_f32' was not declared in this scope
 dsps_biquad_gen_lpf_f32(coeffs, f,  qFactor);
 ^~~~~~~~~~~~~~~~~~~~~~~
exit status 1
'dsps_biquad_gen_lpf_f32' was not declared in this scope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

parlotto, 2023-01-14:
Sorry, I was using arduino 1.8. It works on arduino 2.0.3 by simply a #include <dsps_biquad.h>
But still not make it work on platformIO.

parlotto, 2023-01-16:
in Platformio I still have an error :

**src/main.cpp:3:29: fatal error: dsps_biquad_gen.h: No such file or directory**

platformio.ini  content is :
 
[env:lolin_d32]
platform = https://github.com/platformio/platform-espressif32.git
board = lolin_d32
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
arduino-esp32.gitframework = arduino

mrengineer7777, 2023-01-16:
Try this environment:
[env:lolin_d32_test]
board = lolin_d32
monitor_speed = 115200
platform = espressif32
framework = arduino
monitor_filters = esp32_exception_decoder

Also try closing and reopening VSCode.  PIO manages the include paths for VSCode, but sometimes it takes reopening VSCode for them to be fixed.

parlotto, 2023-01-16:
Same error with this environment even after restarting VSCode.
thank you for you help

mrengineer7777, 2023-01-16:
The following compiles for me.

platformio.ini
```
[env:lolin_d32_test]
board = lolin_d32
monitor_speed = 115200
platform = espressif32
framework = arduino
monitor_filters = esp32_exception_decoder
```

main.cpp
```
#include <Arduino.h>
#include "dsps_biquad_gen.h" 

void setup() {
  Serial.begin(115200);
  float coeffs[15]={0},f=0.4,qFactor=4;
dsps_biquad_gen_lpf_f32(coeffs, f,  qFactor);

for (int i=0;i<15;i++){
  Serial.printf("%f \n",coeffs[i]);
}
}

void loop() {
  

}
```

rftafas, 2025-01-06:
Since PlatformIO does not support Arduino Core for ESP32 newer than 2.0.17, it is impossible to keep supporting this. Please, re-open this issue if tested against newer versions and if the problem remains.

closed 2025-01-06 by rftafas