espressif/arduino-esp32 #7712: USBCDC can only receive but cannot send

Challenger-0, 2023-01-14:
### Board

ESP32S3 Dev Module

### Device Description

ESP32S3 DevKitC

### Hardware Configuration

USB Connected

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

240MHz(WiFi)

### PSRAM enabled

yes

### Upload speed

115200

### Description

USBCDC can only receive but cannot send

### Sketch

```cpp
//USB Serial Example

#if ARDUINO_USB_MODE
#warning This sketch should be used when USB is in OTG mode
void setup(){}
void loop(){}
#else
#include "USB.h"

#if ARDUINO_USB_CDC_ON_BOOT
#define HWSerial Serial0
#define USBSerial Serial
#else
#define HWSerial Serial
USBCDC USBSerial;
#endif

static void usbEventCallback(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data){
  if(event_base == ARDUINO_USB_EVENTS){
    arduino_usb_event_data_t * data = (arduino_usb_event_data_t*)event_data;
    switch (event_id){
      case ARDUINO_USB_STARTED_EVENT:
        HWSerial.println("USB PLUGGED");
        break;
      case ARDUINO_USB_STOPPED_EVENT:
        HWSerial.println("USB UNPLUGGED");
        break;
      case ARDUINO_USB_SUSPEND_EVENT:
        HWSerial.printf("USB SUSPENDED: remote_wakeup_en: %u\n", data->suspend.remote_wakeup_en);
        break;
      case ARDUINO_USB_RESUME_EVENT:
        HWSerial.println("USB RESUMED");
        break;
      
      default:
        break;
    }
  } else if(event_base == ARDUINO_USB_CDC_EVENTS){
    arduino_usb_cdc_event_data_t * data = (arduino_usb_cdc_event_data_t*)event_data;
    switch (event_id){
      case ARDUINO_USB_CDC_CONNECTED_EVENT:
        HWSerial.println("CDC CONNECTED");
        break;
      case ARDUINO_USB_CDC_DISCONNECTED_EVENT:
        HWSerial.println("CDC DISCONNECTED");
        break;
      case ARDUINO_USB_CDC_LINE_STATE_EVENT:
        HWSerial.printf("CDC LINE STATE: dtr: %u, rts: %u\n", data->line_state.dtr, data->line_state.rts);
        break;
      case ARDUINO_USB_CDC_LINE_CODING_EVENT:
        HWSerial.printf("CDC LINE CODING: bit_rate: %u, data_bits: %u, stop_bits: %u, parity: %u\n", data->line_coding.bit_rate, data->line_coding.data_bits, data->line_coding.stop_bits, data->line_coding.parity);
        break;
      case ARDUINO_USB_CDC_RX_EVENT:
        HWSerial.printf("CDC RX [%u]:", data->rx.len);
        {
            uint8_t buf[data->rx.len];
            size_t len = USBSerial.read(buf, data->rx.len);
            HWSerial.write(buf, len);
        }
        HWSerial.println();
        break;
       case ARDUINO_USB_CDC_RX_OVERFLOW_EVENT:
        HWSerial.printf("CDC RX Overflow of %d bytes", data->rx_overflow.dropped_bytes);
        break;
     
      default:
        break;
    }
  }
}

void setup() {
  HWSerial.begin(115200);
  HWSerial.setDebugOutput(true);
  
  USB.onEvent(usbEventCallback);
  USBSerial.onEvent(usbEventCallback);
  
  USBSerial.begin();
  USB.begin();
}

void loop() {
  while(HWSerial.available()){
    size_t l = HWSerial.available();
    uint8_t b[l];
    l = HWSerial.read(b, l);
    USBSerial.write(b, l);
  }
}
#endif /* ARDUINO_USB_MODE */
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
[    88][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface CDC en��[    96][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-01-26:
@Challenger-0 
I tested the sketch you have provided using Win11 and the IDE 1.8.15
Selected `USB Mode: "USB-OTG(TinyUSB)"` and `USB CDC On Boot: "Disabled"`
Flashed an [ESP32-S3 board](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html) that has both ports, UART and USB.

I open the serial monitor to the UART Win COM port and I open another terminal (PuTTY) with the TinyUSB CDC COM Port.
Whatever I type in the PuTTY shows up in the Serial Monitor and vice-versa.

Therefore, it seems to work correclty.

What is the issue you see there? 
How are you testing it?

Challenger-0, 2023-01-26:
I tested using Win11, Arduino IDE 1.0.16 and 2.0.3,  `USB Mode: "USB-OTG(TinyUSB)"` and `USB CDC On Boot: "Disabled"`

Flashed _MuseLab nanoESP32-S3_ and _YD-ESP32S3_ that has both USB and UART ports.

[image]

The window on the left is TinyUSB CDC COM Port (COM30) and on the right is UART COM (COM21) port. 
When I type into COM30, everything can be transfer to COM21, but when I type into COM21, nothing appear at COM30

I send three messages `This is Hardware Serial! Test1` `This is Hardware Serial! Test2` `This is Hardware Serial! Test3` to COM21, nothing appear at COM30, then I send three messages `This is USBCDC Serial! Test1` `This is USBCDC Serial! Test2` `This is USBCDC Serial! Test3`  to COM30, they are all show up COM21.

Challenger-0, 2023-01-26:
> @Challenger-0 I tested the sketch you have provided using Win11 and the IDE 1.8.15 Selected `USB Mode: "USB-OTG(TinyUSB)"` and `USB CDC On Boot: "Disabled"` Flashed an [ESP32-S3 board](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html) that has both ports, UART and USB.
> 
> I open the serial monitor to the UART Win COM port and I open another terminal (PuTTY) with the TinyUSB CDC COM Port. Whatever I type into the PuTTY shows up on the Serial Monitor and vice-versa.
> 
> Therefore, it seems to work correctly.
> 
> What is the issue you see there? How are you testing it?

@SuGlider 
I tested using Win11, Arduino IDE 1.0.16 and 2.0.3,  `USB Mode: "USB-OTG(TinyUSB)"` and `USB CDC On Boot: "Disabled"`

Flashed _MuseLab nanoESP32-S3_ and _YD-ESP32S3_ that has both USB and UART ports.

[image]

The window on the left is TinyUSB CDC COM Port (COM30) and on the right is UART COM (COM21) port. 
When I type into COM30, everything can be transfer to COM21, but when I type into COM21, nothing appear at COM30

I send three messages `This is Hardware Serial! Test1` `This is Hardware Serial! Test2` `This is Hardware Serial! Test3` to COM21, nothing appear at COM30, then I send three messages `This is USBCDC Serial! Test1` `This is USBCDC Serial! Test2` `This is USBCDC Serial! Test3`  to COM30, they are all show up COM21.

lbernstone, 2023-01-26:
Please test using v2.0.6.  USB has been under major development throughout the 2.0 series.

Challenger-0, 2023-01-26:
> Please test using v2.0.6. USB has been under major development throughout the 2.0 series.

@lbernstone 
The test is carried out under the condition of v2.0.6

SuGlider, 2023-01-27:
@Challenger-0 

Maybe the issue is in the Serial Debug Assistant software when used with USB CDC.

As I said, using other terminal software such as Arduino IDE Serial Monitor or PuTTY, it works perfectly well, using Arduino Core 2.0.6.
Could you please try using the setup as I described?

A simple test with USB CDC with a sketch, to read and echo back what has been just read, works using the Arduino IDE Serial Monitor. Therefore, this issue can't be reproduced.

SuGlider, 2023-01-27:
This is the "Echo Back" simple test for UART or CDC:

``` cpp
// simple echo back using UART or CDC:
// IDE Setup: 
//    USB Mode: "USB-OTG(TinyUSB)"
//    USB CDC On Boot: "Disabled"   ==> it will work with UART (Serial = UART)
//    USB CDC On Boot: "Enabled"    ==> it will work with USB CDC (Serial = CDC)
// Open the Serial Monitor using the respective COM port fir UART or for USB
// Type anything and see the result

void setup() {
  Serial.begin(115200);
}

void loop() {
  int data = Serial.read();
  if (data > 0) {
    Serial.printf("Byte Read: %d [CHAR:'%c']\n", data, data);
  }
}
```

Challenger-0, 2023-01-27:
@SuGlider 

It works well. 
Many thanks. I never noticed the issue is in the Serial Debug Assistant software 
[image]
[image]

closed 2023-01-27 by Challenger-0