espressif/arduino-esp32 #7713: ESP32-C6 support

probonopd, 2023-01-14:
### Related area

ESP32-C6 

### Hardware specification

Support for ESP32-C6

### Is your feature request related to a problem?

ESP32-C6 is not supported yet

### Describe the solution you'd like

Support for ESP32-C6

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

probonopd, 2023-01-14:
https://www.espressif.com/en/products/socs/esp32-c6

lbernstone, 2023-01-14:
C6 will not be supported until ESP-IDF 5.1, so arduino will not support it until that is released.

VojtechBartoska, 2023-01-16:
Hello, support of ESP32-C6 is on our Roadmap, @lbernstone is right that it will be supported in ESP-IDF 5.1 so in Arduino, it will be in version 3.X.X. We'll soon create development branch and we will start working on IDF 5 support there.

@probonopd Thanks for opening the issue, we will use it to track general support ESP32-C6 :)

tuct, 2023-01-31:
My first esp32 c6 dev boards v0.0 arrived today, eager to test it!

simogaspa84, 2023-02-07:
Hi @tuct how did you get it the esp32 c6 dev board? which supplier ? thanks

simogaspa84, 2023-02-09:
Thanks a lot @negis02 .. very kind... are you going to try the new features about cyber security in the C6 dev board? 
I am really interested about the trusted execution enviroment and the glitch detector.. Do you know something about that?
Thanks

seajhawk, 2023-02-12:
Hey @simogaspa84 - thanks for posting.
 I'm mainly interested in the Thread support, but I'll have to look at the trusted execution and glitch detector features. I wasn't even aware of those!

I'm building an air quality, temp, humidity, CO2 sensor. 

Does anyone know if we can use a C3 board for basic prototyping until the C6 support is added?

tuct, 2023-02-12:
I got mine from the official store at aliexpress
€ 9,81  1%OFF | ESP32-C6-DevKitC-1 Entwicklung Bord Espressif Systeme ESP32-C6 Serie * Probe * Limit 2 stücke pro Kunden * (siehe beschreibung)
https://a.aliexpress.com/_mLI99g2
I bought 5, so the limit is new, happy to share of you are close to or in Vienna

Trevo525, 2023-02-14:
You can also purchase from Adafruit's [official store](https://www.adafruit.com/product/5672). They have 47 in stock at the time of writing this.

a-d-r-i-a-n-d, 2023-02-22:
I've ordered mine today, looking forward to C6 support.

MingyaoLiu, 2023-02-26:
I just got my esp32-c6 wroom module, looking forward to have arduino capability!

guumeyer, 2023-02-27:
I just got my esp32-c6 wroom module too, looking forward to have arduino capability!

N0ury, 2023-03-04:
idf.py --version
ESP-IDF v5.1-dev-3710-gacac972f70

It looks like v5.1 has been released ;-)

me-no-dev, 2023-03-06:
It has not. This is just the version that will be used in the future. 5.1 is in ESP-IDF master and needs to be split in it's own branch before it will be released. It has not even been feature-frozen yet :)

djmjr77, 2023-03-27:
Update please?

me-no-dev, 2023-03-30:
Guys, please stop asking for update every few days. Before we add support for new chips, there is a lot of work that needs to be done. When that is done (will take way more than few days) most if not all new chips will be supported. We will announce that, so even less reason to keep asking.

Hedda, 2023-04-20:
> Hello, support of ESP32-C6 is on our Roadmap, @lbernstone is right that it will be supported in ESP-IDF 5.1 so in Arduino, it will be in version 3.X.X. We'll soon create development branch and we will start working on IDF 5 support there.
> 
> @probonopd Thanks for opening the issue, we will use it to track progress on support of ESP32-C6 :)

As noted, the official Espressif ESP-IDF v5.1, which is currently in development, will include initial support for the new ESP32-C6 chip, however, while Espressif ESP-IDF v5.1 has not yet been released it should work if using the master branch in esp-idf repository. If you want to track the development status then follow these the updates made to these other issues for ESP32-C6:

https://github.com/espressif/arduino-esp32/issues/7852

https://github.com/espressif/esp-idf/issues/10423

LenReinhart, 2023-05-16:
Another waiting for C6 support. Subscribed.

SuGlider, 2023-06-22:
Ok, we have C6 support already. You can try the C6 branch:
https://github.com/espressif/arduino-esp32/tree/feature/esp32c6-support

SuGlider, 2023-06-22:
This the initial stage of support. At this time we are working in testing and fixing.

UART is fully tested and everything works correclty, including using UART with `setCpuFrequencyMhz()` at 160, 80, 40, 20 and 10MHz.

The same for RMT.

SuGlider, 2023-06-22:
I have tested the Basic WiFi and BLE examples and those are working fine with the current C6 branch.
At this point, please, don't expect to see low power consumption yet.

We still must work on the IDF Power Management component of the Arduino Core for the ESP32-C6.
It will take some time, but we expect to offer similar IDF results to the Arduino Users related to power consumpion in IoT applications with the C6, and maybe also for other SoCs.

Especificaly for the C6, we will also work on the WiFi 6 (802.11ax), adding an Arduino API for controlling the TWT (Target Wait Time) configuration.

#### Some IDF examples using the above referenced IDF APIs:
IDF Power Management Example:
https://github.com/espressif/esp-idf/blob/master/examples/wifi/power_save/README.md

IDF ESP32-C6 WiFi 6 802.11ax Example:
https://github.com/espressif/esp-idf/tree/master/examples/wifi/itwt

LenReinhart, 2023-06-22:
How do I integrate this with the Arduino application? 

 

https://github.com/espressif/esp-idf/blob/master/examples/wifi/power_save/README.md

https://github.com/espressif/esp-idf/tree/master/examples/wifi/itwt

lbernstone, 2023-06-22:
@LenReinhart This is a very early release.  It is intended for the people that will build those pieces. As suglider says, it will take some time.

SuGlider, 2023-06-22:
> How do I integrate this with the Arduino application? 

If you are talknig about how to "install" the C6 Arduino Core branch, it is necessary to use it as a `sketchbook board` by cloning this branch into the local Sketchbook Folder. The process of cloning and creating the `sketchbook board` is described in this link:

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#how-to-update-to-the-latest-code

LenReinhart, 2023-06-22:
Understood. I have a copy of Dr. Ting’s Forth that was designed for the ESP32 dev board. It uses Arduino to build the image download. I correctly guessed that the front end would be as much the same as possible on the compiler and it would take care of the translation. I successfully compiled and loaded it on the C3 dev board simply by changing the target. I have two of the C6 boards and would like to see if it will work for that. If it doesn’t no big deal. Bluetooth did not work, but I didn’t expect it would as it was the LE version. I’m 76 and have been working with computer systems since the 8080, now I just like to mess around with stuff and I thought what was done with using the  Arduino system, to build and install a copy of Forth OS, I thought was very clever. The fact that it has the possibility of being very transportable among processors using the same compiler and the way the Arduino system takes care of the addressing differences between hardware is very nice IMO.

lbernstone, 2023-06-23:
@LenReinhart Sounds good.  Build it on the C3, which has a stable build ATM :)

VojtechBartoska, 2023-06-23:
First initial support PR: https://github.com/espressif/arduino-esp32/pull/8337

VojtechBartoska, 2023-06-26:
## Table with Support overview

_We will keep it up to date.

Date of last update: 26.6.2023_

| **ESP32-C6 Support Overview**   |   Assignee   |                |             |              |
|---------------------------------|:------------:|:--------------:|:-----------:|:------------:|
|          **Peripheral**         |              |   **Testing**  | **Updates** | **Comments** |
|               ADC               | @P-R-O-C-H-Y |                |             |              |
|          Ethernet  SPI          |              |                |             |              |
|               GPIO              | @P-R-O-C-H-Y |                |             |              |
|               GDMA              |              |                |             |              |
|       I2C (Slave & Master)      |              |                |             |              |
|               I2S               |              |                |             |              |
|             LED PWM             |              |                |             |              |
|              MCPWM              |              |                |             |              |
|               PCNT              |              |                |             |              |
|        Parallel IO driver       |              |                |             |              |
|               RMT               |              |                |             |              |
|          SDIO 2.0 Slave         |              |                |             |              |
|               SPI               |              |                |             |              |
|               TWAI              |              |                |             |              |
|           Temp. Sensor          |              |                |             |              |
|               UART              |   @SuGlider  |    Fully Ok    |             |              |
|         USB Serial/JTAG         |              |                |             |              |
|            SigmaDelta           | @P-R-O-C-H-Y |                |             |              |
| Timers (System/Generel-purpose) | @P-R-O-C-H-Y |                |             |              |
|              Touch              |              |                |             |              |
|           **General**           |              |                |             |              |
|              Wi-Fi              |   @SuGlider  | Examples works |             |              |
|            Bluetooth            |              |                |             |              |
|           Bluetooth LE          |   @SuGlider  | Examples works |             |              |
|             802.15.4            |              |                |             |              |
|              Thread             |              |                |             |              |
|              Zigbee             |              |                |             |              |

lboue, 2023-07-01:
> C6 will not be supported until ESP-IDF 5.1, so arduino will not support it until that is released.

[ESP-IDF Release v5.1](https://github.com/espressif/esp-idf/releases/tag/v5.1) has been released.

rdahlin, 2023-07-27:
> > C6 will not be supported until ESP-IDF 5.1, so arduino will not support it until that is released.
> 
> @lbernstone [ESP-IDF Release v5.1](https://github.com/espressif/esp-idf/releases/tag/v5.1) has been released.

Any news here?

lbernstone, 2023-07-27:
Pre-release support as shown in the table above is in the [idf-v5.1 branch](https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs)

VojtechBartoska, 2023-07-31:
Hello, we are now collecting feedback on release 2.0.11 and when we are good to go, above mentioned branch https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs will be merged into master and initial support for ESP32-C6 & ESP32-H2 will be available for development purposes.

petergjoel, 2023-08-01:
@VojtechBartoska in `HardwareSerial.cpp` I find `#define RX1 5` and `#define TX1 4` for UART1 (LP_UART pins) - this is conflicting with the documentation.

However, using the `#define RX1 5` and `#define TX1 4` works fine for our application -- as long as we flip the wiring accordingly.

petergjoel, 2023-08-01:
See also issue with number of UART ports and the esp32-c6: https://github.com/espressif/esp-idf/issues/11989#issue-1830756977

HX2003, 2023-09-29:
Hi all, I have successfully flashed the ESP32-C6 using https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs. Using ble Unified Provisioning I got [`Configuration mismatch. Select BLE Only or BTDM mode from menuconfig`](https://github.com/espressif/esp-idf/blob/cb174b0fe1f150f2ecc2ce87e0f02154405c6677/components/protocomm/src/simple_ble/simple_ble.c#L228C20-L228C88), this error does not occur on the ESP32-S3 using the same build. I wish to continue to use Arduino IDE and not setup platform io.

lbernstone, 2023-09-29:
> Hi all, I have successfully flashed the ESP32-C6 using [`esp-idf-v5.1-libs`](https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs?rgh-link-date=2023-09-29T18%3A04%3A50Z). Using ble Unified Provisioning I got [`Configuration mismatch. Select BLE Only or BTDM mode from menuconfig`](https://github.com/espressif/esp-idf/blob/cb174b0fe1f150f2ecc2ce87e0f02154405c6677/components/protocomm/src/simple_ble/simple_ble.c#L228C20-L228C88), this error does not occur on the ESP32-S3 using the same build. I wish to continue to use Arduino IDE and not setup platform io.

Looks like this is a problem upstream.  There are no CTRL_MODE settings in C6 menuconfig.

HX2003, 2023-10-02:
@lbernstone I am directly invoking from esp idf
`#include <wifi_provisioning/scheme_ble.h>`
`#include <wifi_provisioning/manager.h>`
Although I assume this problem would also occur if one uses `libraries/WiFiProv/examples/WiFiProv` as well.

IsraelMek, 2023-10-07:
Hi, I have added ESP32 core 2.0.14 to the boards manager. Even though in github ESP32C6 is mentioned on the list of supported boards, it does not show up in the selection list of Arduino.

me-no-dev, 2023-10-07:
@IsraelMek you need 3.0.0 for that. It is still in beta and you need to use different BoardManger URL to get it: https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

IsraelMek, 2023-10-07:
@me-no-dev Thanks a lot . Update done to 3.0.0 and I can see ESP32-C6 in the board selection. But it doesn't compile the builtin examples for when C6 is selected. it displays error message "bits/c++config.h: No such file or directory". For ESP32-S3, the IDE compiles the example without any issue.

me-no-dev, 2023-10-07:
It seems that Arduino has some problems extracting the toolchains properly... especially riscv one. Will investigate Monday and come back. You could open the json in the link and download and manually install the toolchain till then

me-no-dev, 2023-10-07:
RV toolchains:
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-x86_64-linux-gnu.tar.xz
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-aarch64-linux-gnu.tar.xz
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-arm-linux-gnueabi.tar.xz
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-i686-linux-gnu.tar.xz
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-x86_64-apple-darwin.tar.xz
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-aarch64-apple-darwin.tar.xz
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-i686-w64-mingw32.zip
https://github.com/espressif/crosstool-NG/releases/download/esp-12.2.0_20230208/riscv32-esp-elf-12.2.0_20230208-x86_64-w64-mingw32.zip

IsraelMek, 2023-10-08:
@me-no-dev The manual installation fixed it. Thanks once again.

Hook654321, 2023-10-09:
How do I have to install it manually?

Atomique13, 2023-10-10:
> @me-no-dev Thanks a lot . Update done to 3.0.0 and I can see ESP32-C6 in the board selection. But it doesn't compile the builtin examples when C6 is selected. it displays error message "bits/c++config.h: No such file or directory". For ESP32-S3, the IDE compiles the example codes without any issue.

Same problem on 3.0.0 with C3.

me-no-dev, 2023-10-10:
> Same problem on 3.0.0 with C3 and H2.

All RiscV chips currently use the same exact toolchain, so yes. same on C3, C6 and H2

me-no-dev, 2023-10-10:
We have found the problem, but not a solution yet. The path to some files inside the toolchain becomes too long for gcc to handle and it does not see the file, though it's there. When installed from git, this issue does not occur, because path is short enough to not trigger it. We do not control where the toolchains are installed, so ... fingers crossed 😆

TD-er, 2023-10-10:
So installing it to something like `C:\A\` (shortest path possible) could be a work-around?

Or is it a problem where the command line of some commands gets too long? (including all paths needed for the command line)

me-no-dev, 2023-10-10:
It's the path to the header that get's too long. GCC seems to limit it to maximum 200 symbols (including the drive). This is how the path looks: 
`c:/users/username/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-12.2.0_20230208/riscv32-esp-elf/include/c++/12.2.0/riscv32-esp-elf/rv32imc_zicsr_zifencei/ilp32/no-rtti/bits/c++allocator.h`

TD-er, 2023-10-10:
I do see a lot of "riscv32-esp-elf" being used in the path.
Isn't that a bit too verbose (and obvious enough) ?
Also "rv32" is being used, so this suggests it can be shortened without loosing its meaning.

TD-er, 2023-10-10:
By the way, not sure if it is a limit of GCC. I think it might be a limit of the (Windows) shell environment of 255 bytes.
If I remember correctly there was a work-around for this as there have been several of these limits in the past. The next one being 32k (not sure if 32k Unicode chars or 32k bytes)

me-no-dev, 2023-10-10:
this is the structure of the toolchain. There are reasons for it and we do not control it, nor can alter it. We control only `riscv32-esp-elf-gcc/esp-12.2.0_20230208` which is the package name and version. We can shorten it slightly, but that would not help for longer usernames... we will see how we will solve this.

me-no-dev, 2023-10-10:
> By the way, not sure if it is a limit of GCC.

It is. Windows supports longer paths than 200 symbols. You can access the files otherwise. Only GCC does not see them.

TD-er, 2023-10-10:
I know Windows does support it, but some shell environments do have different limits.
Not sure if this differs between `cmd` and PowerShell , but that would be an easy fix if these actually differ here.

omerk, 2023-10-10:
As expected, `3.0.0-alpha1` with Arduino IDE 2.2.1 on Windows fails to compile:
```
In file included from c:\users\omerk\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-12.2.0_20230208\riscv32-esp-elf\include\c++\12.2.0\stdlib.h:36,
                 from C:\Users\omerk\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-f0437b945f/esp32c6/include/newlib/platform_include/assert.h:13,
                 from c:\users\omerk\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-12.2.0_20230208\riscv32-esp-elf\sys-include\sys\reent.h:503,
                 from C:\Users\omerk\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-f0437b945f/esp32c6/include/newlib/platform_include/sys/reent.h:9,
                 from c:\users\omerk\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-12.2.0_20230208\riscv32-esp-elf\sys-include\stdio.h:60,
                 from C:\Users\omerk\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha1\cores\esp32/Arduino.h:27,
                 from C:\Users\omerk\AppData\Local\Temp\arduino\sketches\9EA17A2B30510F72480E2CA2A9127035\sketch\Blink.ino.cpp:1:
c:\users\omerk\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-12.2.0_20230208\riscv32-esp-elf\include\c++\12.2.0\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

Using `arduino-cli` inside WSL (Ubuntu 20.04) works just fine though:
```
$ arduino-cli compile --fqbn esp32:esp32:esp32c6 --build-path blink-c6/bin blink-c6
Sketch uses 181666 bytes (13%) of program storage space. Maximum is 1310720 bytes.
Global variables use 8928 bytes (2%) of dynamic memory, leaving 318752 bytes for local variables. Maximum is 327680 bytes.

Used platform Version      Path
esp32:esp32   3.0.0-alpha1 /home/omer/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha1
```

Following [these instructions](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=powershell) didn't fix the issue.

IsraelMek, 2023-10-10:
@Hook654321 For the manual fix,  I search and copied the missing header file(c++config.h) to the folder indicated in the error message which is c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-12.2.0_20230208\riscv32-esp-elf\include\c++\12.2.0\bits.

sfe-SparkFro, 2023-10-10:
FWIW my workaround is to simply rename the folder `esp-12.2.0_20230208` to `esp-12.2.0`. This shortens the path length just enough for GCC to find the needed files, no need to manually install anything anywhere. I understand this isn't a perfect solution for everyone (eg. those with long usernames), but might be helpful for folks wanting to use it right now.

Guytou11, 2023-10-12:
Hi,  Is there any update on when Arduino will support the ESP32-C6 ?    Is there perhaps a Zigbee library available that could be used on PlatformIO for the ESP32-C6 ?

thhkd, 2023-10-20:
> Hi, Is there any update on when Arduino will support the ESP32-C6 ? Is there perhaps a Zigbee library available that could be used on PlatformIO for the ESP32-C6 ?

ESP32-C6 y supported in arduino IDE using the board extension ESP32 by Espressif Systems version 3.0.0 alpha2. I hope it won't take long to have it available in PlatformIO aswel.

VojtechBartoska, 2023-10-20:
Hi, yes, you can already used development branch in Arduino IDE for ESP32-C6 support, keep in mind that not everything is supported yet.

[81 more replies]