espressif/arduino-esp32 #7715: ESP32 Board Not downloaded in Arduino 1.8.19 IDE

mmfloom, 2023-01-15:
### Board

AI thinker ESP32-CAM

### Device Description

AI thinker ESP32-CAM

### Hardware Configuration

Camera [OV2640]

### Version

v2.0.6

### IDE Name

Arduino 1.8.19

### Operating System

windows 10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

1. Hi, i am 
[image]
[image]
trying to install ESP32 by using the Stable Link: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

2. I have also tried after detecting all files in my C:\Users\user\AppData\Local\Arduino15 

3. but no use and I have received the below error message 

 "Error downloading https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json"

4. I tying find solutions online, but I can't find 
[image]
[image]

### Sketch

```cpp
no sketch
```

### Debug Message

```plain
I have received the below error message 

Error downloading https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-01-15:
Are you able to get to https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json in a web browser?

mmfloom, 2023-01-16:
No, I am not able to get in a web browser, but some other links working,
please check this video.

https://user-images.githubusercontent.com/104716903/212587973-c2158265-7df2-4acf-bcfb-7c7b00c12119.mp4

VojtechBartoska, 2023-01-16:
@mmfloom can you try another browser? It's a weird issue. Just an idea, maybe it's blocked by some "security related app" you have installed?

mmfloom, 2023-01-17:
> VojtechBartoska

yes, I tried on mobile also, ESP32 link is not working, but the RP2040 link is working, I have attached that screen shorts for your reference 

https://user-images.githubusercontent.com/104716903/212813654-ef55ebb6-2449-4d4c-9b37-aca781fefefa.mp4

https://user-images.githubusercontent.com/104716903/212813676-b8a8fbbe-989d-4d52-8e52-d0d806791e4b.mp4

[image]

[image]

mmfloom, 2023-01-18:
HI... the issue is solved by **enabling the VPN** !!!
please check the video,
thanks a lot for your support.

https://user-images.githubusercontent.com/104716903/213059289-ebeeacbc-68e6-4f24-a172-35b93136ea0a.mp4

closed 2023-01-18 by VojtechBartoska

rohanpatil-rc, 2023-02-12:
I used vpn to put preference link  first and then downloaded(without vpn) from the board manger. thanks for your support

the-confused-genius, 2023-06-16:
I have got the same issue. anyway, to resolve this without a vpn ?, and what's actually causing the issue?

mmfloom, 2023-06-17:
I don't know exactly, but there is a chance that your country security system ( for example in India The National Cyber Coordination Centre (NCCC)) may mistakenly block this address or may mistakenly banned this address. So you need to use the VPN, otherwise you may do lot of research, if you find any alternative please update here.

camp-easy, 2023-11-04:
hello, i receive the same issue..is the vp the only solution for this problem??? thank you in adveance.