espressif/arduino-esp32 #7716: ESP32 v2.0.6

AlanLL, 2023-01-15:
### Board

ESP32_Cam

### Device Description

None

### Hardware Configuration

No

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

dl_lib_matrix3d.h is missing from release 2.0.6
CamWebServer compiles on v 1.0.6
But NOT on V2.0.6
It appears that the include libaraies are not included in V2.0.6

### Sketch

```cpp
/*
  ESP32CAM Robot Car
  app_httpd.cpp (requires esp32cam-robot.ino)
  Based upon Espressif ESP32CAM Examples
  Uses TBA6612FNG H-Bridge Controller
  
  DroneBot Workshop 2021
  https://dronebotworkshop.com
*/

#include "dl_lib_matrix3d.h"
#include <esp32-hal-ledc.h>
#include "esp_http_server.h"
#include "esp_timer.h"
#include "esp_camera.h"
#include "img_converters.h"
#include "Arduino.h"
```

### Debug Message

```plain
dl_lib_matrix3d.h: No such file or directory
```

### Other Steps to Reproduce

ESP32 V1.0.6 compiles and function sucessfully.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-01-15:
esp-dl has changed since 1.0.6, and many parts were simply scrapped as they were too slow on the esp32.  You can try to include dl_math_matrix.hpp, but it is very possible the function you are looking for is simply gone.

Jason2866, 2023-01-15:
It is gone! See the changed examples included in the core 2.0.6 release.

VojtechBartoska, 2023-02-16:
This have been changed, closing as won't fix.

barnareds, 2023-08-10:
I have code for a esp32-cam car. What was "dl_lib_matrix3d.h" used for and how can I replace it so it works on 2.0.11? 
Thanks for any help.

me-no-dev, 2023-08-10:
it was used for face detection and face recognition. Now API is slightly different. You ca read more [here](https://github.com/espressif/arduino-esp32/blob/release/v2.x/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp)